Koenkk/zigbee2mqtt #2346: Ikea E1524/E1810 no communication after pairing

aforch, 2019-11-16:
Running zigbee2mqtt version 1.7
I was able to pair the remote control exactly as described in the device pairing instructions and received the following output in the logs (debug level)

```
debug 2019-11-15T17:32:26: Device ‘0x000b57fffec49812’ announced itself
info 2019-11-15T17:32:26: MQTT publish: topic ‘homeassistant/sensor/0x000b57fffec49812/action/config’, payload ‘{“icon”:“mdi:gesture-double-tap”,“value_template”:"{{ value_json.action }}",“state_topic”:“zigbee2mqtt/0x000b57fffec49812”,“json_attributes_topic”:“zigbee2mqtt/0x000b57fffec49812”,“name”:“0x000b57fffec49812_action”,“unique_id”:“0x000b57fffec49812_action_zigbee2mqtt”,“device”:{“identifiers”:[“zigbee2mqtt_0x000b57fffec49812”],“name”:“0x000b57fffec49812”,“sw_version”:“Zigbee2mqtt 1.7.0”,“model”:“TRADFRI remote control (E1524/E1810)”,“manufacturer”:“IKEA”},“availability_topic”:“zigbee2mqtt/bridge/state”}’
info 2019-11-15T17:32:26: MQTT publish: topic ‘homeassistant/sensor/0x000b57fffec49812/battery/config’, payload ‘{“unit_of_measurement”:"%",“device_class”:“battery”,“value_template”:"{{ value_json.battery }}",“state_topic”:“zigbee2mqtt/0x000b57fffec49812”,“json_attributes_topic”:“zigbee2mqtt/0x000b57fffec49812”,“name”:“0x000b57fffec49812_battery”,“unique_id”:“0x000b57fffec49812_battery_zigbee2mqtt”,“device”:{“identifiers”:[“zigbee2mqtt_0x000b57fffec49812”],“name”:“0x000b57fffec49812”,“sw_version”:“Zigbee2mqtt 1.7.0”,“model”:“TRADFRI remote control (E1524/E1810)”,“manufacturer”:“IKEA”},“availability_topic”:“zigbee2mqtt/bridge/state”}’
info 2019-11-15T17:32:26: MQTT publish: topic ‘homeassistant/sensor/0x000b57fffec49812/linkquality/config’, payload ‘{“unit_of_measurement”:"-",“value_template”:"{{ value_json.linkquality }}",“state_topic”:“zigbee2mqtt/0x000b57fffec49812”,“json_attributes_topic”:“zigbee2mqtt/0x000b57fffec49812”,“name”:“0x000b57fffec49812_linkquality”,“unique_id”:“0x000b57fffec49812_linkquality_zigbee2mqtt”,“device”:{“identifiers”:[“zigbee2mqtt_0x000b57fffec49812”],“name”:“0x000b57fffec49812”,“sw_version”:“Zigbee2mqtt 1.7.0”,“model”:“TRADFRI remote control (E1524/E1810)”,“manufacturer”:“IKEA”},“availability_topic”:“zigbee2mqtt/bridge/state”}’
```
After that, whenever I push any of the buttons on the remote I get… nothing at all.
I have removed and repaired 3 times, every time is the same; pairing works and then nothing after.
Is there a required firmware version necessary on the remote control itself?

Koenkk, 2019-11-16:
What firmware are you running?

aforch, 2019-11-16:
I don't know, to be honest. I packed up my hubs when I started using zigbee2mqtt, so it's probably been a year since it was updated. That's why I was wondering if there is a device firmware dependency. 
I will have to try to find my ikea hub and see if I can update the remote control.

Koenkk, 2019-11-16:
I don't mean the fw version of the remote control but of your cc2531 adapter (it is logged on zigbee2mqtt startup). Make sure that it is on the latest version.

aforch, 2019-11-16:
Ah, gotcha. I'm running at cc2530 adapter, probably it will need to be updated.
`info  2019-11-15T17:15:40: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20182308}}'`

Koenkk, 2019-11-16:
Yes you need to update, after updating it will work (if not feel free to re-open).

closed 2019-11-16 by Koenkk

adrianmihalko, 2020-01-12:
I have this version and same problem.
zigbee2mqtt:info  2020-01-12 21:01:44: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'

@aforch do you resolved the issue by upgrading?

aforch, 2020-01-12:
> I have this version and same problem.
> zigbee2mqtt:info 2020-01-12 21:01:44: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'
> 
> @aforch do you resolved the issue by upgrading?

Yes, I upgraded to "revision":20190619 and now it's working