Koenkk/zigbee2mqtt #2347: last_seen config keeps getting overwritten

tavalin, 2019-11-16:
# Bug Report
## What happened

I have set the `last_seen` attribute in the `configuration.yaml` to:

`last_seen: ISO_8601_local`

However for some reason it keeps getting overwritten with:

`last_seen: epoch`

and therefore the `last_seen `data in the topics are output in epoch format instead of my local time.

## What did you expect to happen

The `last_seen` format to stay as specified - which is ISO_8601_local as expected by my integrations (which notify me if a device hasn't been seen for a period of time).

The fact they are being output in epoch is currently breaking my integration.

## How to reproduce it (minimal and precise)

I am running z2m through docker:

1. Stop docker container 
2. Set `last_seen` attribute to  `ISO_8601_local` in `configuration.yaml`
3. Start docker container
4. Wait ~10 seconds for start up to complete
4. Check file - `last_seen` is displayed as `last_seen: epoch`

## Debug Info

Currently using the following z2m version:

`{"version":"1.7.1+dev","commit":"e51ed89","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}`

sjorge, 2019-11-16:
Might be due to this commit https://github.com/Koenkk/zigbee2mqtt/commit/44db557a0c83f419d66755d14e460cd78bd6204e

tavalin, 2019-11-16:
The docker image I'm using is from an earlier commit.  I'd also say it's something more fundamental than just the logger.js change as z2m is actually overwriting my config (rather than just ignoring it)

carldebilly, 2019-11-16:
Set the `AutosetLastSeen` to `false` in _Zigbee2MqttAssistant_: https://github.com/yllibed/Zigbee2MqttAssistant#settings

carldebilly, 2019-11-16:
Has been fixed in Zigbee2MqttAssistant v0.3.105.  Try an update!

tavalin, 2019-11-17:
Oh wow, didn't even come to mind that this behaviour was from Zigbee2MqttAssistant.  I've pulled the latest Zigbee2MqttAssistant image and it has resolved this.

closed 2019-11-17 by tavalin