Koenkk/zigbee2mqtt #2348: After pairing any device I get the error "no lqi"

kitopopo, 2019-11-16:
# Bug Report
  
Hello, after pairing any zigbee device to the cc2531 coordinator or any repeater i have the following problem: My two repeaters (smart plug xiaomi) are shown on the network map the message: "no lqi" and all devices paired to my repeaters disappear, i need repair the 2 repeaters pressing the button 10 seconds . once they have been repaired i have to make a short press the button of all devices of my network.

The only way to pair any device correctly for me withouts problems is:

1) unplug my 2 repeaters (xiaomi smart plug)
2) Pair the device
3) plug again the two reppeaters

following these stepts my device is paired correctly and the repeaters don't show "no lqi" message, all is ok in the network map. Why appear the "no lqi" message in the network map?
The problem also appears in the 1.6.0 zigbe2mqtt version.

Thanks in advanced. Best regards

## What happened
my zigbe devices are unpaired and my repeaters show the message of "no lqi"

## What did you expect to happen
As the ideological operation would be to be able to link any zigbee device near the desired repeater or cordinator, that this device are paired correctly and that all devices remain paired correctly and without displaying the message "no lqi"

## How to reproduce it (minimal and precise)
Only pair any zigbee device to my network zigbee with my 2 repeaters  plugged (xiaomi smart plug)

## Debug Info
zigbee2mqtt version: 1.7.1
CC2531 firmware version: 20190608

rosscullen, 2019-11-16:
Having similar issue:

``
info  2019-11-16T22:00:00: Starting network scan (includeRoutes 'false')
error 2019-11-16T22:00:10: Failed to execute LQI for '0x0017880103f5024d'
error 2019-11-16T22:00:20: Failed to execute LQI for '0x00158d00030a2aab'
error 2019-11-16T22:00:31: Failed to execute LQI for '0x000d6ffffe9038e4'
error 2019-11-16T22:00:41: Failed to execute LQI for '0x00158d00036d31d3'
info  2019-11-16T22:00:41: Network scan finished
``

For reference, the devices are:
0x0017880103f5024d: 915005106701 - Philips Hue white and color ambiance LightStrip plus (Router)
0x00158d00030a2aab: QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
0x000d6ffffe9038e4: E1746 - IKEA TRADFRI signal repeater (Router)
0x00158d00036d31d3: QBKG12LM - Xiaomi Aqara double key wired wall switch (EndDevice)

The Aqara switches are having an issue at the moment with a long delay before executing but not sure if thats related (still looking into it).

zigbee2mqtt version: 1.7.1
CC2531 firmware version: 20190608
Synology Docker & Home Assistant 0.101

Koenkk, 2019-11-18:
Network scans are very heavy on the network, therefore devices could fail to respond or the CC2531 is not able to handle all requests. The CC2531 source routing firmware could improve this (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

kitopopo, 2019-11-18:
> Network scans are very heavy on the network, therefore devices could fail to respond or the CC2531 is not able to handle all requests. The CC2531 source routing firmware could improve this (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

Hi, thank you for you answer.
 I currently have the firmware CC2531_DEFAULT_20190608., but I have also tested the firmware CC2531 20181027 (this firmware is the oldest I know) and the problem of "no lqi" when pairing a new device with the xiaomi routers plugged in still occurs with the two firmwares .
It is really strange, when the 2 routers enter "no lqi mode"  , no device that had been paired before into the 2 routers is drawn on the map anymore. For solve the problem first I have to repair the 2 routers with a 10-second press and once the 2 routers are repaired. After the routers are repaired  I have to make a small click on each device that had been paired before and they are already drawn again on the network map ( I do not make a new pairing, only one small click).

I will wait a new firmware to see if the problem is solved. thanks for your help.

Best regards

majdzik84, 2019-12-04:
same problem with osram plug

kitopopo, 2019-12-31:
Hi,

The problem has been resolved after upgrade to 1.8.0 version.  Now i can add new devices without unplug repeaters. good job.

Thanks and best regards

kitopopo, 2020-05-22:
Dear @Koenkk 

5 months later we back to the same problem. Randomly the smart sockets zigbee repeaters again show the message:

`Failed to execute LQI for '0x00158d00039bba90'`

I have 3 reapeaters an the problem appears in the 3 reapeaters- I lost 2 hours this morning trying to mount again my zigbe net. Every time thath i pair a device appears the no lqi message.

the zigbe network again returns to be unstable even does not show lines in the zigbee2mqtt map sometimes. I read something yesterday about another user who had the same problem in my old xiaomi pulatiun failure detection issue. Can you solve the problem? I have the 13.1.0 dev branch version. thanks in advanced

kwetnico, 2020-05-22:
i have the same problem since 2 days... :(

[https://github.com/Koenkk/zigbee2mqtt/issues/3586](url)

Koenkk, 2020-05-22:
It's probably not something that can directly be solved from Zigbee2mqtt. Network scan (=LQI) are very heavy for the network. Doing this on a CC2531 and a larger network can make the adapter crash.

kwetnico, 2020-05-22:
> It's probably not something that can directly be solved from Zigbee2mqtt. Network scan (=LQI) are very heavy for the network. Doing this on a CC2531 and a larger network can make the adapter crash.

What can I do ? we can deactivate LQI and my peripherals are working again (I don't know this part)
I no longer have any device that talks to the CC2531 and wants to pair. I haven't added anything for several days.

Koenkk, 2020-05-22:
@kwetnico LQI are executed on the request of a user (network scan), zigbee2mqtt doesn't do this automatically. It could be that a zigbee2mqtt Home Assistant or node red addon does this automatically, so try disabling this.

kitopopo, 2020-05-22:
> Probablemente no sea algo que se pueda resolver directamente desde Zigbee2mqtt. La exploración de red (= LQI) es muy pesada para la red. Hacer esto en un CC2531 y una red más grande puede hacer que el adaptador se bloquee.

Hi Mr @Koenkk ,

I have been using the cc2531 for more than a year and now the lqi is very heavy? I also have to buy the CC2652R? Wow, what used to be wonderful now is no longer worth anything. After opening this issue you solved the lqi problem in a new deb brach and it worked perfectly for 6 months, I have been doing network scans daily and it takes less than 1 second to run the scan on a rapberry pi 4 with 4g ram. -

kitopopo, 2020-05-22:
I have no problem buying the new zzh, but there are no stocks to make the purchase, so far everything has worked well with the cc2531. thanks and best regards

closed 2020-07-26 by stale[bot]

SPEC1AL1ST, 2020-10-22:
Same problem