Koenkk/zigbee2mqtt #2349: Symfonisk remote generates too many events.

freakshock88, 2019-11-16:
# Bug Report
## What happened
I bought a symfonisk remote and paired it to zigbee2mqtt without problems.
However, it is generating 2 'events' per action. For instance when I press the button: 

```
zigbee2mqtt:info  2019-11-16T13:56:42: MQTT publish: topic 'zigbee2mqtt/0xccccccfffebd8924', payload '{"linkquality":13,"rate":195,"brightness":14,"battery":87,"action":"play_pause"}'
zigbee2mqtt:info  2019-11-16T13:56:42: MQTT publish: topic 'zigbee2mqtt/0xccccccfffebd8924', payload '{"linkquality":13,"rate":195,"brightness":14,"battery":87,"action":""}'
zigbee2mqtt:info  2019-11-16T13:56:42: MQTT publish: topic 'zigbee2mqtt/0xccccccfffebd8924', payload '{"linkquality":15,"rate":195,"brightness":14,"battery":87,"action":"play_pause"}'
zigbee2mqtt:info  2019-11-16T13:56:42: MQTT publish: topic 'zigbee2mqtt/0xccccccfffebd8924', payload '{"linkquality":15,"rate":195,"brightness":14,"battery":87,"action":""}'
```
This causes double triggers on automations, of course. 

Especially then I turn the volume up (just slightly in one smooth go), a lot of events get generated:
https://pastebin.com/H6vg5TWQ
Especially turning the volume button locks up my HA instance for a while, and writes a lot to the recorder DB (I excluded it now). 

## What did you expect to happen
I expect button presses to only register once. I expect volume changes to only register one delta, or at least less events to not overload home assistant. 

## How to reproduce it (minimal and precise)
Pair symfonisk to zigbee2mqtt and press button once, for volume problem turn button once.

## Debug Info
zigbee2mqtt version: 1.7.0 (commit #551a58a)
CC253X firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

james-fry, 2019-11-16:
Same root cause as issue https://github.com/Koenkk/zigbee2mqtt/issues/2337
Looks like there needs to be some way to throttle the MQTT messages to <n> per second.

freakshock88, 2019-11-16:
You are right, I solved the problem by adding the option to my device config:
`debounce: 1`
Thanks!

closed 2019-11-16 by freakshock88

tobimori, 2019-11-24:
> 
> 
> You are right, I solved the problem by adding the option to my device config:
> `debounce: 1`
> Thanks!

Didn't make it for me. I only get 'rotate_stop' or the click events.

freakshock88, 2019-11-24:
That's the case also for me, but I just respond to rotate stop commands, and not rotate commands.

tobimori, 2019-11-24:
@freakshock88 How do you determine the rotation direction? Could you send me your script?

freakshock88, 2019-11-24:
> @freakshock88 How do you determine the rotation direction? Could you send me your script?

Here's what I have so far, it also supports multiple media players and runs actions on the currently active one:
https://pastebin.com/Y1SG3Tft

tobimori, 2019-11-24:
> 
> 
> > @freakshock88 How do you determine the rotation direction? Could you send me your script?
> 
> Here's what I have so far, it also supports multiple media players and runs actions on the currently active one:
> https://pastebin.com/Y1SG3Tft

This works really well. Thanks!