Koenkk/zigbee2mqtt #2350: Can't bind Controller with bulp

bncbnc83, 2019-11-16:
I tryed to bind Controllers with Bulp.

It says Successfully bound but it does not work

I tryed do delete configuration and database and start over. but it did not help

Im running with latest 1.7.1 and with firmware CC2531_DEFAULT_20190608.zip

https://pastebin.com/fJSDKVVr

In this ex. 
TRADFRI remote control and TRADFRI bulb E27 CWS opal 600lm

hope someone can help :-)
 

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

didiht, 2019-11-17:
Which firmware version has the IKEA remote control?

There is a different behaviour in the newest one, see #2325

Kryzek, 2019-11-17:
Are you trying to bind remote directly to bulb? Ikea remote can be bound only to group. So you would need to create a group and add bulb to group and then bind remote to group.

bncbnc83, 2019-11-17:
> Which firmware version has the IKEA remote control?
> 
> There is a different behaviour in the newest one, see #2325

{"id":11,"type":"EndDevice","ieeeAddr":"0x000d6ffffe25891c","nwkAddr":22167,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1a1f7","endpointID":1}]}},"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}

from that i think 1.2.223

> Are you trying to bind remote directly to bulb? Ikea remote can be bound only to group. So you would need to create a group and add bulb to group and then bind remote to group.

I tryed to make it directly...i think it has been working that way before.

But tryed to make a group and got the same result

Kryzek, 2019-11-17:
If it is older Ikea firmware, change logging-level to debug. Then you should see groupID the remote has in the log when pressing buttons. Then create group with that ID and add bulb to that group.

bncbnc83, 2019-11-17:
> If it is older Ikea firmware, change logging-level to debug. Then you should see groupID the remote has in the log when pressing buttons. Then create group with that ID and add bulb to that group.

That did the trick.

Is it something new. I did not do that a year ago ?
And there it was working

closed 2019-11-17 by bncbnc83