Koenkk/zigbee2mqtt #2353: Xiaomi door contact sensor won't pair

lancelon, 2019-11-17:
[Re: this thread on the HA forum](https://community.home-assistant.io/t/still-confused-zigbee2mqtt-unable-to-pair-new-zigbee-devices/132659/13?u=daneboom) I'm totally unable to pair a new device.  I'm running 1.5.1 [with the network pass fix](https://github.com/Koenkk/zigbee2mqtt/issues/1756#issuecomment-513544916) as I got stung  when something changed and suddenly a network pass was required.
I'd gladly upgrade to 1.7 but can anyone tell me if the upgrade will likely go smoothly?! I don't have time for debug/fix these days so just want minimum hassle.

Thank you!

sjorge, 2019-11-17:
I overall had less hassle on 1.7.0 with my Xiaomi/Aqara devices than I had on 1.6.0 (not used 1.5.x)
I was running dev most of the time, I do believe you can't got back to 1.6.0 or older once you are on 1.7.0, so something to keep in mind if you do want to check out 1.7.0.

BasTijs, 2019-11-23:
I'm running the newest version and started using it today. I managed to get an ikea bulb working like a charm, but the xiaomi devices won't pair.... (MCCGQ11LM, RTCGQ11LM). Nothing shows up in the log file. Any suggestions?

BasTijs, 2019-11-23:
Mine started working after PI reboot

lancelon, 2019-11-23:
Tried a couple of host reboots, no dice. I've tried 'resetting' the sensor by holding the button in for 10s but seems to have no effect. I'm running Hass 0.92.1., I don't suppose that's the reason for 1.70 not working?  Wish I could try this sensor against another setup but don't know anyone in my city with zigbee/ha stuff.
@Koenkk don't suppose you have ahy thoughts on whether zigbee2mqtt 1.7 HAS to be used with later version of Hass to pair with zigbee devices? Thanks!!

Koenkk, 2019-11-25:
That shouldn't matter, what do you see in your log when enabling debug log and putting the device into pairing mode?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

darnokb, 2019-12-02:
I have the same problem. I have 2 Mija Temp & Hum sensors, 2 Aqara Temp & Hum sensors, 1 Mija door sensor. None is able to join zigbee network.
I have zigbee2mqtt 1.7.1, joining is enabled:
zigbee2mqtt:warn  2019-12-02T17:53:17: `permit_join` set to  `true` in configuration.yaml.

There is no further information in logs, even after setting log level to debug in configuration.yaml.
Any ideas?

darnokb, 2019-12-02:
I had to take out stick from USB port and stick it again. This solved the problem. Strange as hell.

lancelon, 2020-01-21:
Just a quick query, trying to debug this: when you reach your max number of paired devices, would you get any feedback at all from Zigbee2mqtt at that point? Or would it just silently fail to interact in any way with devices? Reason I ask is as well as the Xiaomi sensors failing to pair, I'm now struggling to get an IKEA blind to pair. AFAIK I have one more 'slot' available for a new device but nothing ever shows up in my log...

Koenkk, 2020-01-22:
You dont get any message (the CC2531 does notify zigbee2mqtt about this). You need to make sure that there are enough routers in your network to handle the amount of devices.

lancelon, 2020-01-22:
Thanks. Just to confirm you meant to say that the CC2531 DOES notify zigbee2mqtt that it has reached the max number of devices but zigbee2mqtt doesn't pass it on?

I've deleted a device to make room for more. On my network map I have 13 devices. I am hoping to try again pairing tonight.

Koenkk, 2020-01-22:
The cc2531 does not (unfortunately).

lancelon, 2020-02-03:
Hiya, just wondering, I've got hold of another CC2531. I have used it with TI's packet sniffer to [verify ](https://imgur.com/a/Rc88u1y)that the Xiaomi sensors and the IKEA blind are BOTH definitely  transmitting zigbee packets. I began to wonder if I'd got SUPER unlucky and nothing was actually working...
Anyway I was wondering. Do you reckon I could use another instance of Zigbee2Mqtt and obtain a pairing with the sensors then somehow manually copy and paste the magic info into my 'production' zigbee2mqtt database? Would this even work as an approach?!  My theory is that either the CC2531 dongle itself or zigbee2mqtt is silently rejecting any new pairings. Because the existing sensors which are already paired work flawlessly.
Thanks

Koenkk, 2020-02-03:
No this is not possible, you can try refreshing your CC2531, start zigbee2mqtt and immediately pair the device after that.

closed 2020-04-11 by stale[bot]

drew-kun, 2020-06-09:
Same issue here... Paired 12 devices almost without any issues.. An then two xiaomi devices ([power plug](https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html) and [aqara cube](https://www.zigbee2mqtt.io/devices/MFKZQ01LM.html)) just would not pair.. I tried to restart the HA, restarted zigbee2mqtt, removed and plugged again the CC2531.. 
@Koenkk, you said the post above:
```
try refreshing your CC2531, start zigbee2mqtt and immediately pair the device after that
```
Could you please suggest what you mean by `refreshing your CC2531`?

nekeress, 2020-10-26:
As someone who stumbled into this while having issues with pairing the sensors and searching on Google:
I also couldn't pair the motion sensors, and I solved it by removing the USB stick (the TI chip), then plugging it back and restarting the OS. To me it seemed that the config.yaml for zigbee2mqtt was locked somehow, and no matter if I set the auto-join to true, it didn't matter. I'm not sure which step resolved the problem, but I can report it was fixed and I'm happily using the automatic light in my bathroom now :-)