Koenkk/zigbee2mqtt #2355: IKEA Tradfri light weird behavior

syphernl, 2019-11-17:
On each floor I have a Tradfri bulb. Ideal as a router since its placed pretty centrally. With Z2M 1.6 and below it was working fine. Once in a while it would "drop off" the network but simply powercyling resolved it. Nothing to worry about.

However, last week I updated to 1.7.0 and 1.7.1 and one of the lights is now being marked as unavailable. It cannot be pinged/shows LQI failed in the map. However, according to the map it still serves as a router and has connections to several other devices. It however cannot be controlled or queried, yet it routes. Odd.

I have tried:

1. Removing it from the network, but this didn't work.
2. Resetting / re-pairing it
3. Pairing it to my Deconz stick: First run it was added, yet it could not be controlled (or perhaps I was to impatient
4. Paired it again to the CC-stick, nothing changed.
5. Tried removing it from `database.db` and all other files, eventually it got added again but with `interview_failed`. 
6. Paired it to Deconz again: It could control the light just fine. On. Off. Setting brightness. No problems here.. So the light is fine, something else must be causing this.
7. Tried pairing it again to CC. Again, not working. Interview failed.
8. I have also tried to powercycle all of the routers (a few bulbs and signal repeaters) but this didn't seem to make any difference.

When the bulb turns on, Z2M says that the bulb has announced itself. It briefly becomes "online" / available but it cannot be controlled and after a short while it turns unavailable again.

## Debug Info
zigbee2mqtt version: 1.7.1+dev
CC253X firmware version: 20190619 (source routing)

Koenkk, 2019-11-18:
It could be that the bulb immediately joins the Deconz network as it was connected to that before (I've seen this behaviour multiple times for TRADFRI devices).

syphernl, 2019-11-19:
Once I turn it on again I immediately see:
```
info  2019-11-19 16:34:25: MQTT publish: topic 'zigbee2mqtt/lamp_overloop/availability', payload 'online'
```

And shortly thereafter:
```
error 2019-11-19 16:34:35: Failed to setup reporting for '0x000b57fffeda561e' - Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
error 2019-11-19 16:34:45: Failed to read state of 'lamp_overloop' after reconnect
error 2019-11-19 16:35:00: Publish 'set' 'state' to 'lamp_overloop' failed: 'Error: Timeout - 27211 - 1 - 14 - 6 - 11 after 15000ms'
```

As a test I have the Deconz network open but I don't see it joining that network.

Koenkk, 2019-11-19:
The reset procedure of IKEA bulb can be tricky sometimes, this screenshot is from a users who managed to pair his bulb after a lot of attempts:

[image]

syphernl, 2019-11-23:
Yeah it is a bit tricky but I never had the issue I'm seeing now. I've done the off/on thing for a bunch of times now but the interview keeps on failing.

syphernl, 2019-11-23:
Debug logs:
Immediate after turning on the bulb:

Turn on:
```
  serialport/bindings/poller received "readable" +13s
  serialport/binding-abstract read +13s
  serialport/bindings/unixRead Starting read +13s
  serialport/bindings/unixRead Finished read { bytesRead: 18,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 9b 15 34 01 02 00 00 00 00 d1 fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +0ms
  serialport/stream binding.read finished { bytesRead: 18 } +13s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,75,106,75,106,30,86,218,254,255,87,11,0,142,200] +13s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,75,106,75,106,30,86,218,254,255,87,11,0,142,200] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [75,106,75,106,30,86,218,254,255,87,11,0,142] - 200 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":27211,"nwkaddr":27211,"ieeeaddr":"0x000b57fffeda561e","capabilities":142} +13s
  zigbee-herdsman:controller:log Device announce '0x000b57fffeda561e' +13s
zigbee2mqtt:debug 2019-11-23 12:12:36: Device 'lamp_overloop' announced itself
winston:file written true false
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
zigbee2mqtt:debug 2019-11-23 12:12:36: Setup reporting for '0x000b57fffeda561e' - 1 - genOnOff
winston:file written true false
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":27211,"srcaddr":"0x000b57fffeda561e","srcendpoint":1,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b0018ed1774","dstendpoint":1} +13s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,75,106,30,86,218,254,255,87,11,0,1,6,0,3,116,23,237,24,0,75,18,0,1,55] +13s
  serialport/stream _write 28 bytes of data +6ms
  serialport/binding-abstract write 28 bytes +7ms
  serialport/bindings/unixWrite Starting write 28 bytes offset 0 bytesToWrite 28 +13s
winston:file logged 102659 debug 2019-11-23 12:12:36: Device 'lamp_overloop' announced itself

  serialport/stream _read reading { start: 290, toRead: 65246 } +0ms
  serialport/binding-abstract read +1ms
winston:file logged 102742 debug 2019-11-23 12:12:36: Setup reporting for '0x000b57fffeda561e' - 1 - genOnOff

  serialport/bindings/unixRead Starting read +8ms
  serialport/bindings/unixWrite write returned: wrote 28 bytes +1ms
  serialport/bindings/unixWrite Finished writing 28 bytes +0ms
  serialport/stream binding.write write finished +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +8ms
  serialport/bindings/poller received "readable" +15ms
  serialport/binding-abstract read +15ms
  serialport/bindings/unixRead Starting read +15ms
  serialport/bindings/unixRead Finished read { bytesRead: 6,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 9b 15 34 01 02 00 00 00 00 d1 fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +0ms
  serialport/stream binding.read finished { bytesRead: 6 } +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0} +13s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  serialport/stream _read reading { start: 296, toRead: 65240 } +1ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
zigbee2mqtt:error 2019-11-23 12:12:46: Failed to setup reporting for '0x000b57fffeda561e' - Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
winston:file written true false
winston:file logged 103155 error 2019-11-23 12:12:46: Failed to setup reporting for '0x000b57fffeda561e' - Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```
[... 354 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  serialport/stream _read reading { start: 870, toRead: 64666 } +1ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
  zigbee-herdsman:controller:device Interview - failed for device '0x000b57fffeda561e' with error 'Error: AREQ - ZDO - nodeDescRsp after 10000ms
[... cut ...]
zigbee2mqtt:info  2019-11-23 12:13:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"lamp_overloop"}}'
winston:file written true false
winston:file logged 106711 error 2019-11-23 12:13:51: Failed to interview 'lamp_overloop', device has not successfully been paired

winston:file logged 106883 info  2019-11-23 12:13:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"lamp_over
loop"}}'

```

mpettitt, 2019-11-23:
Getting exactly the same error messages with three Tradfri bulbs here, while a fourth paired without any issues. All had previously been connected as part of a Hassio install, using the same Zigbee dongle. Have tried the reset procedure suggested above, and it made no difference to the results 

The paired one is an LED1545G12 (with white temperature control), and at least one of the others is the model without white temperature - I don't know if this could be affecting the process in some way?

Koenkk, 2019-11-25:
I think the problem depends on how the bulb is being reset, it seems to make a difference e.g. if the on/off's  are shorter/longer. I hope I can soon solve https://github.com/Koenkk/zigbee2mqtt/issues/2396 for IKEA devices, this would provide a much easier and reliable reset procedure.

mpettitt, 2019-11-26:
I'm not convinced about the reset process being the problem here - the process which works reliably under 0.6.0 fails under 0.7.1, even with the same devices being used. I've also reset IKEA bulbs using a Hue Remote, and again, works for 0.6.0, fails for 0.7.1.

I could potentially see it being an issue with a specific firmware version of the bulbs, but I'm not sure how to verify what version they are on without digging out the hub.

Koenkk, 2019-11-26:
@mpettitt the only difference I can think of right now is that shepherd tries to query the node descriptor more times. zigbee-herdsman tries it twice, logic can be found here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L321

Let me know if you are comfortable in changing this, otherwise please post your `node_modules/zigbee-herdsman/dist/controller/model/device.js` file.

mpettitt, 2019-11-26:
I've reverted to 1.6.0 for the moment, for domestic harmony reasons (e.g. wife getting annoyed with lights that were misbehaving or couldn't be controlled), so that file will be whatever was in the 1.6.0 Docker release - I've not changed it at all. (Obviously, I typoed with the version numbers here - all above should be 1.7.1 rather than 0.7.1...) From looking at changelogs, it may not even exist in that version!

I'm guessing that using the Hue remote reset method is equivalent to the Touchlink process? It seems more reliable than the on-off method, although not as convenient if you have high ceilings.

Koenkk, 2019-11-26:
@mpettitt yes it is the same as the hue dimmer switch reset method.

janmisar, 2019-12-14:
Any news here? I tried everything, reset with on/offs, reset with Hue dimmer, nothing works. Still the same error in interview 😞 I''d be glad to send some logs or anything which could help 🙏

exetico, 2019-12-15:
I have the same problem, and after spending more than an hour with my bulb, i simply took the power, and went to bed.

I tried the tips, which are mentioned here, and multiple other places - but it's still an issue. I can't get it to pair successfully.

So, next step for me, will be to update all my IKEA bulbs on the IKEA Gateway. Hereafter, i'll update my ZigBee hardware, add a router (which should arrive, in a few days), and update my docker.

I don't think it will solve all my problems, however, hopefully it help out some of the issues. I would also like to give the "group"-function a try, which arn't an option with my current firmware (on the ZigBee hardware).

If i can help with logs or anything else, i would be happy to do that :-)

_... Cause.. After removing all IKEA bulbs from the Zigbee network, updating them with the IKEA Gateway, and trying to put them back in my main Zigbee-network, i really hope that it will work (o: Else, i'll be in big trouble..._

Koenkk, 2019-12-15:
You can try pairing it with the latest zigbee2mqtt dev branch (it might fix the pairing issues as more attempt are done there).

exetico, 2019-12-15:
Thanks for the tip, @Koenkk ! I'll buy one of the buttons from IKEA, and give a single bulb a try. If the pairing goes smoothly, i'll fix the rest of them If not, i gues i need to rethink what to do (regarding the firmware update for the bulbs).

I have a good amount of bulbs, so i need to have at least some of them working. Softbuttons is mounted on top of the switches, so going manually, will be a pain :D 

I'm still not quite sure about which firmware i should go with for the ZigBee USB-device, itself. If i should just use the lastest one, or what? However, i don't think it's a good idea to put that question here, now that the main issue/discussion is about the IKEA paring-process.

janmisar, 2019-12-16:
> You can try pairing it with the latest zigbee2mqtt dev branch (it might fix the pairing issues as more attempts are done there).

I installed update 1.8.0, which is based on latest dev branch afaik and I get still the same error 😞 The only difference is that it takes longer to fail. Before update it failed after few seconds, now it takes almost a minute - I guess it's caused by more trials as you described.
It's really frustrating... Can I help in any way?

Koenkk, 2019-12-17:
@janmisar it is really strange that in your case nothing works, even the touchlink factory reset. Did this bulb ever manage to pair successfully with zigbee2mqtt? Are any other zigbee network running or perhaps it could be interference? Did you connect your CC2531 through an usb extension cable?

janmisar, 2019-12-17:
1. It's brand new FLOALT panel and it has not been paired with anything yet.
2. I have one led strip connected to Hue bridge (because of sync with philips ambilight tv) so I suppose it is other zigbee network.
3. Yes, I have my CC2531 connected with USB extension but I have never had any problem with this. I paired everything smoothly few months ago even with ext cable 🤷‍♂ 

I will definitely try to turn off Hue bridge to avoid possible interference. And another possible problem could be that my CC2531 is pretty far away from floalt but again... my old floalt is even further and it worked anyway. Could this be a problem?

janmisar, 2019-12-17:
OK, there's definitely something wrong with my floalt 😞 I also have another new one which I didn't even try to pair when first one didn't work... I tried it now and it paired in a second, just works like a charm 🧙‍♂️ The first floalt still without success, so I will try it with Hue or I will just return it to Ikea.

So at least from my point of view, issue is solved, thank you!

exetico, 2019-12-30:
_I just want to make a quick follow up._

My bulb which wouldn't do anything, is now paired again :+1:

I'm on the latest docker branch. But i've also updated the bulb itself.

Everything went smoothly, after two resets (i don't think it was close enough the first time).

Next step for me, is to:
- Grap the rest of the IKEA bulbs
- Update them with the IKEA Gateway
- Pair them again

However, i'm also going to update the firmware on the zigbee-coordinator, itself. But i'm not sure if the public link is the best firmware to go with? Of if i should try out, something else than that? :-)

janmisar, 2019-12-30:
> OK, there's definitely something wrong with my floalt 😞 I also have another new one which I didn't even try to pair when first one didn't work... I tried it now and it paired in a second, just works [...]

Before returning my floalt as broken I tried @Koenkk advice to **turn off all other zigbee devices** and it works - all my floalts are paired 🎉 Thanks again, I hope it will help to the others some time 🙌

ikbenfrank, 2020-01-24:
I had the same problem as mentioned and couldn't get the bulb (LED1545G12) to connect. Tried the method with slow on/off and connected to a ikea remote before connecting to zigbee2mqtt again but all didn't work.
The solution in the end was using the experimental feature touchlink reset as mentioned here: #2396 . The bulb reset in a matter of seconds and reconnected immediatly.

closed 2020-03-31 by stale[bot]

fss, 2020-05-13:
For me power-cycle was the only option that worked :) Not sure if I have the latest firmware, but getting the stick out to re-flash it is not an option atm ;)

WeterPeter, 2022-07-31:
Worked for me, thanks!

drublin, 2024-03-04:
This SLOW cycling didn't work for me.

what worked for me was updating SONOFF 3.0 CC2652P firmware from 20210708 to 20230507. then Force remove and pair again like a charm.

Dongle firmware update here:
https://forum.iobroker.net/post/535605