Koenkk/zigbee2mqtt #2356: Сhanges in the control structure of switch topics Xiaomi QBKG03LM and QBKG04LM

directman66, 2019-11-17:
Hey. QBKG03LM switches could previously be controlled through the topics
zigbee2mqtt/0x00158d00022bc3ab/left/set by sending the payload {"state_left": "OFF"} and zigbee2mqtt/0x00158d00022bc3ab/right/set by sending the payload {"state_left": "OFF"}.

Everything worked well. But after updating to version 1.7.0 the management topics changed. Now you can control it only by sending zigbee2mqtt/0x00158d00022bc3ab/right/set by sending the payload {"state": "OFF"}. I have already adapted the code and control methods when the result of the command corresponds to the command being sent. I do not like that sending the state parameter I see the result in the state_right topic. Can I add an old control option in addition to the existing one?

Koenkk, 2019-11-18:
I'm not aware of any changes regarding this in 1.7, at least it wss not covered by the tests or documented. However when publishing to `zigbee2mqtt/0x00158d00022bc3ab/right/set` the endpoint is already defined, why should zigbee2mqtt allow specifying it two times? (Also in the payload)

directman66, 2019-11-18:
I would like to change the parameter that is returned in  json  response format from the device. This greatly simplifies the work of setting up and integrating equipment.

sjorge, 2019-11-18:
AFAIK if you publish `state_left` or `state_right` to `zigbee2mqtt/0x00158d00022bc3ab/set` you should get the same response back? on `zigbee2mqtt/0x00158d00022bc3ab` ?

Koenkk, 2019-11-19:
Yes this would indeed be more logical but for now it has not been implemented.

directman66, 2019-11-19:
Сan you expect in the near future? It seems there’s nothing much to add to the need :-) everything is simple.

Koenkk, 2019-11-20:
Implemented! (in latest dev branch).

closed 2019-11-20 by Koenkk

directman66, 2019-11-25:
Thanks, similar problem with LLKZMK11LM relay.