Koenkk/zigbee2mqtt #2359: Eurotronic Sprit get local temperature

sti0, 2019-11-17:
# Bug Report
## What happened

Tried to get the `local_temperature `by publishing payload
```
{
    "local_temperature": ""
}
```

to zigbee2mqtt/[device]/get

won't got the actual `local_temperature`, instead it publishs the `local_temperature_calibration `if set before.

## What did you expect to happen

Getting the actual local temperature.

## How to reproduce it (minimal and precise)

1. set a `local_temperature_calibration `(this won't be published, so the payload shows non calibration or an old value)
2. get the `local_temperature`
3. look at the logs: `local_temperature_calibration `value was updated, the `local_temperature` stays on the old value

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190425

sjorge, 2019-12-17:
Yep, I have the same behavior :(

sjorge, 2019-12-17:
I also tried adding 'tz.local_temperature' to the device, but that results in the same.

Koenkk, 2019-12-19:
2 things could be going on:
- The device respond wrong
- The converter is wrong

@sjorge a point to start debugging is here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L2092 I don't have this device so I cannot test it.

sjorge, 2019-12-19:
Is there a way to get the logger in a fromZigbee converter?
I know `meta.logger.debug` works in toZigbee as I used it while working on the system_node stuff for this device that is open for testing, but there seems to be no meta object for fromZigbee.

Koenkk, 2019-12-23:
You can use `console.log('test')`, this will log to stdout.

sjorge, 2019-12-27:
I did some debugging, I got this:
thermostat_att_report::main
thermostat_att_report::localTemperatureCalibration

I added a console.log in thermostat_att_report main block and 2 more, one in each if for localTemp and localTemperatureCalibration... so the get request is hitting the wrong one in thermostat_att_report

sjorge, 2019-12-27:
@Koenkk it looks like entityPublish.js is picking the wrong convertor!

```
zigbee2mqtt:debug 2019-12-27 11:51:06: Received MQTT message on 'zigbee2mqtt/bedroom/radiator/get' with data '{"local_temperature": ""}'
zigbee2mqtt:debug 2019-12-27 11:51:06: Publishing get 'get' 'local_temperature' to 'bedroom/radiator'
thermostat_local_temperature_calibration
```

I added a log entry to both `thermostat_local_temperature_calibration` and `thermostat_local_temperature` and the wrong one gets called!

sjorge, 2019-12-27:
I think I found the problem!

```
const converter = converters.find((c) => c.key.includes(key));
```

`c.key` can be both string or array... (based on some logging for the TRV)

```
occupied_heating_setpoint
unoccupied_heating_setpoint
local_temperature_calibration
system_mode
[ 'eurotronic_host_flags', 'eurotronic_system_mode' ]
eurotronic_error_status
setpoint_raise_lower
control_sequence_of_operation
remote_sensing
current_heating_setpoint
eurotronic_trv_mode
eurotronic_valve_position
thermostat_local_temperature_calibration
```

both local_temperature_calibration and thermostat_local_temperature_calibration would match as `.includes` also works on a string and both strings contain `local_temperature`!

I'm not sure what the correct fix here is, I guess making `key` always an array would be easiest! However that seems like a big change to do. I think two things are needed to fix this.

1. add fz.local_temperature to the trv
2. fix the matching on a string/array

sjorge, 2019-12-27:
Going to experiment with
```const converter = converters.find((c) => (typeof c === 'string' ? c.key == key : c.key.includes(key)));```

In theory it should work

sjorge, 2019-12-27:
OK, that does not seem to work.

sti0, 2019-12-28:
@sjorge @Koenkk The problem is fixed with the PR #2623. Thank you. 

Now the question is, should the device be forced to measure and deliver a actual value if I publish a `local_temperature` payload? Or would it deliver the last published value?

sjorge, 2019-12-28:
Based on the debug logging it reads the value from the device, so it should be up to date.

I’m not sure on the reading of the calibration value, I think it depends on the convertGet part of the toZigbee function. I’m traveling so can’t check the code.

~ sjorge

sti0, 2020-01-09:
This issue is fixed. Thanks.

closed 2020-01-09 by sti0