Koenkk/zigbee2mqtt #2360: Raspberry Pi 4 not pairing

notherealmarco, 2019-11-17:
#What happened

Just installed Zigbee2mqtt on my new Raspberry Pi 4. All seemed to be fine, I started pairing the first device by following the pairing instructions for that device.
I wasn't able to pair the device and I didn't get anything in the logs. I then tried with my older Pi 3 B+: it worked.
Basically, it works with every Raspberry (even a friend's Pi 4), but not with my Pi 4. It does not show anything in the logs, no errors, just nothing.

## What did you expect to happen
Devices being able to pair.

## How to reproduce it (minimal and precise)
- enable joining mode
- follow the instructions for the device to pair
- device doesn't pair

## Debug Info
zigbee2mqtt version: 1.7.1
CC2531 firmware version: CC2531_DEFAULT_20190608

Koenkk, 2019-11-18:
Do you have the CC2531 connected through an USB extension cable? I read that the pi 4 produces more interference.

notherealmarco, 2019-11-18:
No extension cable, just connected to the USB port. Tried both 3.0 and 2.0 USB ports.

snlk, 2019-11-21:
I think that I have the same problem. On my RPI4 zigbee2mqtt starts properly (debug enabled). There are no errors. Pairing mode enabled. When I try to pair any device literally nothing happens. I have done so many things allready (restarting, using diffrend usb ports, reflashing, reinstalling z2m). Nothing seems to work. Than I installed zigbee2mqtt on my PC (UBUNTU) and everything works perfect.
I am lost...

Koenkk, 2019-11-21:
Do you use an extension cable?

snlk, 2019-11-21:
no, it's connected directly

Koenkk, 2019-11-21:
Try with an extension cable, pi 4 seems to interfere otherwise.

snlk, 2019-11-21:
@Koenkk It fixed this problem. Thank you for help.

Koenkk, 2019-11-21:
Thanks for confirming, I've added this to the docs.

closed 2019-11-21 by Koenkk