Koenkk/zigbee2mqtt #2361: Ikea Tradfri LEDs - Connection lost after switching off the lights for a longer period

Donkie77, 2019-11-18:
Until 2 weeks ago I had 2 ikea tradfri LED's running smoothly on Hass.io with Zigbee2MQTT and a self flashed CC2531 sniffer (with the latest coordinator firmware).
But after adding 7 more LEDs the problems started. I was able to pair then one by one without issues. Controlling was also fine. However when I switch off the lights hardwire at night (cutting the power by the wall switch) and turn them on all connections seem to be lost the next morning. A "short" switch-off, lets say for a minute was not problem.

I've tried everything. Using another CC2531 (from my brother who has no issues at all), reinstalling Hass.io and using the normal or Edge version of Zigbee2MQTT (which is running now). But still the same issues.

Log: https://pastebin.com/6bVRv164

Many thanks for your support!

mhgvdd, 2019-11-19:
Update: when restarting Zigbee2MQTT I get a errors starting the hersmen:
info  2019-11-19 07:19:07: Logging to directory: '/share/zigbee2mqtt/log/2019-11-19.07-19-06'
info  2019-11-19 07:19:07: Starting zigbee2mqtt version 1.7.1+dev (commit #ba18c5f773c160b54977908ed241cd40547a2fff)
info  2019-11-19 07:19:07: Starting zigbee-herdsman...
error 2019-11-19 07:19:15: Error while starting zigbee-herdsman

Rebooting Hass.io resolved the issue until I switch-off my light for a longer period.

Any clues how to resolve this?

Koenkk, 2019-11-19:
Is there anything logged when repowering the bulbs?

Please check with debug logging enabled.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

mhgvdd, 2019-11-20:
Hi Koen,
First of all much appreciation for the work you are doing!!!
I turned off some lights (with the wall switch) for appr. 10 minutes and turned it back on again and the connection is lost.
Logfile when trying controlling them: https://pastebin.com/PsqdK25i

Koenkk, 2019-11-21:
- What firmware version are you using?
- Do you see anything in the log when powering on the device?

closed 2020-01-27 by stale[bot]