Koenkk/zigbee2mqtt #2362: Change request, alternative mqtt topic

jenskaa, 2019-11-18:
I am using node-red to controle zigbee devices, it would be nice if I could have two topics when communicating with zigbee2mqtt. 

One topic where I could send my request to (ex. mqtt2zigbee) and one that contains the responces (ex zigbee2mqtt)

In that way it would be easy the prevent listening to my own requests.

The zigbee2mqtt implementation could listen on both topics but only publish on the "primary" topic. This solution would also be backward compatible.

My installation:
zigbee2mqtt version: 1.6.0

sjorge, 2019-11-18:
I use node-red too and I just filter out all the messages with a topic ended in '/set' as those are the only ones being send out?

jenskaa, 2019-11-19:
You are right, but then you need to write some node-red code to do the filtering. For my usage of zigbee2mqtt I will also have to filter messages that ends with /set/state, because I have chosen to use that format.

sjorge, 2019-11-19:
Still not hard to do with regex, all my mqtt_in nodes have a node after that do filtering anyway...

Something like this should match `\/set(\/?[a-z]*)$` then just return null if matched or msg if not.

jenskaa, 2019-11-19:
The issue is not if its hard to do or not. But it results in unnecessary cluttering the node-red "scripts" with an extra function for each mqtt receiver node. This could be avoid with my suggestion, I really do not want to take action on my own messages .

closed 2020-01-26 by stale[bot]