Koenkk/zigbee2mqtt #2363: Xiaomi Aqara WSDCGQ11LM - Stops responding after around an hour and needs repair

jameswarner62, 2019-11-18:
Have my first Xiaomi Aqara WSDCGQ11LM (on supported list) sensor attached - running stable  1.7.1

The device works for around an hour after pairing with the network, and then stops responding. It seems to have some variable refresh rate based on changes in sensor values, but otherwise pings back at a basic rate. I have tried changing the battery on the device. The only way to get it back is to repair the device.  Anyone seen similar behaviour before?

brendanmullan, 2019-11-18:
How far away from the coordinator is the sensor? If the connection is very weak that might explain it dropping? Try placing the sensor near the coordinator and see how long it stays responsive.

jameswarner62, 2019-11-18:
It was in the next room, but now re-paired and sitting adjacent - link quality for the time it was last responding was around 50, but slowly dropped to it's last value of 18. You may have something there - it's almost like the device is ramping the transmit power down over the course of 30 minutes...

jameswarner62, 2019-11-20:
OK it's been stable since much closer to the coordinator. Any idea why a loss of signal causes permanent dis-join from the network?

zorruno, 2019-11-21:
I can add a 'me too' to this, 

2x Xiaomi Aqara WSDCGQ11LM and they drop off after short order and stop reporting. Not sure of the timeframe yet.  It appears that pressing the button on them seems to get them back reporting to mqtt (I don't have to re-pair)

- Link q is no less than 60 when first connected 
- these are my only zigbee devices connected
- using a CC2531 with antenna 
- devices are a maximum of 12m away and close to line of sight  
- battery reports 80% 
- using docker koenkk/zigbee2mqtt (I'll pull the latest for more testing)

I'll attempt to do some more testing over the next week and also have a 3rd Aquara device I'll sit much closer to see if it is a range issue.  I don't own any other zigbee devices currently to test my setup, but I'll try and find something.

pcb1962, 2019-11-22:
> * battery reports 80%

I have found that 80% is the point at which Aqara sensors start to become unreliable. By then voltage is down to about 2.9 volts, the 80% figure doesn't mean that you're only 20% of the way to needing a new battery.

zorruno, 2019-11-27:
The 80% figure was the lower of the two for me (one is mid 90s) but I'll grab some new ones to eliminate that.  I'll be doing more testing as I get time but at the moment they certainly don't seem to stay regularly reporting for some reason.  I've turned on 'last seen' and 'elapsed' in config advanced section and I'll try and graph that over a few weeks.

zorruno, 2019-12-21:
I can add more here to say that my network with a couple of WSDCGQ11LM sensors is now pretty reliable.  I've plotted for a couple of weeks and here is some data for one sensor for the last 3 days.  It is around 12m away from the coordinator (CC2531 with antenna) with a couple of walls between.  

I do have a CC2530 router that is closer but my network map shows it isn't paired to it.  Graphs show Temp, and elapsed timeframes of LastSeen.  There is a second plot of reported battery and link quality.

https://snapshot.raintank.io/dashboard/snapshot/oBfgekoisRYhNcoQXxxwGS5CzRfXebrf?orgId=2

It looks like there only a couple of times where reporting seems to be a bit delayed, and maximum period is 60 minutes.

closed 2020-02-27 by stale[bot]