Koenkk/zigbee2mqtt #2365: Innr SP-120 not reporting Power

FutureCow, 2019-11-18:
Running latest 1.7.1-dev

I've connected 2 new Innr SP-120 power sockets. The sockets are working correctly for turning ON/OFF. But they should also report power measurement, but i don't see any report in the MQTT topic.

Database.db:
```{"id":9,"type":"Router","ieeeAddr":"0x00158d000342af39","nwkAddr":35504,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 120","epList":[1,2],"endpoints">
{`
MQTT topic:
```
debug 2019-11-18T21:31:01: Received MQTT message on 'zigbee2mqtt/Schakel_Lamp_Bank/set' with data 'ON'
debug 2019-11-18T21:31:01: Publishing 'set' 'state' to 'Schakel_Lamp_Bank'
info  2019-11-18T21:31:01: MQTT publish: topic 'zigbee2mqtt/Schakel_Lamp_Bank', payload '{"state":"ON","last_seen":1574112661660}'
```
Debug  error connecting the plug:
```debug 2019-11-18T21:16:01: Device 'Schakel_Lamp_Bank' announced itself
info  2019-11-18T21:16:01: Configuring 'Schakel_Lamp_Bank'
error 2019-11-18T21:16:11: Failed to configure 'Schakel_Lamp_Bank', attempt 2 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/home/frans/domotica/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

Debug error pairing the plug:
```info  2019-11-18T21:27:08: Device 'Schakel_Lamp_Bank' joined
info  2019-11-18T21:27:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Schakel_Lamp_Bank"}}'
info  2019-11-18T21:27:08: Starting interview of 'Schakel_Lamp_Bank'
info  2019-11-18T21:27:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Schakel_Lamp_Bank"}}'
debug 2019-11-18T21:27:08: Device 'Schakel_Lamp_Bank' announced itself
debug 2019-11-18T21:27:08: Received Zigbee message from 'Schakel_Lamp_Bank', type 'readResponse', cluster 'genBasic', data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}' from endpoint 1 with groupID 0
debug 2019-11-18T21:27:08: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}'
debug 2019-11-18T21:27:08: Received Zigbee message from 'Schakel_Lamp_Bank', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
debug 2019-11-18T21:27:08: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":2}'
debug 2019-11-18T21:27:08: Received Zigbee message from 'Schakel_Lamp_Bank', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}' from endpoint 1 with groupID 0
debug 2019-11-18T21:27:08: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}'
info  2019-11-18T21:27:08: Successfully interviewed 'Schakel_Lamp_Bank', device has successfully been paired
info  2019-11-18T21:27:08: Device 'Schakel_Lamp_Bank' is supported, identified as: Innr Smart plug (SP 120)
info  2019-11-18T21:27:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Schakel_Lamp_Bank","model":"SP 120","vendor":"Innr","description":"Smart plug","supported":true}}'
```

Koenkk, 2019-11-19:
What commit are you on? As this commit may fix it: https://github.com/Koenkk/zigbee2mqtt/commit/eea660374f3bfb69ae69512131db6a9ff20ee7e1

FutureCow, 2019-11-19:
Did a update to the latest: zigbee2mqtt version 1.7.1+dev (commit #461544e)

Still no power report:
```
info  2019-11-19 18:44:33: Device '0x00158d0003892907' joined
info  2019-11-19 18:44:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0003892907"}}'
info  2019-11-19 18:44:33: Starting interview of '0x00158d0003892907'
info  2019-11-19 18:44:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0003892907"}}'
debug 2019-11-19 18:44:33: Device '0x00158d0003892907' announced itself
info  2019-11-19 18:44:41: MQTT publish: topic 'zigbee2mqtt/Sensor_PIR_Achterdeur', payload '{"battery":100,"voltage":3005,"linkquality":33,"occupancy":false,"last_seen":1574185391515,"elapsed":90001}'
info  2019-11-19 18:44:43: MQTT publish: topic 'zigbee2mqtt/Sensor_PIR_Vrij', payload '{"occupancy":false,"linkquality":36,"battery":100,"voltage":3005,"last_seen":1574185393029,"elapsed":90001}'
debug 2019-11-19 18:44:47: Received Zigbee message from '0x00158d0003892907', type 'readResponse', cluster 'genBasic', data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}' from endpoint 1 with groupID 0
debug 2019-11-19 18:44:47: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}'
debug 2019-11-19 18:44:47: Received Zigbee message from '0x00158d0003892907', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
debug 2019-11-19 18:44:47: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":2}'
debug 2019-11-19 18:44:48: Received Zigbee message from '0x00158d0003892907', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}' from endpoint 1 with groupID 0
debug 2019-11-19 18:44:48: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}'
info  2019-11-19 18:44:48: Successfully interviewed '0x00158d0003892907', device has successfully been paired
info  2019-11-19 18:44:48: Device '0x00158d0003892907' is supported, identified as: Innr Smart plug (SP 120)
info  2019-11-19 18:44:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d0003892907","model":"SP 120","vendor":"Innr","description":"Smart plug","supported":true}}'
debug 2019-11-19 18:44:48: Received Zigbee message from '0x00158d0003892907', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}' from endpoint 1 with groupID 0
debug 2019-11-19 18:44:48: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}'
```
```
debug 2019-11-19 19:26:50: Received MQTT message on 'zigbee2mqtt/0x00158d0003892907/set' with data 'ON'
debug 2019-11-19 19:26:50: Publishing 'set' 'state' to '0x00158d0003892907'
info  2019-11-19 19:26:50: MQTT publish: topic 'zigbee2mqtt/0x00158d0003892907', payload '{"state":"ON","last_seen":1574188010506}'
```

Koenkk, 2019-11-19:
Can you pist the log when starting zigbee2mqtt until 30 seconds?

FutureCow, 2019-11-19:
https://pastebin.com/raw/0xbc8j34

Koenkk, 2019-11-19:
The problem is
```
error 2019-11-19 18:38:00: Failed to configure 'Schakel_Lamp_Kerstboom', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/home/frans/domotica/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

For some reasons the plug fail to configure. As I also have a SP 120 I've checked again but for me it works properly. I guess you also won't get state changes when turning the switch of via the button on it.

I'm not sure why the configure fails, but could you try to temporarily move the plug very close to the coordinator and check if it configures there?

FutureCow, 2019-11-19:
- Removed the device
- Placed in next to the coordinator (50 cm)
- Repaired
- Restarted z2m:

```
info  2019-11-20 00:02:27: Configuring '0x00158d0003892907'
error 2019-11-20 00:02:37: Failed to configure '0x00158d0003892907', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/home/frans/domotica/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```
I don't get a status change when pushing the button.

Koenkk, 2019-11-20:
Do you by any change have another CC2531 where you can sniff the traffic with? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

FutureCow, 2019-11-20:
I've flashed my old CC2531 with the sniffer firmware, and running ZBOSS and see the traffic. But I don't know how and what to read? What do you want to see?

Koenkk, 2019-11-20:
Somewhere in the log you will see a bind request, does the switch respond with a bind response?

FutureCow, 2019-11-21:
The messages are flying around in WireShark:
[image]

Koenkk, 2019-11-22:
Could you send me the wireshark trace? (if you want to send it in private contact me on telegram, @koenkk).

runningman84, 2019-11-30:
My sp 120 report power just fine. But they send a lot of messages especially for voltage change. It looks like my xiamoi sensor messages sometimes get lost... Maybe because the network is overloaded?

FutureCow, 2019-11-30:
The problem is not the overload of the network or the SP-120. But the new coordinator (1352p). It can't configure the SP-120 right...

Koenkk, 2019-12-04:
I've just checked with the CC1352-P but it works correctly in my case, so that doesn't seem to be the problem (honestly I don't know what causes it now).

[image]

FutureCow, 2019-12-04:
Can reflashing the CC1352 solve the problem (without repairing al devices?) Or start Z2M from scratch? Or...?

Koenkk, 2019-12-05:
@FutureCow you can try to reflash the CC1352-P (doesn't require repairing as zigbee2mqtt will automatically restore `coordinator_backup.json`)

FutureCow, 2019-12-12:
Reflashed my CC1352 (erased twice) with UniFlash, but the same problem is still there...

Koenkk, 2019-12-12:
Can you try with temporarily moving `data/coordinator_backup.json` (so that it won't restore your network), reflash the CC1352, change the channel and pair the SP 120 (so that only the SP 120 and the CC1352P are in the network)

wisekki, 2019-12-13:
I'm currently having the same problem with my Innr sockets. I have three of them and they worked with CC2530 very nicely. I changed my coordinator to CC2531+CC2591 and used the latest firmware CC2530_CC2591_DEFAULT_20190608.zip. Zigbee2mqtt is 1.7.1+dev. Now they do not report power consumption at all :(

FutureCow, 2019-12-14:
- reflashed my 1352-P2
- Started a new z2m from scratch
- Paired the SP-120

```zigbee2mqtt:info  2019-12-14 15:09:25: Device '0x00158d000342af39' joined
zigbee2mqtt:info  2019-12-14 15:09:25: MQTT publish: topic 'zigbee2mqtt3/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d000342af39"}}'
zigbee2mqtt:info  2019-12-14 15:09:25: Starting interview of '0x00158d000342af39'
zigbee2mqtt:info  2019-12-14 15:09:25: MQTT publish: topic 'zigbee2mqtt3/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d000342af39"}}'
zigbee2mqtt:debug 2019-12-14 15:09:25: Device '0x00158d000342af39' announced itself
zigbee2mqtt:debug 2019-12-14 15:09:25: Received Zigbee message from '0x00158d000342af39', type 'readResponse', cluster 'genBasic', data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-12-14 15:09:25: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}'
zigbee2mqtt:debug 2019-12-14 15:09:25: Received Zigbee message from '0x00158d000342af39', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-12-14 15:09:25: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2,"appVersion":1,"stackVersion":2}'
zigbee2mqtt:debug 2019-12-14 15:09:25: Received Zigbee message from '0x00158d000342af39', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-12-14 15:09:25: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}'
zigbee2mqtt:info  2019-12-14 15:09:25: Successfully interviewed '0x00158d000342af39', device has successfully been paired
zigbee2mqtt:info  2019-12-14 15:09:25: Device '0x00158d000342af39' is supported, identified as: Innr Smart plug (SP 120)
zigbee2mqtt:info  2019-12-14 15:09:25: MQTT publish: topic 'zigbee2mqtt3/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d000342af39","model":"SP 120","vendor":"Innr","description":"Smart plug","supported":true}}'
zigbee2mqtt:info  2019-12-14 15:09:25: Configuring '0x00158d000342af39'
zigbee2mqtt:info  2019-12-14 15:09:25: Succesfully configured '0x00158d000342af39'
zigbee2mqtt:debug 2019-12-14 15:09:33: Received Zigbee message from '0x00158d000342af39', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":232}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-12-14 15:09:33: MQTT publish: topic 'zigbee2mqtt3/0x00158d000342af39', payload '{"voltage":232,"linkquality":87}'
```
Can there be something wrong with the state.json? I don't like te repair all my devices again....

wisekki, 2019-12-15:
I made a fresh install of Zigee2mqtt and joined SP120 to the network. It seems it doesn't report power at all anymore.. only voltage and link :(

`joulu 15 20:54:17 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:17: Device '0x00158d00034450be' joined
joulu 15 20:54:17 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d00034450be"}}'
joulu 15 20:54:17 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:17: Starting interview of '0x00158d00034450be'
joulu 15 20:54:17 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d00034450be"}}'
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: MQTT publish: topic 'homeassistant/switch/0x00158d00034450be/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x00158d00034450be/set","state_topic":"zigbee2mqtt/0x00158d00034450be","json_attributes_topic":"zigbee2mqtt/0x00158d00034450be","name":"0x00158d00034450be_switch","unique_id":"0x00158d00034450be_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00034450be"],"name":"0x00158d00034450be","sw_version":"Zigbee2mqtt 1.8.0","model":"Smart plug (SP 120)","manufacturer":"Innr"},"availability_topic":"zigbee2mqtt/bridge/state"}'
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: MQTT publish: topic 'homeassistant/sensor/0x00158d00034450be/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/0x00158d00034450be","json_attributes_topic":"zigbee2mqtt/0x00158d00034450be","name":"0x00158d00034450be_power","unique_id":"0x00158d00034450be_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00034450be"],"name":"0x00158d00034450be","sw_version":"Zigbee2mqtt 1.8.0","model":"Smart plug (SP 120)","manufacturer":"Innr"},"availability_topic":"zigbee2mqtt/bridge/state"}'
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: MQTT publish: topic 'homeassistant/sensor/0x00158d00034450be/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00034450be","json_attributes_topic":"zigbee2mqtt/0x00158d00034450be","name":"0x00158d00034450be_linkquality","unique_id":"0x00158d00034450be_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00034450be"],"name":"0x00158d00034450be","sw_version":"Zigbee2mqtt 1.8.0","model":"Smart plug (SP 120)","manufacturer":"Innr"},"availability_topic":"zigbee2mqtt/bridge/state"}'
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: Successfully interviewed '0x00158d00034450be', device has successfully been paired
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: Device '0x00158d00034450be' is supported, identified as: Innr Smart plug (SP 120)
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d00034450be","model":"SP 120","vendor":"Innr","description":"Smart plug","supported":true}}'
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: Configuring '0x00158d00034450be'
joulu 15 20:54:18 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:18: Succesfully configured '0x00158d00034450be'
joulu 15 20:54:25 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:54:25: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":242,"linkquality":57}'
joulu 15 20:55:22 zigbeegw npm[31097]: zigbee2mqtt:info  2019-12-15 20:55:22: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":240,"linkquality":59}'`

CSchlipp, 2019-12-25:
I'm having exactly the same issues after pairing four Innr Plugs to my CC2530. Everything else is working fine, so I'd really appreciate a solution that does not force me to repair all other devices...

elsbrock, 2019-12-27:
It took some time for me until it started reporting power.

CSchlipp, 2020-01-03:
Until today, I've still just received mentioned error messages whenever I've plugged in one of the Innr Plugs. They did neither send any state updates when turned on manually, nor power/current/voltage readings with the most recent dev or stable versions

I've finally worked around this by using zigbee2mqtt version 1.6 with the old zigbee-shepherd module:
Using the old version, all plugs could be configured sucessfully and immediately reported all readings after repairing. The plugs continue to work and keep reporting the values correctly after reverting back to the most recent dev-version.

Seems like this error is related to Zigbee-Herdsman...

dersparkiie, 2020-01-08:
Similar problem here after updating from Version 1.5.1 to 1.8.0.
With Version 1.5.1 there were no problems. Readings have been updated every few seconds. Under Version 1.8.0 only on/off states are commited. Power readings seem to be wrong:
`2020-01-08 08:06:24: MQTT publish: topic 'zigbee2mqtt/Plug_Alexa', payload '{"voltage":229,"linkquality":52,"state":"OFF","power":7,"current":0.067}'`
The plug is turned off but power reading isn't 0!

runningman84, 2020-01-08:
It works for me just fine in 1.8.0. Although I think the amount of updates are too much... but I guess you can change the update interval.

wisekki, 2020-01-08:
> It works for me just fine in 1.8.0. Although I think the amount of updates are too much... but I guess you can change the update interval.

Could you please be kind enough and tell me how to do that? :) I have multiple Innr's and they really flood the logs.

About the devices itself.. 1.8.0 keeps forgetting the devices. I have to rejoin them few times a month. Didn't happen with 1.6.0 :(

runningman84, 2020-01-08:
I have found this:
https://zigbee2mqtt.discourse.group/t/configure-attribute-reporting-for-innr-sp120-or-any-zigbee-device/457

wisekki, 2020-01-08:
> I have found this:
> https://zigbee2mqtt.discourse.group/t/configure-attribute-reporting-for-innr-sp120-or-any-zigbee-device/457

Yes that's my original post :) but if you read it i didn't get it to work...

0rn0lf, 2020-01-18:
Any update about how to change the reporting interval? I installed the plug last week and its flooding my HA database. Over 3GB in a week just from the plug.

dersparkiie, 2020-01-18:
> Any update about how to change the reporting interval? I installed the plug last week and its flooding my HA database. Over 3GB in a week just from the plug.

Totaly different here. My SP120 doesn't update the current at all. 1.8.0 dev commit e766335

0rn0lf, 2020-01-18:
> Totaly different here. My SP120 doesn't update the current at all. 1.8.0 dev commit [e766335](https://github.com/Koenkk/zigbee2mqtt/commit/e76633519f00ae7355afd046323ab98276c13220)

I'm on 1.9.0 and it updates nearly every second.
[image]

dersparkiie, 2020-01-18:
My SP120 can not be configured.
`Jan 18 22:43:43 sparkie xiaomi[2015]: zigbee2mqtt:error 2020-01-18 22:43:43: Failed to configure 'Plug_Alexa', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Jan 18 22:43:43 sparkie xiaomi[2015]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Jan 18 22:43:43 sparkie xiaomi[2015]:     at listOnTimeout (internal/timers.js:531:17)
Jan 18 22:43:43 sparkie xiaomi[2015]:     at processTimers (internal/timers.js:475:7))
Jan 18 22:43:43 sparkie xiaomi[2015]: zigbee2mqtt:info  2020-01-18 22:43:43: Configuring '0x00158d0003569804'
Jan 18 22:43:53 sparkie xiaomi[2015]: zigbee2mqtt:error 2020-01-18 22:43:53: Failed to configure '0x00158d0003569804', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Jan 18 22:43:53 sparkie xiaomi[2015]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Jan 18 22:43:53 sparkie xiaomi[2015]:     at listOnTimeout (internal/timers.js:531:17)
Jan 18 22:43:53 sparkie xiaomi[2015]:     at processTimers (internal/timers.js:475:7))`

runningman84, 2020-01-19:
@0rn0lf @wisekki 
@Koenkk wrote two posts how to reduce the amount of message:
Changing the interval (this seems to be the best solution but needs code change):
https://github.com/Koenkk/zigbee2mqtt/issues/588#issuecomment-575166329
Adding the debounce option (this is just a configuration entry):
https://github.com/Koenkk/zigbee2mqtt/issues/2114

From my point of view the first method is the best because it would also reduce the zigbee traffic. The second option just holds back messages from homeassistant. I hope we get some option in the future which would allow us the change the interval from the configuration.yaml...

IcedEarth, 2020-01-19:
> @0rn0lf @wisekki
> @Koenkk wrote two posts how to reduce the amount of message:
> Changing the interval (this seems to be the best solution but needs code change):
> [#588 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/588#issuecomment-575166329)
> Adding the debounce option (this is just a configuration entry):
> #2114
> 
> From my point of view the first method is the best because it would also reduce the zigbee traffic. The second option just holds back messages from homeassistant. I hope we get some option in the future which would allow us the change the interval from the configuration.yaml...

I agree on making it changeable. Indeed, I experience similar isssues with status reports flooding my FHEM server putting a huge load on my zigbee network making other sensor elss responsible.

runningman84, 2020-01-19:
My impression is that alot of motion events get's lost since I have installed the sp-120 plugs (I have about 10 of them).  Furthermore my mariadb server is under a lot of load... (which I could reduce using the debounce option):
[image]

Koenkk, 2020-01-22:
Instead of making this configurable, the default should be good out-of-the-box.

If somebody can experiment with changing:
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L240
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L249
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L258

to `minimumReportInterval: 5,`, this should change the min report interval to 5 seconds (instead of 1)

After making these changes in `devices.js` trigger a configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

Guide how to get to `devices.js`: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

If this fixes the above issues I can integrate this.

wisekki, 2020-01-23:
I did the experiment, thnx @Koenkk,  for one of the innr's. Seems to be 5 seconds now except for a few lines.. but you guys can decide if it works :)

First one is before the modification. Latter after the change. I also turned off the plug to see how it reports off-status.

<pre>2020-01-23 09:47:47: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":236,"linkquality":65,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:49:11: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":65,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:49:25: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":236,"linkquality":65,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:50:02: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":236,"linkquality":65,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:51:03: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":65,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:51:04: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":65,"power":0,"current":0,"state":"ON"}'
2020-01-23 09:51:05: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":0,"current":0,"state":"ON"}'
2020-01-23 09:51:07: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":22,"current":0.31,"state":"ON"}'
2020-01-23 09:51:08: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":40,"current":0.344,"state":"ON"}'
2020-01-23 09:51:09: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":68,"current":0.566,"state":"ON"}'
2020-01-23 09:51:10: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":80,"current":0.643,"state":"ON"}'
2020-01-23 09:51:11: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":48,"current":0.605,"state":"ON"}'
2020-01-23 09:51:12: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":48,"current":0.452,"state":"ON"}'
2020-01-23 09:51:14: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":37,"current":0.428,"state":"ON"}'
2020-01-23 09:51:18: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":34,"current":0.428,"state":"ON"}'
2020-01-23 09:51:19: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":37,"current":0.39,"state":"ON"}'
2020-01-23 09:51:20: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":34,"current":0.39,"state":"ON"}'
2020-01-23 09:51:21: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":34,"current":0.386,"state":"ON"}'
2020-01-23 09:51:23: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":37,"current":0.386,"state":"ON"}'
2020-01-23 09:51:24: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":65,"current":0.413,"state":"ON"}'
2020-01-23 09:51:25: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":68,"current":0.559,"state":"ON"}'
2020-01-23 09:51:26: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":60,"current":0.574,"state":"ON"}'
2020-01-23 09:51:27: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":30,"current":0.51,"state":"ON"}'
2020-01-23 09:51:28: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":30,"current":0.36,"state":"ON"}'</pre>

After mod:
<pre>2020-01-23 09:56:20: Configuring '0x00158d00034450be'
2020-01-23 09:56:21: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":20,"current":0.306,"state":"ON"}'
2020-01-23 09:56:21: Successfully configured '0x00158d00034450be'
2020-01-23 09:56:22: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":52,"power":20,"current":0.306,"state":"ON"}'
2020-01-23 09:56:26: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":19,"current":0.31,"state":"ON"}'
2020-01-23 09:56:31: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":19,"current":0.307,"state":"ON"}'
2020-01-23 09:56:43: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":19,"current":0.31,"state":"ON"}'
2020-01-23 09:56:49: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":19,"current":0.307,"state":"ON"}'
2020-01-23 09:56:52: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":17,"current":0.307,"state":"ON"}'
2020-01-23 09:56:54: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":54,"power":17,"current":0.31,"state":"ON"}'
[... cut ...]
2020-01-23 09:58:10: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":62,"power":0,"current":0.31,"state":"OFF"}'
2020-01-23 09:58:13: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":65,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:58:31: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":236,"linkquality":59,"power":0,"current":0,"state":"OFF"}'
2020-01-23 09:58:45: MQTT publish: topic 'zigbee2mqtt/0x00158d00034450be', payload '{"voltage":239,"linkquality":65,"power":0,"current":0,"state":"OFF"}'</pre>

wisekki, 2020-01-23:
I'm not sure if this modification has nothing to do with the 'off' -statusreporting but I measured and compared off-reporting from two devices for one hour. Newly configured device reported off-status for 144 times and an old one only 90 times.

Koenkk, 2020-01-23:
@wisekki thanks, I've made 5 seconds the default now.

turboproc, 2020-01-29:
Running 

info  2020-01-29 14:59:29: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
info  2020-01-29 14:59:29: Starting zigbee-herdsman...
info  2020-01-29 14:59:31: zigbee-herdsman started
info  2020-01-29 14:59:31: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}
}'

I've exactly the same issue with 2 SP 120 devices that refuse to be configured.

`{"id":14,"type":"Router","ieeeAddr":"0x00158d000388f6ff","nwkAddr":21318,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 120","epList":[
1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,4,3,6,8,5,2820,1794,10],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"modelI
d":"SP 120","manufacturerName":"innr","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}},"genOnOff":{
"attributes":{"onOff":1}}},"binds":[]},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":1,"stackV
ersion":2,"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1580248978754}
{"id":15,"type":"Router","ieeeAddr":"0x00158d000388f73e","nwkAddr":12532,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 120","epList":[
1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,4,3,6,8,5,2820,1794,10],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"modelI
d":"SP 120","manufacturerName":"innr","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}}},"binds":[]}
,"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20
171027-100","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1580306330952}

so interview phase seems completed but configuration is not.

Any idea how to solve this?`

turboproc, 2020-01-29:
Providing some further details when forcing a configure of the device:

`zigbee2mqtt:debug 2020-01-29 19:10:23: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data '0x00158d000388f73e'
zigbee2mqtt:info  2020-01-29 19:10:23: Configuring '0x00158d000388f73e'
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":12532,"srcaddr":"0x00158d000388f73e","srcendpoint":1,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b00193664ab","dstendpoint":1} +7s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,244,48,62,247,136,3,0,141,21,0,1,6,0,3,171,100,54,25,0,75,18,0,1,177] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0} +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
zigbee2mqtt:error 2020-01-29 19:10:33: Failed to configure '0x00158d000388f73e', attempt 6 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
`

Koenkk, 2020-01-29:
@turboproc could you check if it works correctly with zigbee2mqtt 1.6.0?

turboproc, 2020-01-29:
@Koenkk : Oops, downgrade ? Can of course give it a try. Any instruction to do so ?

Koenkk, 2020-01-29:
@turboproc I would recommend to do it on a separate (temporary) setup, perhaps with a different CC2531. Do you have one?

turboproc, 2020-01-29:
@Koenkk . Yes, that would be possible. Having another raspberry available and CC2531. Just need to understand how to install version 1.6.0

Koenkk, 2020-01-30:
@turboproc when following https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html, after `sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt` do

```
sudo git checkout tags/1.6.0
```

turboproc, 2020-01-31:
Weird, can;t get the second CC2531 (same firmware) with z2m v1.6.0 not really into action. Trying to bind the SP-120  doesn;t show any message although joining is enabled.

Koenkk, 2020-01-31:
@turboproc probably because it pairs to your existing network. Try powering that down.

turboproc, 2020-01-31:
Indeed, had to remove the device from my main network using config/remove. Good news, the SP-120 pairs with the 1.6.0 version of z2m.

turboproc, 2020-01-31:
`zigbee2mqtt:info 2/1/2020, 12:17:21 AM Removed 0x00158d000388f73e
  zigbee2mqtt:debug 2/1/2020, 12:17:22 AM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 2/1/2020, 12:17:22 AM Successfully removed 0x00158d000388f73e
  zigbee2mqtt:info 2/1/2020, 12:17:22 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d000388f73e"}'

  zigbee-herdsman:unpi:parser <-- [254,12,69,202,172,125,62,247,136,3,0,141,21,0,0,0,136] +2m
  zigbee-herdsman:unpi:parser --- parseNext [254,12,69,202,172,125,62,247,136,3,0,141,21,0,0,0,136] +1ms
  zigbee-herdsman:unpi:parser --> parsed 12 - 2 - 5 - 202 - [�}>���] - 136 +1ms
  zigbee-herdsman:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":32172,"extaddr":"0x00158d000388f73e","parentaddr":0} +2m
  zigbee-herdsman:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:unpi:parser <-- [254,13,69,193,172,125,172,125,62,247,136,3,0,141,21,0,142,221] +76ms
  zigbee-herdsman:unpi:parser --- parseNext [254,13,69,193,172,125,172,125,62,247,136,3,0,141,21,0,142,221] +1ms
  zigbee-herdsman:unpi:parser --> parsed 13 - 2 - 5 - 193 - [�}�}>����] - 221 +1ms
  zigbee-herdsman:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":32172,"nwkaddr":32172,"ieeeaddr":"0x00158d000388f73e","capabilities":142} +81ms
  zigbee-herdsman:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":32172,"nwkaddrofinterest":32172} +2m
  zigbee-herdsman:unpi:writer --> frame [254,4,37,2,172,125,172,125,35] +2m
  zigbee-herdsman:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:unpi:parser <-- [254,1,101,2,0,102] +326ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 2 - [] - 102 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +2m
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:unpi:parser <-- [254,18,69,130,172,125,0,172,125,1,64,142,102,17,127,80,0,0,0,80,0,0,18] +29ms
  zigbee-herdsman:unpi:parser --- parseNext [254,18,69,130,172,125,0,172,125,1,64,142,102,17,127,80,0,0,0,80,0,0,18] +1ms
  zigbee-herdsman:unpi:parser --> parsed 18 - 2 - 5 - 130 - [�}�}@�PP] - 18 +0ms
  zigbee-herdsman:znp:AREQ <-- ZDO - nodeDescRsp - {"srcaddr":32172,"status":0,"nwkaddr":32172,"logicaltype_cmplxdescavai_userdescavai":1,"apsflags_freqband":64,"maccapflags":142,"manufacturercode":4454,"maxbuffersize":127,"maxintransfersize":80,"servermask":0,"maxouttransfersize":80,"descriptorcap":0} +366ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":32172,"nwkaddrofinterest":32172} +365ms
  zigbee-herdsman:unpi:writer --> frame [254,4,37,5,172,125,172,125,36] +364ms
  zigbee-herdsman:unpi:parser <-- [254,1,101,5,0,97] +15ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,5,0,97] +1ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 5 - [] - 97 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - activeEpReq - {"status":0} +49ms
  zigbee-herdsman:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:unpi:parser <-- [254,8,69,133,172,125,0,172,125,2,1,2,201] +27ms
  zigbee-herdsman:unpi:parser --- parseNext [254,8,69,133,172,125,0,172,125,2,1,2,201] +1ms
  zigbee-herdsman:unpi:parser --> parsed 8 - 2 - 5 - 133 - [�}�}] - 201 +0ms
  zigbee-herdsman:znp:AREQ <-- ZDO - activeEpRsp - {"srcaddr":32172,"status":0,"nwkaddr":32172,"activeepcount":2,"activeeplist":{"type":"Buffer","data":[1,2]}} +49ms
  zigbee-herdsman:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":32172,"nwkaddrofinterest":32172,"endpoint":1} +50ms
  zigbee-herdsman:unpi:writer --> frame [254,5,37,4,172,125,172,125,1,37] +50ms
  zigbee-herdsman:unpi:parser <-- [254,1,101,4,0,96] +16ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,4,0,96] +1ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 4 - [] - 96 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - simpleDescReq - {"status":0} +50ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:unpi:parser <-- [254,38,69,132,172,125,0,172,125,32,1,94,192,16,0,2,9,0,0,4,0,3,0,6,0,8,0,5,0,4,11,2,7,10,0,3,3,0,25,0,10,0,92] +27ms
  zigbee-herdsman:unpi:parser --- parseNext [254,38,69,132,172,125,0,172,125,32,1,94,192,16,0,2,9,0,0,4,0,3,0,6,0,8,0,5,0,4,11,2,7,10,0,3,3,0,25,0,10,0,92] +1ms
  zigbee-herdsman:unpi:parser --> parsed 38 - 2 - 5 - 132 - [�}�} ^�

  zigbee-herdsman:unpi:parser
  zigbee-herdsman:unpi:parser ] - 92 +1ms
  zigbee-herdsman:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":32172,"status":0,"nwkaddr":32172,"len":32,"endpoint":1,"profileid":49246,"deviceid":16,"deviceversion":2,"numinclusters":9,"inclusterlist":[0,4,3,6,8,5,2820,1794,10],"numoutclusters":3,"outclusterlist":[3,25,10]} +51ms
  zigbee-herdsman:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":32172,"nwkaddrofinterest":32172,"endpoint":2} +51ms
  zigbee-herdsman:unpi:writer --> frame [254,5,37,4,172,125,172,125,2,38] +52ms
  zigbee-herdsman:unpi:parser <-- [254,1,101,4,0,96] +16ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,4,0,96] +1ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 4 - [] - 96 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - simpleDescReq - {"status":0} +52ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:unpi:parser <-- [254,16,69,132,172,125,0,172,125,10,2,94,192,0,16,2,1,0,16,0,68] +27ms
[... cut ...]
  zigbee-herdsman:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100 +1ms
  zigbee-herdsman:znp:SRSP <-- AF - dataRequest - {"status":0} +129ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:unpi:parser <-- [254,3,68,128,0,1,26,220] +23ms
  zigbee-herdsman:unpi:parser --- parseNext [254,3,68,128,0,1,26,220] +0ms
  zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 220 +1ms
  zigbee-herdsman:znp:AREQ <-- AF - dataConfirm - {"stat`

turboproc, 2020-02-01:
@Koenkk 

Made a first comparison between V1.6.0 and V1.9.0. May be these snippets provide a hint already:

In v.1.6.0 the following happens:
````zigbee-herdsman:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":32172,"status":0,"nwkaddr":32172,"len":32,"endpoint":1,"profileid":49246,"deviceid":16,"deviceversion":2,"numinclusters":9,"inclusterlist":[0,4,3,6,8,5,2820,1794,10],"numoutclusters":3,"outclusterlist":[3,25,10]} +51ms
  zigbee-herdsman:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":32172,"nwkaddrofinterest":32172,"endpoint":2} +51ms
  zigbee-herdsman:unpi:writer --> frame [254,5,37,4,172,125,172,125,2,38] +52ms
  zigbee-herdsman:unpi:parser <-- [254,1,101,4,0,96] +16ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,4,0,96] +1ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 4 - [] - 96 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - simpleDescReq - {"status":0} +52ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:unpi:parser <-- [254,16,69,132,172,125,0,172,125,10,2,94,192,0,16,2,1,0,16,0,68] +27ms
  zigbee-herdsman:unpi:parser --- parseNext [254,16,69,132,172,125,0,172,125,10,2,94,192,0,16,2,1,0,16,0,68] +2ms
  zigbee-herdsman:unpi:parser --> parsed 16 - 2 - 5 - 132 - [�}�}
  zigbee-herdsman:unpi:parser ^�] - 68 +0ms
  zigbee-herdsman:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":32172,"status":0,"nwkaddr":32172,"len":10,"endpoint":2,"profileid":49246,"deviceid":4096,"deviceversion":2,"numinclusters":1,"inclusterlist":[4096],"numoutclusters":0,"outclusterlist":[]} +51ms
  zigbee-herdsman:unpi:parser --- parseNext [] +2ms
PuTTY  zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":32172,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":3,"options":48,"radius":30,"len":9,"data":{"type":"Buffer","data":[0,3,0,4,0,5,0,7,0]}} +78ms
  zigbee-herdsman:unpi:writer --> frame [254,19,36,1,172,125,1,1,0,0,3,48,30,9,0,3,0,4,0,5,0,7,0,198] +78ms
  zigbee-herdsman:unpi:parser <-- [254,1,100,1,0,100] +45ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100 +1ms
  zigbee-herdsman:znp:SRSP <-- AF - dataRequest - {"status":0} +79ms
  zigbee-herdsman:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:unpi:parser <-- [254,3,68,128,0,1,3,197] +18ms
  zigbee-herdsman:unpi:parser --- parseNext [254,3,68,128,0,1,3,197] +1ms
  zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 197 +1ms
  zigbee-herdsman:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3} +69ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:unpi:parser <-- [254,48,68,129,0,0,0,0,172,125,1,1,0,55,0,29,63,29,0,0,28,24,3,1,4,0,0,66,4,105,110,110,114,5,0,0,66,6,83,80,32,49,50,48,7,0,0,48,1,172,125,29,216] +18ms
  zigbee-herdsman:unpi:parser --- parseNext [254,48,68,129,0,0,0,0,172,125,1,1,0,55,0,29,63,29,0,0,28,24,3,1,4,0,0,66,4,105,110,110,114,5,0,0,66,6,83,80,32,49,50,48,7,0,0,48,1,172,125,29,216] +2ms
  zigbee-herdsman:unpi:parser --> parsed 48 - 2 - 4 - 129 - [�}7?BinnrBSP 1200�}] - 216 +0ms
  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":32172,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":1916701,"transseqnumber":0,"len":28,"data":{"type":"Buffer","data":[24,3,1,4,0,0,66,4,105,110,110,114,5,0,0,66,6,83,80,32,49,50,48,7,0,0,48,1]}} +23ms
````
Where as in v.1.9.0 the following occurs:
````
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":25003,"status":0,"nwkaddr":25003,"len":32,"endpoint":1,"profileid":49246,"deviceid":16,"deviceversion":2,"numinclusters":9,"inclusterlist":[0,4,3,6,8,5,2820,1794,10],"numoutclusters":3,"outclusterlist":[3,25,10]} +89ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '1' device '0x00158d000388f73e' +78ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":25003,"nwkaddrofinterest":25003,"endpoint":2} +102ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,171,97,171,97,2,38] +102ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96] +57ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0} +112ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,16,69,132,171,97,0,171,97,10,2,94,192,0,16,2,1,0,16,0,68] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,16,69,132,171,97,0,171,97,10,2,94,192,0,16,2,1,0,16,0,68] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 16 - 2 - 5 - 132 - [171,97,0,171,97,10,2,94,192,0,16,2,1,0,16,0] - 68 +2ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":25003,"status":0,"nwkaddr":25003,"len":10,"endpoint":2,"profileid":49246,"deviceid":4096,"deviceversion":2,"numinclusters":1,"inclusterlist":[4096],"numoutclusters":0,"outclusterlist":[]} +88ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '2' device '0x00158d000388f73e' +90ms
[... cut ...]
zigbee2mqtt:debug 2020-02-01 10:31:41: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SP 120","manufacturerName":"innr","powerSource":1}'
 ````

So both version find the same endpoints and within these endpoints they find the same clusters. In the next step where cluster 0 of endpoint 1 is being queried, the difference starts as v.1.6.0 queries for 
````
"data":{"type":"Buffer","data":[0,3,0,4,0,5,0,7,0]}
````

and v1.9.0 queries for 
````
"data":{"type":"Buffer","data":[16,14,0,5,0,4,0,7,0]}
````

''

turboproc, 2020-02-01:
@Koenkk , learning fast, when doing a deep dive it seems that V1.6.0 and v.1.9.0 have (by design some fundamental difference:
V.1.6.0: reads all endpoint and cluster within endpoints as part of the binding process
V.1.9.0: reads only on the genBasic endpoint and when done it considers Interviewing complete. As a next phase it starts the Configuration phase to get the other data. (If I'm correct).

What I notice regarding the Configuration phase, z2m sends a bindReq where it looks like srcaddr and dstaddr have been swapped (at least compared to all dataRequest that have done before). If this is not intended it would explain why z2m would never receive a response.

````
 zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":3259,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":3,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,4,0,3,0,6,0,0,64]}} +82ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,187,12,1,1,0,0,3,0,30,9,16,4,0,3,0,6,0,0,64,196] +82ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +83ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,3,197] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,3,197] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,3] - 197 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3} +58ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,53,68,129,0,0,0,0,187,12,1,1,0,21,0,212,78,103,0,0,33,24,4,1,3,0,0,32,1,6,0,0,66,12,50,48,49,55,49,48,50,55,45,49,48,48,0,64,0,66,3,50,46,48,30,99,28,169] +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,53,68,129,0,0,0,0,187,12,1,1,0,21,0,212,78,103,0,0,33,24,4,1,3,0,0,32,1,6,0,0,66,12,50,48,49,55,49,48,50,55,45,49,48,48,0,64,0,66,3,50,46,48,30,99,28,169] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 53 - 2 - 4 - 129 - [0,0,0,0,187,12,1,1,0,21,0,212,78,103,0,0,33,24,4,1,3,0,0,32,1,6,0,0,66,12,50,48,49,55,49,48,50,55,45,49,48,48,0,64,0,66,3,50,46,48,30,99,28] - 169 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":3259,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":6770388,"transseqnumber":0,"len":33,"data":{"type":"Buffer","data":[24,4,1,3,0,0,32,1,6,0,0,66,12,50,48,49,55,49,48,50,55,45,49,48,48,0,64,0,66,3,50,46,48]}} +34ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":3,"status":0,"dataType":32,"attrData":1},{"attrId":6,"status":0,"dataType":66,"attrData":"20171027-100"},{"attrId":16384,"status":0,"dataType":66,"attrData":"2.0"}],"Cluster":{"ID":0,"attributes":{"zclVersion":{"ID":0,"type":32,"name":"zclVersion"},"appVersion":{"ID":1,"type":32,"name":"appVersion"},"stackVersion":{"ID":2,"type":32,"name":"stackVersion"},"hwVersion":{"ID":3,"type":32,"name":"hwVersion"},"manufacturerName":{"ID":4,"type":66,"name":"manufacturerName"},"modelId":{"ID":5,"type":66,"name":"modelId"},"dateCode":{"ID":6,"type":66,"name":"dateCode"},"powerSource":{"ID":7,"type":48,"name":"powerSource"},"appProfileVersion":{"ID":8,"type":48,"name":"appProfileVersion"},"swBuildId":{"ID":16384,"type":66,"name":"swBuildId"},"locationDesc":{"ID":16,"type":66,"name":"locationDesc"},"physicalEnv":{"ID":17,"type":48,"name":"physicalEnv"},"deviceEnabled":{"ID":18,"type":16,"name":"deviceEnabled"},"alarmMask":{"ID":19,"type":24,"name":"alarmMask"},"disableLocalConfig":{"ID":20,"type":24,"name":"disableLocalConfig"}},"name":"genBasic","commands":{"resetFactDefault":{"ID":0,"parameters":[],"name":"resetFactDefault"}},"commandsResponse":{}}},"address":3259,"endpoint":1,"linkquality":21,"groupID":0}' +91ms
zigbee2mqtt:debug 2020-02-01 10:31:12: Received Zigbee message from '0x00158d000388f73e', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-01 10:31:12: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20171027-100","swBuildId":"2.0"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +11ms
  zigbee-herdsman:controller:device Interview - got 'hwVersion,dateCode,swBuildId' for device '0x00158d000388f73e' +89ms
  zigbee-herdsman:controller:device Interview - completed for device '0x00158d000388f73e' +7ms
  zigbee-herdsman:controller:log Succesfully interviewed '0x00158d000388f73e' +23ms
zigbee2mqtt:info  2020-02-01 10:31:12: Successfully interviewed '0x00158d000388f73e', device has successfully been paired
zigbee2mqtt:info  2020-02-01 10:31:12: Device '0x00158d000388f73e' is supported, identified as: Innr Smart plug (SP 120)
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,187,12,62,247,136,3,0,141,21,0,1,6,0,3,171,100,54,25,0,75,18,0,1,194] +106ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69] +42ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0} +105ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
````

Koenkk, 2020-02-01:
@turboproc there is a very big difference between 1.6.0 and 1.7.0 + (zigbee-shepherd, partially herdsman) vs full zigbee-herdsman, basically the whole underlying zigbee library has been changed). 

- Does the power reporting/configure work correctly on zigbee2mqtt 1.6.0?
- If yes, could you create a sniff log for both 1.6.0 and 1.9.0 (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html). These sniffer logs shows exactly what is happening and we can easily compare them.

> What I notice regarding the Configuration phase, z2m sends a bindReq where it looks like srcaddr and dstaddr have been swapped (at least compared to all dataRequest that have done before).

I don't think this is the issue, `srcaddr` should be the address you want to bind (= SP 120) and `dstaddress` the address you want to bind it to (= coordinator). If this would be wrong, binding shouldn't work at all in Zigbee2mqtt (for no device) but this doesn't seem to be the case.

dersparkiie, 2020-02-01:
Same problem here since updating from 1.5.1 to 1.9.0. Repairing doesn't solve it!

turboproc, 2020-02-01:
@Koenkk 

- reporting and configure under 1.6.0 works OK, getting values that make sense
- sniffing is ofcourse an option, just need to order an addition CC2531 for this (having 1 now on 1.6.0 and one on 1.9.0)
- what I learned in the meantime is that when deleting the "await bind ...." line in devices.js int the block that defines the SP-120, the configuration process gets completed but values are still not being reported. So deletion is not the option, but change of that line is still required.

turboproc, 2020-02-02:
@Koenkk , while waiting for additional C2531, I explored a bit futher the traces and what I see now is that V1.6.0 does not use reporting, in fact it does a Discover of attributes and then reads the attributes.
So changed now in V1.9.0 the devices.js file: removed the bind line and the configure report lines (all). When configuring the SP-120, it comes to life, reporting status, activepower, rmsvoltage and rms current. So that's more now.
When requesting a new status via MQTT, z2m reaches out to the SP-120 and provides an updated status. However, when requesting an new activepower it throws an error "no converter available". Do understand yet where to look for creating of the convertor.

Koenkk, 2020-02-02:
@turboproc zigbee2mqtt 1.6.0 also bound it (https://github.com/Koenkk/zigbee-herdsman-converters/blob/zigbee-shepherd-converters/devices.js#L2006) and the power value was definitely not read manually. 

The strange thing: I also have this device and it configures fine with the latest zigbee2mqtt dev branch. I don't understand why it doesn't work on some cases

FutureCow, 2020-02-02:
I did some testing 2 months ago and if i remember correctly this was the outcome:

z2m 1.6.0 fresh install --> Pairing SP-120 without a problem
z2m 1.8.0 fresh install --> Pairing SP-120 without a problem
z2m 1.6.0 fresh install --> Paring SP-120 --> Update to 1.8.0 --> without a problem
z2m update from 1.6.0 --> 1.8.0 --> Pairing SP-120, Not working correctly!

So if you already used the SP-120 in a older version of z2m, you don't have any problems. But if you update from 1.6.0 to a newer version and pair the SP-120 after it, it doesn't work correctly. 
Maybe turboproc can verify this?

turboproc, 2020-02-02:
@Koenkk , @FutureCow , done some more work on a rainy Sunday. Suddenly understood that for requesting values on demand from the SP 120 one needs a toZigbee converter t get the commands out. So quickly added a tz.SP-120 convertor that can request power, current, voltage. That works. (Not really exciting but hey, first time everything is a bit exciting.)

So the key question that remains is why reporting cannot be configured.  Sunday isn;t over yet...

````
mosquitto_pub -t "zigbee2mqtt/0x00158d000388f73e/get" -m "{\"power\":\"\"}" -d
````
results in 
````
 Received Zigbee message from '0x00158d000388f73e', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":24}' from endpoint 1 with groupID 0
````
Same for the other attributes.

[5 more replies]

closed 2020-02-02 by Koenkk