Koenkk/zigbee2mqtt #2366: TRADFRI bulb E27 WW 806lm' is not supported. Version

wes1993, 2019-11-18:
# Bug Report
## What happened

TRADFRI bulb E27 WW 806lm' is not supported.
 From what version of Zigbee2MQTT is supported?
Also the Ikea E1743 and E1603
Zigbee2MQTT map won’t update Also, only white screen.
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Log of the Map:

2019-11-17 18:22:27 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zigbee2mqtt_networkmap, service=update, service_data=> 2019-11-17 18:22:27 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=mqtt, service=publish, service_data=topic=your topic (optional, default zigbee2mqtt)/bridge/networkmap, payload=graphviz>

zigbee2mqtt version: 1.6.0
CC253X firmware version:

Koenkk, 2019-11-19:
It is supported since zigbee2mqtt 1.7.0

closed 2019-11-19 by Koenkk

wes1993, 2019-11-19:
Thanks a lot!!
Also for your beautiful work!!
Stefano