Koenkk/zigbee2mqtt #2367: zigbee2mqtt:error 2019-11-19T11:13:36: Failed to interview '0xd0cf5efffef919e6', device has not successfully been paired

sander816, 2019-11-19:
# Bug Report
## What happened

Trying to re pair 2 muller lights: https://www.zigbee2mqtt.io/devices/MLI-404011.html

## How to reproduce it (minimal and precise)
upgrade to latest zigbee2mqtt and try to add muller light
## Debug Info
zigbee2mqtt version: 1.7.1 (commit #b459c35)
CC253X firmware version: 20181119 (i have no debugger to upgrade)

https://pastebin.com/qcKAi6nn

sander816, 2019-11-19:
same for xiaomi smoke sensor:
zigbee2mqtt:error 2019-11-19T12:28:48: Failed to interview '0x00158d0002ad12a0', device has not successfully been paired
zigbee2mqtt:info  2019-11-19T12:28:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0002ad12a0"}}'

Koenkk, 2019-11-19:
Can you post the log when running with `DEBUG=* npm start`?

CMDR-Sloma, 2019-11-20:
I have the same problem with pairing devices:

```
zigbee2mqtt:info  2019-11-20T13:00:01: Device '0x90fd9ffffed54a16' joined
zigbee2mqtt:info  2019-11-20T13:00:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x90fd9ffffed54a16"}}'
zigbee2mqtt:info  2019-11-20T13:00:01: Starting interview of '0x90fd9ffffed54a16'
zigbee2mqtt:info  2019-11-20T13:00:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x90fd9ffffed54a16"}}'
zigbee2mqtt:debug 2019-11-20T13:00:07: Device '0x90fd9ffffed54a16' announced itself
zigbee2mqtt:error 2019-11-20T13:00:21: Failed to interview '0x90fd9ffffed54a16', device has not successfully been paired
zigbee2mqtt:info  2019-11-20T13:00:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x90fd9ffffed54a16"}}'
```

I'm on hassio with zigbee2mqtt addon v.1.7.1 and latest firmware on cooridnator.

sander816, 2019-11-20:
smoke sensors are working after some tries
this is for the aqara cube
https://pastebin.com/0tHNY0QN

sander816, 2019-11-21:
i checked oiut with dev, now the cube is recognized.
The muller light is still not working
https://pastebin.com/rnLzZJHV

Koenkk, 2019-11-21:
Please provide the log with `DEBUG=* npm start`, otherwise there is not enough information to determine the issue.

sander816, 2019-11-21:
I am running it on raspbian
I edited configuration.yaml
With loglevel=debug

leo8484, 2019-11-22:
> OIUT с Dev

And in more detail you can

leo8484, 2019-11-22:
> i checked oiut with dev, now the cube is recognized.
> The muller light is still not working
> https://pastebin.com/rnLzZJHV

And in more detail you can?

leo8484, 2019-11-22:
> i checked oiut with dev, now the cube is recognized.
 And in more detail you can?

sander816, 2019-11-22:
tried again:
permit_join: true
advanced:
  log_level: debug

https://pastebin.com/Ph3jaVEz

looks like its getting detected correctly according to the log:
`{"friendly_name":"0xd0cf5efffef919e6","model":"404000/404005/404012","vendor":"Müller Licht","description":"Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white","supported":true}}'
`
but in database i see this:
`{"id":18,"type":"Router","ieeeAddr":"0xd0cf5efffef919e6","nwkAddr":35548,"manufId":4635,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"ZBT-ExtendedColor","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096,65244],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-ExtendedColor","manufacturerName":"MLI","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181119-71","swBuildId":"4.4"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181119-71","swBuildId":"4.4","zclVersion":2,"interviewCompleted":true,"meta":{}}`

domoticz doesn not recognize it either as muller light:
`2019-11-22 20:47:07.298 (zigbeenetwork) Device 404000/404005/404012 0xd0cf5efffef919e6 (0xd0cf5efffef919e6)
`
And the controls are not working

Koenkk, 2019-11-22:
I think the device needs to be added here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin, from the zigbee2mqtt side things are working properly now. Please create an issue there, it's maintained by someone else.

sander816, 2019-11-22:
i hade it working before, with v1.5.1. or is the domoticz plugin not ok anymore, since i upgraded to 1.7.1?

Koenkk, 2019-11-22:
I don't have knowledge about the domitcz plugin. But if you use an MQTT client and send to topic: `zigbee2mqtt/0xd0cf5efffef919e6/set` payload `toggle` and it works it means that the zigbee2mqtt part is OK.

sander816, 2019-11-23:
Am I doing something wro g then?
On my aqara hub I have zero problems getting all my Sensors recognized. (Except Müller Lights)
I'm my cc2531 it takes several attempts, or some never get recognized. Device has been not successfully paired.
Is it maybe because of the older Firmware?

I delete them from the hub. Press the button on the mend tion sensor for 5-6 seconds....I assume that's correct? Some get paired directly, some not

Koenkk, 2019-11-25:
I'm not sure wether this firmware can cause these problems (too long time ago). Is your CC2531 connected through an USB extension cable to avoid interference issues?

sander816, 2019-11-25:
no, its connected to a hub, the hub is connected with a 20 cm cable.
Is it better to add an usb extension cable to the cc2531?

Koenkk, 2019-11-26:
yes

UJStudio, 2019-12-14:
I got the same situation when try to pair gas  and water leak sensor.
OS: Raspbian Buster
node: v10.17.0
npm: v6.11.3
zigbee2mqtt: v1.7.1 
Debug info: see the attached file

[0x00124b001c48f997_pairing_error_debug_info.log](https://github.com/Koenkk/zigbee2mqtt/files/3963396/0x00124b001c48f997_pairing_error_debug_info.log)
[0x00124b001c48f997_pairing_error_info.log](https://github.com/Koenkk/zigbee2mqtt/files/3963397/0x00124b001c48f997_pairing_error_info.log)

sander816, 2019-12-14:
i think it had to do with other qifi signals distorting the pairing process
I followed the rules, set all different channels for my 2 wifi networks, and the zigbee network.
Since that i have no problems pairing anymore
And i got an other cc2531 with antenna, and flashed the latest firmware

UJStudio, 2019-12-17:
good news i did as per your instruction, adjust 2.4Ghz wifi channel from 11 to 6, it works , paired the two devices.

dzungpv, 2020-01-03:
I have the same error, never happen with version prior 1.6

KevinRiordan, 2020-01-06:
I was able to pair the HEIMAN smoke sensor when running HASS.IO on a RPi but now when switching to the new version running on linux+dockers I am getting the same error. The device does report but its not appearing as a device or entity and i am not seeing anything inth mqtt logs.

`zigbee2mqtt:warn  2020-01-06 14:29:53: Device '0x000d6f00120b28fd' left the network
zigbee2mqtt:info  2020-01-06 14:29:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000d6f00120b28fd"}}'
zigbee2mqtt:info  2020-01-06 14:30:39: Device '0x000d6f00120b28fd' joined
zigbee2mqtt:info  2020-01-06 14:30:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f00120b28fd"}}'
zigbee2mqtt:info  2020-01-06 14:30:39: Starting interview of '0x000d6f00120b28fd'
zigbee2mqtt:info  2020-01-06 14:30:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f00120b28fd"}}'
zigbee2mqtt:error 2020-01-06 14:31:12: Failed to interview '0x000d6f00120b28fd', device has not successfully been paired
zigbee2mqtt:info  2020-01-06 14:31:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000d6f00120b28fd"}}'
zigbee2mqtt:info  2020-01-06 14:35:20: MQTT publish: topic 'zigbee2mqtt/0x000d6f00120b28fd', payload '{"water_leak":true,"tamper":true,"battery_low":false,"linkquality":110}'
zigbee2mqtt:info  2020-01-06 14:35:20: MQTT publish: topic 'zigbee2mqtt/0x000d6f00120b28fd', payload '{"water_leak":false,"tamper":true,"battery_low":false,"linkquality":110}'`

Koenkk, 2020-01-08:
In order to help with finding the root cause of this issue, please:
- Provide the device you are trying to pair
- The herdsman debug logging when pairing the device

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

KevinRiordan, 2020-01-08:
Reimaged PC, reinstalled dockers, and HASS and reflashed the cc25XX stick and repeated the process with no issues.

dzungpv, 2020-01-11:
@Koenkk  This is my log:

0x00124b000dxxx551 is DNCAT002, is an support devices after that it show failed to interview
```
2020-01-11T12:55:11.221Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,126,16,1,1,0,57,0,255,4,112,0,0,7,24,33,10,0,0,16,0,126,16,29,83]
2020-01-11T12:55:11.227Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,126,16,1,1,0,57,0,255,4,112,0,0,7,24,33,10,0,0,16,0,126,16,29,83]
2020-01-11T12:55:11.233Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,126,16,1,1,0,57,0,255,4,112,0,0,7,24,33,10,0,0,16,0,126,16,29] - 83
2020-01-11T12:55:11.234Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":4222,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":57,"securityuse":0,"timestamp":7341311,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,33,10,0,0,16,0]}}
2020-01-11T12:55:11.255Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":33,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":"genOnOff","commands":{"off":{"ID":0,"parameters":[],"name":"off"},"on":{"ID":1,"parameters":[],"name":"on"},"toggle":{"ID":2,"parameters":[],"name":"toggle"},"offWithEffect":{"ID":64,"parameters":[{"name":"effectid","type":32},{"name":"effectvariant","type":32}],"name":"offWithEffect"},"onWithRecallGlobalScene":{"ID":65,"parameters":[],"name":"onWithRecallGlobalScene"},"onWithTimedOff":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"commandsResponse":{}}},"address":4222,"endpoint":1,"linkquality":57,"groupID":0}'
2020-01-11T12:55:11.262Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '4222', skipping...
2020-01-11T12:55:11.270Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-11T12:55:17.837Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,197,89,1,1,0,107,0,193,85,112,0,0,7,24,11,10,0,0,16,0,135,229,28,187]
2020-01-11T12:55:17.837Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,197,89,1,1,0,107,0,193,85,112,0,0,7,24,11,10,0,0,16,0,135,229,28,187]
2020-01-11T12:55:17.838Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,197,89,1,1,0,107,0,193,85,112,0,0,7,24,11,10,0,0,16,0,135,229,28] - 187
2020-01-11T12:55:17.839Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":22981,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":107,"securityuse":0,"timestamp":7361985,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,11,10,0,0,16,0]}}
2020-01-11T12:55:17.842Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":11,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":"genOnOff","commands":{"off":{"ID":0,"parameters":[],"name":"off"},"on":{"ID":1,"parameters":[],"name":"on"},"toggle":{"ID":2,"parameters":[],"name":"toggle"},"offWithEffect":{"ID":64,"parameters":[{"name":"effectid","type":32},{"name":"effectvariant","type":32}],"name":"offWithEffect"},"onWithRecallGlobalScene":{"ID":65,"parameters":[],"name":"onWithRecallGlobalScene"},"onWithTimedOff":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"commandsResponse":{}}},"address":22981,"endpoint":1,"linkquality":107,"groupID":0}'
2020-01-11T12:55:17.842Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '22981', skipping...
2020-01-11T12:55:17.843Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-11T12:55:32.842Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,33,4,1,1,0,63,0,236,12,113,0,0,7,24,8,10,0,0,16,1,33,4,29,103]
2020-01-11T12:55:32.842Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,33,4,1,1,0,63,0,236,12,113,0,0,7,24,8,10,0,0,16,1,33,4,29,103]
2020-01-11T12:55:32.843Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,33,4,1,1,0,63,0,236,12,113,0,0,7,24,8,10,0,0,16,1,33,4,29] - 103
2020-01-11T12:55:32.844Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":1057,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":7408876,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,8,10,0,0,16,1]}}
[... cut ...] ,"commandsResponse":{}}},"address":1057,"endpoint":1,"linkquality":63,"groupID":0}'
zigbee2mqtt:debug 2020-01-11 19:55:32: Received Zigbee message from '0x00124b000dxxx551', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-11 19:55:32: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-01-11 19:55:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Koenkk, 2020-01-13:
@dzungpv can you share the complete log (post it on pastebin.com and link it here).

dzungpv, 2020-01-17:
@Koenkk I think it will be fix with 1.9.0 https://github.com/Koenkk/zigbee2mqtt/issues/2499 because it not read the Zigbee model, i will test and report back

talondnb, 2020-02-16:
I'm having trouble re-pairing a TRADFRI bulb, getting the same error:

info  2020-02-16 18:46:53: Switching log level to 'debug'
info  2020-02-16 18:46:53: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.10.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'
debug 2020-02-16 18:46:58: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
info  2020-02-16 18:46:58: Zigbee: allowing new devices to join.
info  2020-02-16 18:46:58: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.10.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
info  2020-02-16 18:47:32: Starting interview of '**0xccccccfffea91b17**'
info  2020-02-16 18:47:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"**0xccccccfffea91b17**"}}'
debug 2020-02-16 18:47:50: Saving state to file /share/zigbee2mqtt/state.json
debug 2020-02-16 18:48:29: Received Zigbee message from '0x00124b001c299b42', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-02-16 18:48:29: MQTT publish: topic 'zigbee2mqtt/0x00124b001c299b42', payload '{"state":true,"linkquality":33,"led_state":false}'
debug 2020-02-16 18:48:40: Received Zigbee message from '0x00158d0002e9e092', type 'attributeReport', cluster 'genBasic', data '{"65282":[{"elmType":16,"elmVal":0},{"elmType":33,"elmVal":3045},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":34},{"elmType":32,"elmVal":87}]}' from endpoint 1 with groupID 0
info  2020-02-16 18:48:40: MQTT publish: topic 'zigbee2mqtt/0x00158d0002e9e092', payload '{"contact":true,"linkquality":36,"battery":100,"voltage":3045}'
error 2020-02-16 18:48:52: **Failed to interview '0xccccccfffea91b17', device has not successfully been paired**
**info  2020-02-16 18:48:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xccccccfffea91b17"}}'**
debug 2020-02-16 18:49:16: Received Zigbee message from '0x00158d00022809d8', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-02-16 18:49:16: MQTT publish: topic 'zigbee2mqtt/0x00158d00022809d8', payload '{"battery":100,"voltage":3015,"linkquality":96,"occupancy":true}'
info  2020-02-16 18:49:21: MQTT publish: topic 'zigbee2mqtt/0x00158d00022809d8', payload '{"battery":100,"voltage":3015,"linkquality":96,"occupancy":false}'
debug 2020-02-16 18:49:29: Received Zigbee message from '0x00124b001c299b42', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-02-16 18:49:29: MQTT publish: topic 'zigbee2mqtt/0x00124b001c299b42', payload '{"state":true,"linkquality":36,"led_state":false}'

Any advice?

closed 2020-04-25 by stale[bot]

tinxx, 2021-01-11:
Any updates on the matter?
I came by a Müller 404002 and always get a `interview_failed` :(