Koenkk/zigbee2mqtt #2368: Xiaomi JTYJ-GD-01LM/BW selftest not working via router

KennieNL, 2019-11-19:
# Bug Report
## What happened
I'm trying to send a {"selftest":""} to the set topic for the smoke dectector. Unfortunate it does not respond to this and does not execute the selftest, same counts for setting sensitivity:

zigbee2mqtt:error 2019-11-19T14:46:25: Publish 'set' 'selftest' to '0x00158d000411f845' failed: 'Error: Timeout - 39617 - 1 - 2 - 1280 - 4 after 35000ms'

The smoke detector is sending information to a IKEA bulb that is also a router.
On the spot its located there is a very small to no link directly to the coordinator (1-4 to none).

I tried executed the selftest both when having the small link, and no link on the network map to the coordinator... in both situations it wont execute it.

When moving close to the coordinator, it does execute the selftest.

Could it be that it does not supported to be routed through the IKEA bulb?

## What did you expect to happen
Selftest to the routed via the router/IKEA bulb

## How to reproduce it (minimal and precise)
Move the smoke detector far away from the coordinator to be out of reach, but yet to be within reach of a router. Execute selftest

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190223

Koenkk, 2019-11-19:
I think the bulb doesn't hold the message properly for the smoke detector (it has to do this for max 30 seconds because this device only wakes up every 30 seconds). 

From the zigbee2mqtt side we cannot influence this behaviour. One thing you could do is adding a CC2531 router to your network and make sure that the traffic of the smoke sensor is routed through there.

kentux, 2019-12-11:
Hi @Koenkk !

I think I have the same problem, but I have a question, how can I force a device to use the CC2531 router ?

Koenkk, 2019-12-12:
Pair it while having it close to the CC2531 should do it.

closed 2020-02-10 by Koenkk