Koenkk/zigbee2mqtt #2369: Freeze 24h after last boot

Rykian, 2019-11-19:
# Bug Report
## What happened

Since 1.7.0, every 24h my Raspberry Pi 4 under Manjaro with zigbee2mqtt seems to be frozen (but there is logs in zigbee2mqtt, I just can't connect on SSH/HTTPS)

The logs shows journalctl detect a reboot, but zigbee2mqtt seems to continue to run (every logs here at the time of reboot : https://hastebin.com/amodihovaz.md, and I'm GMT+1 fyi)

It happened before when I was on the docker branch `latest-dev` and I rollbacked it to latest to avoid this bug, but it's now the `latest` :)

I already reinstall everything from scratch and it freeze anyway.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1 (docker)
CC253X firmware version: 20190608

Kryzek, 2019-11-19:
What kind of power supply you have? I’ve seen weird actions with Pi using a bit under spec power supply. Z2M should not cause whole system to die.

Rykian, 2019-11-19:
I'm using a 3A/5V power supply. The Pi is used sometimes for more power hungry tasks like video decoding under, so I don't think it comes from that.

Koenkk, 2019-11-20:
I've checked the logs you provided but they don't show a crash or communication error with the CC2531.

Rykian, 2019-11-23:
For now, I've set up a cron restarting my zigbee2mqtt every 23 hours to avoid this freeze and it seems to work. But it's a messy solution... I will try to investigate more next week.

closed 2020-01-30 by stale[bot]