Koenkk/zigbee2mqtt #2371: Gledopto GL-C-008 connectivity issues

Kryzek, 2019-11-20:
I was wondering why I could not get my nice moody lighting on and I noticed that my Gledopto GL-C-008 was offline. I'm using 1.7.1+dev #e33b49e. This light has been working since my last repairing session (after https://github.com/Koenkk/zigbee2mqtt/issues/2338)

I tried to unplugging and re-plugging it to wall and this happened:
```
debug 15:26:38: Device '0x00124b001d3ab1b9' announced itself
info  15:26:38: MQTT publish: topic 'zigbee/0x00124b001d3ab1b9/availability', payload 'online'
error 15:26:48: Failed to read state of '0x00124b001d3ab1b9' after reconnect
###### This fails to control light
debug 15:26:51: Received MQTT message on 'zigbee/0x00124b001d3ab1b9/set' with data '{"state": "ON"}'
debug 15:26:51: Publishing 'set' 'state' to '0x00124b001d3ab1b9'
###### /fail
error 15:28:48: Failed to ping '0x00124b001d3ab1b9'
```

I tried re-powering multiple times but the behavior is the same. Device announces itself to network, does not respond to commands and finally gets marked as offline after a ping checks for its availability.

Is this something to investigate more for possible explanation for future issues or do I just go ahead and try to re-pair and see if the issue is solved with that?

Kryzek, 2019-11-20:
After restarting Zigbee2MQTT device stays on network again.

closed 2019-11-20 by Kryzek

ricreis394, 2020-01-30:
I'm having the same issue but with a Philips 9290011370.
Unpluging and pluging from wall plug doesn't make it appear as available, but appears a message:
`zigbee2mqtt:error 2020-01-30 22:10:30: Failed to read state of 'philips_hue_sala' after reconnect`
Any thoughts?