Koenkk/zigbee2mqtt #2373: Improve log management

SeLLeRoNe, 2019-11-20:
# Feature Request
## What happened
On every restart the log file have a different name and makes it hard to track the "current" log compared to the old ones.

## What did you expect to happen
It would be much better being able to:
- Introducing log rotation with the ability to configure how many days of logs to keep
- Have the "log.txt" default file always be related to "today" and have log-YYYYMMDD.txt for passed days or, have the log.txt as a symlink to the "current" log.

## Debug Info
zigbee2mqtt version: 1.7.1+dev
CC253X firmware version: 2019042

Koenkk, 2019-11-20:
Currently there is already an option to change the `log_directory` (https://www.zigbee2mqtt.io/information/configuration.html). But something more advanced is indeed not in place yet.

SeLLeRoNe, 2019-11-21:
I did change the log directory and I am logging in logs/ so to have the logs only there, but still that doesn't change the confusion generated by the logs name structure :)

KorbinianP, 2020-01-02:
This would be a very cool enhancement, I would like to use [frontail](https://github.com/mthenw/frontail) to see quickly whats going on without an ssh. But as the log file name or folder always changes, this is highly complex. For me a symlink to the latest log file would be enough, all other behaviour can stay the same.

closed 2020-03-10 by stale[bot]