Koenkk/zigbee2mqtt #2374: Xiaomi Aqara single wirelles switch (WXKG03LM): Trigger actions "click" and "action" doesn't work

uxdesignerhector, 2019-11-20:
# Bug Report
## What happened
"Actions" and "Clicks" triggers for Aqara single wirelles switch (WXKG03LM) not enabled in automations UI in homeassistant (Hassio) while "Link quality" and "Battery level" appears. I have tried writing the automation on my own but that throws me the next log error in HA. So that also seems to be broken.
**_Invalid config for [automation]: value is not allowed for dictionary value @ data['type']. Got None._**

I'm new to zigbee2mqtt so I can't tell you is this was working on previous versions.  

**Models being affected**: 
- Aqara single key wireless wall switch (**WXKG03LM**)

## What did you expect to happen
Action and Click triggers enabled for Aqara switches at the automation tool inside Home assistant. Correct behaviour if I write the automation directly at the automation.yaml file

## How to reproduce it
Go to automations in Home Assistant, create a new one, go to trigger section, select device, select an Aqara single key wireless wall switch (WXKG03LM) click at next label trigger. It will only shows "Battery Level" and "Link quality" triggers.
- **Screen record showing the problem**: https://gfycat.com/allspitefulbumblebee

## Debug Info
- **My manual automation**: https://pastebin.com/zsYW15h8
- **Zigbee2mqtt Log** (there are not warms or previous errors) : https://pastebin.com/ufP9HqdQ
- **zigbee2mqtt version**:  1.7.1 (zigbee-herdsman)
- **CC253X firmware version**: Z-Stack_Home_1.2 (default)

Note: Other Aqara wired switches are working.  I'm a seeing double payloads at zigbee2mqtt log for (WXKG03LM) like in the log provided here, is that normal?

Koenkk, 2019-11-20:
- It seems to be available under the 'State':
[image]

- Double payloads are normal, this is needed to trigger a state change in home assisstant

uxdesignerhector, 2019-11-21:
Thank you so much for the info, now it's working.

closed 2019-11-21 by uxdesignerhector

alekse-kuznecow, 2021-03-09:
It looks like that recommended solution does not work.
I have tried to implement it but all my possibilities failed.
I propose to provide the more detailed recommendations to the community