Koenkk/zigbee2mqtt #2375: Ikea Fyrtur blind doesn't pair (repeater works)

tanuva, 2019-11-20:
# Bug Report
## What happened
1. Successfully paired the Ikea Tradfri repeater (E1746) with a CC2531 stick, shows up in HASS. This is a fresh zigbee2mqtt and HASS installation. (I'm a newbie.)
1. Pressed both buttons on a Ikea Fyrtur blind so that the white light starts to "breathe" for about 2 minutes.
1. Made the paired repeater touch the blind's case in various orientations around the chain-link "pair here" sticker.
1. As much as it hurts to say it: "nothing happens". There doesn't seem to be any pairing activity going on.

I added `zigbee_herdsman_debug: true` to `configuration.yaml`. No additional log lines after startup appear when I initiate the pairing process. `permit_join: true` is set.

Here's the full [zigbee2mqtt log](https://pastebin.com/Mny3vSna).

## What did you expect to happen
I expected the blind to join the Zigbee network successfully.

## How to reproduce it (minimal and precise)
*see "What happened"*

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: Stick vendor says "June 2019", so 20190608.

Is there something in the pairing process that I missed? The Tradfri stuff seems to be rather finicky when used with hardware from outside the Tradfri world.

Koenkk, 2019-11-20:
Can you try holding the blind close to the coordinator when pairing it? Perhaps it's a range issue. If that doesn't help you could try to ask for support here: #829 (many people seem to have paired it successfully with zigbee2mqtt).

tanuva, 2019-11-21:
> If that doesn't help you could try to ask for support here: #829
I guess you meant to link #892. :) Will ask there, thanks.
> (many people seem to have paired it successfully with zigbee2mqtt).
That's the part that confuses me. It can't be that hard after all, can it? (Now that you've done the work in z2m, that is.)

closed 2019-11-21 by tanuva