Koenkk/zigbee2mqtt #2376: IKEA Tradfri doesn't pair (interview failed)

frauhottelmann, 2019-11-20:
# Bug Report
## Tradfri bulb doestn't pair because Interview failed.
`zigbee2mqtt:info  2019-11-20T16:20:45: Device '0x90fd9ffffefae31a' joined
zigbee2mqtt:info  2019-11-20T16:20:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x90fd9ffffefae31a"}}'
zigbee2mqtt:info  2019-11-20T16:20:45: Starting interview of '0x90fd9ffffefae31a'
zigbee2mqtt:info  2019-11-20T16:20:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x90fd9ffffefae31a"}}'
zigbee2mqtt:error 2019-11-20T16:21:05: Failed to interview '0x90fd9ffffefae31a', device has not successfully been paired
zigbee2mqtt:info  2019-11-20T16:21:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x90fd9ffffefae31a"}}'`

## What did you expect to happen
Interview beeing successful and bulb ist paired.

## How to reproduce it (minimal and precise)
Reseting the bulb while   "permit_join": true. I restartet the addon as well. The reset of the bulb is successful because it blinks.

## Debug Info
zigbee2mqtt version: 1.7.1
CC2531 firmware version:
`{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}}`

Log:
https://gist.github.com/frauhottelmann/a49d83d75b66f2d63137d4ec70368299

Koenkk, 2019-11-20:
Please check https://github.com/Koenkk/zigbee2mqtt/issues/2355#issuecomment-555673674

CMDR-Sloma, 2019-11-20:
I have moved my post from other similar thread here because it of exactly the same issue.

I have the same problem with pairing devices (Ikea bulb LED1545G12 below):

```
zigbee2mqtt:info  2019-11-20T13:00:01: Device '0x90fd9ffffed54a16' joined
zigbee2mqtt:info  2019-11-20T13:00:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x90fd9ffffed54a16"}}'
zigbee2mqtt:info  2019-11-20T13:00:01: Starting interview of '0x90fd9ffffed54a16'
zigbee2mqtt:info  2019-11-20T13:00:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x90fd9ffffed54a16"}}'
zigbee2mqtt:debug 2019-11-20T13:00:07: Device '0x90fd9ffffed54a16' announced itself
zigbee2mqtt:error 2019-11-20T13:00:21: Failed to interview '0x90fd9ffffed54a16', device has not successfully been paired
zigbee2mqtt:info  2019-11-20T13:00:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x90fd9ffffed54a16"}}'
```

I'm on hassio with zigbee2mqtt add-on v.1.7.1 and latest firmware on coordinator. Sometimes it works, sometimes I have to restart pairing by switching on-off the bulb.

UPDATE:

Overall, pairing after upgrade from 1.6 to 1.7.1 is total pain because of "Failed to interview" errors. Some of my Ikea bulbs pair on first try, other times it is after second or third but some I simply cannot pair because of this. I had no issues before, bulbs were always pairing on first try. I was previously on default firmware and had no issues but because I have more than 40 devices I wanted to try alternative firmware and pairing with this was much worse. I gave up after fourth device which I tried to pair probably 20 times with no luck. Now, I went back to default firmware and it's harder and harder after I paired 16 devices... Those are bulbs only....

> 
> 
> Please check [#2355 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2355#issuecomment-555673674)

I think it started after migration to zigbee-herdsman. Because of pairing problems I have downgraded my zigbee2mqtt to 1.6 and I don't have any issues pairing Ikea devices. Everything paired at first try. I have 9 Ikea GU10 Spotlights, 3 "candle" light LEDs, 5 Ikea sockets and 5 other Ikea bulbs. All gave me pairing problems under zigbee-herdsman but not under zigbee-shepherd.

frauhottelmann, 2019-11-20:
> Please check [#2355 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2355#issuecomment-555673674)

That did work. Maybe it is a firmware issue with the bulbs.

@CMDR-Sloma I didn't have any issues with other devices.

closed 2019-11-20 by frauhottelmann

Koenkk, 2019-11-20:
@CMDR-Sloma I'm not sure wether zigbee-herdsman has regression regarding pairing or if that zigbee-shepherd does not log errors at all.

CMDR-Sloma, 2019-11-20:
I'm sorry for re-opening but I want to be sure:

@frauhottelmann Do you mean other Ikea devices or simply other devices?

@Koenkk I noticed under zigbee-shepherd it eventually pairs after several seconds. I don't know if it helps here is my log for successful pairing of another IKEA device:

```
11/20/2019, 4:10:44 PM - error: Cannot get the Node Descriptor of the Device: 0x90fd9ffffef4cb72 (Error: Timed out after 10000 ms)
11/20/2019, 4:10:54 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb E27 WS opal 980lm","powerSource":1}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:54 PM - info: New device 'TRADFRI bulb E27 WS opal 980lm' with address 0x90fd9ffffef4cb72 connected!
11/20/2019, 4:10:54 PM - debug: Saving state to file /share/zigbee2mqtt/state.json
11/20/2019, 4:10:54 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x90fd9ffffef4cb72","meta":{"modelID":"TRADFRI bulb E27 WS opal 980lm"}}'
11/20/2019, 4:10:54 PM - info: MQTT publish: topic 'homeassistant/light/0x90fd9ffffef4cb72/light/config', payload '{"brightness":true,"color_temp":true,"schema":"json","command_topic":"zigbee2mqtt/0x90fd9ffffef4cb72/set","state_topic":"zigbee2mqtt/0x90fd9ffffef4cb72","name":"0x90fd9ffffef4cb72_light","unique_id":"0x90fd9ffffef4cb72_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x90fd9ffffef4cb72"],"name":"0x90fd9ffffef4cb72","sw_version":"Zigbee2mqtt 1.6.0","model":"TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (LED1545G12)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/0x90fd9ffffef4cb72/availability"}'
11/20/2019, 4:10:54 PM - info: MQTT publish: topic 'homeassistant/sensor/0x90fd9ffffef4cb72/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x90fd9ffffef4cb72","json_attributes_topic":"zigbee2mqtt/0x90fd9ffffef4cb72","name":"0x90fd9ffffef4cb72_linkquality","unique_id":"0x90fd9ffffef4cb72_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x90fd9ffffef4cb72"],"name":"0x90fd9ffffef4cb72","sw_version":"Zigbee2mqtt 1.6.0","model":"TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (LED1545G12)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/0x90fd9ffffef4cb72/availability"}'
11/20/2019, 4:10:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x90fd9ffffef4cb72/availability', payload 'online'
11/20/2019, 4:10:54 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170331","swBuildId":"1.2.217"}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden"}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"10":{"type":"Buffer","data":[76,69,68,49,53,52,53,71,49,50,69,50,55,69,85]},"modelId":"TRADFRI bulb E27 WS opal 980lm","dateCode":"20170331","powerSource":1}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"65533":1}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'devInterview' with data '"0x90fd9ffffef4cb72"'
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'devInterview' with data '"0x90fd9ffffef4cb72"'
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genGroups","data":{"65533":1,"nameSupport":0}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:55 PM - debug: Received zigbee message of type 'devInterview' with data '"0x90fd9ffffef4cb72"'
11/20/2019, 4:10:56 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:56 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"65533":1}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:56 PM - debug: Received zigbee message of type 'devInterview' with data '"0x90fd9ffffef4cb72"'
11/20/2019, 4:10:56 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"16387":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}}' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffef4cb72) of endpoint 1
11/20/2019, 4:10:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x90fd9ffffef4cb72', payload '{"state":"ON","linkquality":42}'
[... cut ...]
11/20/2019, 4:10:58 PM - debug: Setup reporting for '0x90fd9ffffef4cb72' - genLevelCtrl - 1
11/20/2019, 4:10:59 PM - debug: Setup reporting for '0x90fd9ffffef4cb72' - lightingColorCtrl - 3
11/20/2019, 4:10:59 PM - debug: Successfully setup reporting for '0x90fd9ffffef4cb72' - genOnOff - 1
11/20/2019, 4:10:59 PM - debug: Successfully setup reporting for '0x90fd9ffffef4cb72' - genLevelCtrl - 1
11/20/2019, 4:10:59 PM - debug: Successfully setup reporting for '0x90fd9ffffef4cb72' - lightingColorCtrl - 3
```

frauhottelmann, 2019-11-20:
> I'm sorry for re-opening but I want to be sure:
>
> @frauhottelmann Do you mean other Ikea devices or simply other devices?
>
> @Koenkk I noticed under zigbee-shepherd it eventually pairs after several [...]

Both. I had no problems with another IKEA bulb and an IKEA repeater as well as my Xiaomi sensors and a switch.

Koenkk, 2019-11-21:
@CMDR-Sloma can you post the log when running with `DEBUG=* npm start` and pairing the device (on zigbee2mqtt 1.7.1).

CMDR-Sloma, 2019-11-21:
@Koenkk I'm on hassio with zigbee2mqtt add-on and I'm afraid, I don't have the ability to run npm

Koenkk, 2019-11-22:
@CMDR-Sloma on the latest edge addon you can set `zigbee_herdsman_debug: true` in the config and it will show the debug logging.

timdonovanuk, 2020-03-13:
@Koenkk I have `-e DEBUG=*` added to my docker. 
It doesn't provide any additional info when pairing fails, but I do notice that at z2m startup, it shows:

`0x90fd9ffffe4d91b2 (0x90fd9ffffe4d91b2): Not supported (Router)`

This is also the device that later fails to pair:

```
debug 2020-03-13 16:44:15: Device '0x90fd9ffffe4d91b2' announced itself
info  2020-03-13 16:44:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x90fd9ffffe4d91b2"}}'
info  2020-03-13 16:44:17: Starting interview of '0x90fd9ffffe4d91b2'
error 2020-03-13 16:45:17: Failed to interview '0x90fd9ffffe4d91b2', device has not successfully been paired
```

#2779 #2761

HA-TB303, 2020-03-13:
Pairing via switch on/off/onoff/on/off/on/off reset:

\2020-03-13T17:13:13.333Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":55311,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":48,"securityuse":0,"timestamp":859493,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,139,10,0,0,16,0]}}
2020-03-13T17:13:13.335Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":139,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":55311,"endpoint":1,"linkquality":48,"groupID":0}'
zigbee2mqtt:debug 2020-03-13 18:13:13: Received Zigbee message from '0x00158d000309bedb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-03-13 18:13:13: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-03-13 18:13:13: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2020-03-13T17:13:13.338Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

HA-TB303, 2020-03-13:
I am not changing back to my CC2530 because switching back and forth costs me a day to re-pairn everything again.

Hi @Koenkk . What can I do to help analyse this issue? I have half of my house lights and sensors not working now, I have to WAF bitchin'...... So I am extremely motivated to help ;)

timdonovanuk, 2020-03-13:
Heh same, WAF is awful right now!

I just noticed zigbee_herdsman_debug logs seem to go to stdout which is a complete pain when running z2m in a docker using portainer. Any idea how to redirect them to a file?