Koenkk/zigbee2mqtt #2378: New livolo gateway

MiguelAngelLV, 2019-11-21:
Livolo launch a new gateway a news devices with use this gateway.

Any information about the support of thats devices? Will we need a new hardware to connect them?

https://es.aliexpress.com/item/4000343575116.html?spm=a219c.12010612.8148356.2.711e3c6cJANUpC

sfstar, 2019-11-24:
It seems that the new devices are backwards compatible with the old devices.
"Friend, our new gateway can compatible our old version and new version/socket zigbee switch, but the old gateway  can't compatible with new version zigbee switch/ socket. So we advice you order the new version." -- Qoute from the livolo alieexpress rep

Therefore, I speculate that the new devices might no longer be hard locked to channel 26 and might behave like routers. 
However, until I receive the new series outlets I ordered, this remains speculation at this point. (Livolo didn't comment on requests for clarification on channel use and whether the devices behave like routers)

MiguelAngelLV, 2019-11-25:
Ok, thanks.

I hope that the new devices works as router and can be supported by z2m :)

derelict, 2019-12-11:
Let us know ... i'm also interested in this.

sfstar, 2020-01-30:
Livolo Initially delayed the shipping of the outlets to allow for some quality improvements.
Today the outlet arrived and I tried to pair it to zigbee2mqtt to find it's device ID (before following the how to add an device guide) and to my surprise it paired as an normal livolo switch out of the box.
So either the livolo outlets do not have an unique device type identifier or zigbee2mqtt assumes all livolo products to be switches.

closed 2020-04-07 by stale[bot]