Koenkk/zigbee2mqtt #2379: "Hidden information" (battery status) of devices

kisseler, 2019-11-21:
Hey there,
first of all, Thank you for the great software! I use it with CC2530, RPi4 and Openhab (Mosquitto). I started pairing a few devices and I am wondering if there are any "hidden" data that are not being sent with the default MQTT message.
In my case (so far) it concerns the battery status of
- [Eurotronic Thermostate](https://www.zigbee2mqtt.io/devices/SPZB0001.html)
- [Aqara Door/Window Contact](https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html)
- [Aqara Double Switch](https://www.zigbee2mqtt.io/devices/WXKG02LM.html)
So generally the question, is there a way to "trigger" status information via MQTT?
Thanks in advance!

Koenkk, 2019-11-21:
You cannot trigger this, eventually they will send the data (can take 24 hours).

kisseler, 2019-12-10:
Thanks! I can confirm, that the information after some time (up to 24 h) will be published in the end.

closed 2019-12-10 by kisseler