Koenkk/zigbee2mqtt #2380: disable zigbee repeaters (xiaomi smart plug)

kitopopo, 2019-11-21:
hi,

Does anyone know if you can disable repeater mode on zigbee devices such as zigbee smart plug sockets?
I have 3 smart plug sockets that I would like to include within my zigbe2mqtt network but I am not interested in working as signal repeaters. these 3 plugs would not be plugged into plugs from UPS and I would not be interested in pairing any open sensor to these plugs.  My alarm would not work if the power goes out in my house and some sensor is paired to these signal repeaters. How i can dissable de router mode in my smart plug sockets?

Tanks in advanced.

Best regards

zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608

Koenkk, 2019-11-21:
In Zigbee it is not possible to do this, however I would expected that in case the power goes down the traffic of the alarm is routed through another device.

kitopopo, 2019-11-21:
> In Zigbee it is not possible to do this, however I would expected that in case the power goes down the traffic of the alarm is routed through another device.

Thanks for your response, I have read that end devices such as Xiaomi aperture sensors do not seek a new routing, I do not know if I am correct.

Thanks again

pcb1962, 2019-11-22:
> I have read that end devices such as Xiaomi aperture sensors do not seek a new routing, I do not know if I am correct.

I can confirm from my testing that Aqara end devices (I have around 40 door contacts, motion sensors and temperature/humidity sensors) will continue to find a route to the coordinator if you take down the router that they are shown connected to in the network map **if you're using the regular coordinator firmware**. I found the mesh to be much less resilient when I tried the source routing firmware, not an issue for me as I use zigbee2mqtt for sensors only, my bulbs remain on the Tradfri gateway.

kitopopo, 2019-11-22:
Hi, ok. Thanks for your answer. I will test if my final devices can take a new network route. Can i make a new answer? After i will close this issue.
Is it possible to hide the information of the zigbbee2mqtt log in a raspberry pi? My raspberry pi log is filled with zigbee information every minute. since the log I can see it through the command: journalctl -u zigbee2mqtt.service -f So I would like to know how I can hide the debug information only in my raspberry log. thanks in advanced. Best regards

pplucky, 2019-12-08:
> In Zigbee it is not possible to do this, however I would expected that in case the power goes down the traffic of the alarm is routed through another device.

What if we change the type from 'Router' to 'EndDevice' on the database.db file, won't this allow it?

sallazda, 2020-01-01:
> What if we change the type from 'Router' to 'EndDevice' on the database.db file, won't this allow it?

I am interested in the answer. I would like also choose which device is not a Router.

kitopopo, 2020-01-01:
Yes, I am also interested in this answer. It would be very interesting to be able to change 'router' to 'end device' on the database.db device and that these devices only work as end devices. 
I have 4 xiaomi plugs that I don't want to work as routers, just as end devices. I have these plugs unused and unplugged also have 4 Philips Hue lamps that I would like to work only as final devices, I could not add these lamps to the zigbe2mqtt because I do not want them to work as repeaters since I cut the power through the switch and then I would have routing problems with all my devices .

If the changes in the database.db file don't work, could you find another way to achieve this? 
I think a lot of people would be interested in getting this

thanks for your job, best regards.

kitopopo, 2020-01-02:
Hi dear friend,

As the fellow @pplucky  commented, today I tried replacing "router" with "end device" in the database.db file. After making this modification I generated a network map. I could see that my xiaomi samart socket has changed from blue to yellow in the network map (the routers are blue) and the devices that were linked to my smart socket now appear without lines. Does this modification really make my samart socket work only as an end device? or still working as a router? thanks, best regards

kitopopo, 2020-01-26:
According to my last tests carried out during the last month, 
I observe that changing in the configuration file "end device" instead of "router" I get the xiaomi smart plug to work as final devices. No end device is linked to the smart socket, but if the samart socket is linked to other smart sockets. Also on the network map  we can see how change the smart socket from blue (router) to yellow (end device). 
This operation can be seen and confirmed by generating the zigbee network map.

closed 2020-01-26 by kitopopo

steve41200, 2020-10-14:
I have try. It works, but we lost the possibility to ping device. it's too bad. Zigbee2mqtt see it as a End Device he thinks on battery so impossible to ping it :(. Have you a tips to know if the device is online or offine ?

andrazek, 2024-03-29:
If that truly works, why there's no option to change this under the Zigbee2MQTT settings for every device?

heisenberg2980, 2024-06-06:
> According to my last tests carried out during the last month, I observe that changing in the configuration file "end device" instead of "router" I get the xiaomi smart plug to work as final devices. [...]

@Koenkk according to @kitopopo it seems it is possible to change the settings file so a router is consider an end device and it doesn´t route traffic from other devices. I am having issues at the moment with a router that is not routing messages properly so I will try this option, but it would be really handy to have the option added to z2m, do you think this would be possible?

g-work, 2024-10-17:
Can someone confirm if editing the database really makes a router device an end device? Or does it only show up as end device in HA and continues routing messages?

fakorte, 2024-10-23:
This problem has been bothering me for a long time and I would be very interested in a solution.

louis-lau, 2025-08-29:
The first answer of this thread already says it's not possible. Changing a value in the db does not suddenly make it possible. Wether a device routes is based on the firmware of the device, nothing else can control it.

Yes, I hate it as well. I think ZigBee should have accounted for this. They did not. But if anyone finds this thread, let me take the confusion caused by the past few messages away: it can not be done.

TLDR: Listen to the creator of this project. They know what they're doing.