Koenkk/zigbee2mqtt #2381: Need help: advice for hardware wall thermostat + heater thermostats

bartplessers, 2019-11-21:
Hi everyone,

I need some advice.
I want to buy 
- a wireless wall thermostat and 
- a wireless heater thermostat

compatible with zigbee2mqtt.

In example
### Bitron AV2010/32
https://www.zigbee2mqtt.io/devices/AV2010_32.html

and 

### Eurotronic SPZB0001
https://www.zigbee2mqtt.io/devices/SPZB0001.html

Main purpose:
- huge WAF (Wife Acceptance Factor)
  - easy to use
  - "normal, classic" behaviour of wall thermostat: turn on/off, set temp, etc...
  - ability to simple program (daily, weekend, etc..) on wall thermostat
  - wall thermostat regulates the individual heater thermostats (master/slave model): temperature is set by wall thermostat, heater thermostats follow temp set by wall thermostat

- together with hug GEEK-factor
  - connect to zigbee2mqtt
  - get control with home assistant
  - etc...

So the big picture is: my wife should be able to regulate the temp without any knowledge of  hass or app or stuff like that. Just the old-fashioned way.
I want to surprise her with home assistant and automatisation...

Is this possible?
What hardware do you recommend?
Oh, did I mention that it has to be cheap too? :-)

kind regards,
Bart

bartplessers, 2019-12-04:
Nobody on this one? Really?

sjorge, 2019-12-04:
I'm also not sure a those TRV's can request heating on/off from the wall thermostat... maybe with some kind of binding? But AFAIK nobody has done it yet.

I'll be in similar need in about half a year once I move into a new place... I picked up a SPZB0001 last week and there are still some oddities here that need some work. I'll try to fix some of these though but so far it has been a complex and temperamental device.

bartplessers, 2019-12-14:
Maybe I have to look for Wi-Fi wall units, and do the logic in hass/domoticz/node-red.

Maybe something like 
https://m.nl.aliexpress.com/item/32852946950.html

Has anybody experience with this device? 
Can it be used as a "master" for the TRV's? (Thermostat wall unit --> Wi-Fi--> home assistant --> mqtt --> zigbee2mqtt  --> zigbee --> TRV's)

Thanx for responses

closed 2020-02-19 by stale[bot]