Koenkk/zigbee2mqtt #2385: version": "1.7.1: Missing attribute status: offline / online

zibous, 2019-11-22:
I see that the MQTT payload has changed, with some devices I am missing the status "Offline / Online". I can not find any attitude so I can get this with the new version again.

Is there an alternative for that?

## "version": "1.6.0"
```
{
	"occupancy": false,
	"linkquality": 42,
	"last_seen": "2019-08-05T11:27:15+02:00",
	"battery": 100,
	"voltage": 3075,
	"no_occupancy_since": 120,
	"elapsed": 90000,
	"device": {
		"ieeeAddr": "0x00158d000358e8cd",
		"friendlyName": "bewegungsmelder1",
		"type": "EndDevice",
		"nwkAddr": 3532,
		"manufId": 4151,
		"manufName": "LUMI",
		"powerSource": "Battery",
		"modelId": "lumi.sensor_motion",
		"hwVersion": "unknown",
		"swBuildId": "unknown",
		"dateCode": "unknown",
		"status": "online"                  < ----- allready present 
	}
}
```
## "version": "1.7.1+dev"
```
{
	"occupancy": false,
	"linkquality": 42,
	"last_seen": "2019-11-22T08:50:54+01:00",
	"battery": 100,
	"voltage": 3045,
	"no_occupancy_since": 240,
	"elapsed": 59999,
	"device": {
		"friendlyName": "bewegungsmelder1",
		"ieeeAddr": "0x00158d000358e8cd",
		"networkAddress": 47876,
		"type": "EndDevice",
		"manufacturerID": 4151,
		"manufacturerName": "LUMI",
		"powerSource": "Battery",
		"applicationVersion": 11,
		"stackVersion": 2,
		"zclVersion": 1,
		"hardwareVersion": 2,
		"dateCode": "20160524",
		"softwareBuildID": "3000-0001"
	}
}
```

Koenkk, 2019-11-22:
The status has been removed as it was completely unreliable, if you want to the have the device availability you should use the availability feature (https://www.zigbee2mqtt.io/information/availability.html)

zibous, 2019-11-22:
@Koenkk 

Thanks for this 👍 
Is there any way to set the availability_timeout with mqtt command ?

Koenkk, 2019-11-22:
No, this is not possible.

zibous, 2019-11-22:
So i have to change the settings and restart.
ok.

closed 2019-11-22 by zibous