Koenkk/zigbee2mqtt #2386: SmartThings Multi-Sensor (3321-S) not reporting contact.

jonnyrider, 2019-11-22:
I know that the 3321-S is supported, and it is reporting link quality and temperature correctly, but the actual contact status is not reported.

The only thing in the add-on log is:
zigbee2mqtt:info  2019-11-22T09:08:17: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bbaa7cc', payload '{"temperature":22.25,"linkquality":68}'

And an error appears in the developer log:
No matching payload found for entity: 0x000d6f000bbaa7cc_contact with state_topic: zigbee2mqtt/0x000d6f000bbaa7cc

## Debug Info
zigbee2mqtt version: 1.7.1
CC2531 firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
Device: 0x000d6f000bbaa7cc (0x000d6f000bbaa7cc): 3321-S - SmartThings Multi Sensor (2015 model) (EndDevice)

Thanks

Koenkk, 2019-11-22:
Can you share `database.db`?

jonnyrider, 2019-11-22:
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/3880283/database.txt)

Koenkk, 2019-11-22:
Interviewing failed it seems. Can you try repairing the device when running with `DEBUG=* npm start`?

jonnyrider, 2019-11-22:
I'm using hassio. If I turn on debug logging and re-pair will that give you
the same result?

Koenkk, 2019-11-22:
For hassio  switch to edge and set `zigbee_herdsman_debug: true` in the config.

bassicrob, 2019-11-25:
I'm having similar issue with standalone install of Zigbee2mqtt on separate Pi from Hassio.  Stopped working when I upgraded to 1.6.1-dev and still not reporting contact on 1.7.1. All other attributes report fine.

jonnyrider, 2019-11-25:
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/3885348/debug.txt)

Thanks

Koenkk, 2019-11-25:
Can you check if this is fixed in the latest dev?

jonnyrider, 2019-11-25:
Yes, it is. I had to remove and re-link it, but it works perfectly now.

Thanks very much!

closed 2019-11-25 by jonnyrider