Koenkk/zigbee2mqtt #2390: Aqara wxkg11lm - More than one broadcast for one action

WouterJN, 2019-11-22:
# Bug Report
## What happened

Zigbee2mqtt sent out broadcast two MQTT messages when pressing an Aqara wxkg11lm. The first message is with the actual amount of clicks, the second message is empty.

```
zigbee2mqtt:info  2019-11-22T21:00:52: MQTT publish: topic 'zigbee2mqtt/Bedroom_Aqara_Button_Bed', payload '{"battery":100,"voltage":3025,"linkq
uality":41,"last_seen":"2019-11-22T21:00:52.223Z","click":"single"}'                                                                            
zigbee2mqtt:info  2019-11-22T21:00:52: MQTT publish: topic 'zigbee2mqtt/Bedroom_Aqara_Button_Bed', payload '{"battery":100,"voltage":3025,"linkq
uality":41,"last_seen":"2019-11-22T21:00:52.223Z","click":""}'  
```
## What did you expect to happen
Only the first MQTT broadcast

## Debug Info
zigbee2mqtt version: 1.7.1

Koenkk, 2019-11-22:
This is expected, see https://github.com/Koenkk/zigbee2mqtt/issues/1155#issuecomment-467094619

Duplicate of #1155 and #1212

closed 2019-11-22 by Koenkk

WouterJN, 2019-11-22:
Thank you, it is clear to me now.

Was the behaviour different in the past?