Koenkk/zigbee2mqtt #2391: advanced settings are not merged

runningman84, 2019-11-22:
# Bug Report
## What happened
I have configured a network key and cannot start zigbee2mqtt anymore:
```
Nov 22 21:20:33 raspberrypi systemd[1]: Started zigbee2mqtt.
Nov 22 21:20:34 raspberrypi npm[3470]: > zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
Nov 22 21:20:34 raspberrypi npm[3470]: > node index.js
Nov 22 21:20:36 raspberrypi npm[3470]: /opt/zigbee2mqtt/lib/util/logger.js:12
Nov 22 21:20:36 raspberrypi npm[3470]: const directory = settings.get().advanced.log_directory.replace('%TIMESTAMP%', timestamp);
Nov 22 21:20:36 raspberrypi npm[3470]:                                          ^
Nov 22 21:20:36 raspberrypi npm[3470]: TypeError: Cannot read property 'log_directory' of null
Nov 22 21:20:36 raspberrypi npm[3470]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:12:42)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Module._compile (internal/modules/cjs/loader.js:701:30)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Module.load (internal/modules/cjs/loader.js:600:32)
Nov 22 21:20:36 raspberrypi npm[3470]:     at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Function.Module._load (internal/modules/cjs/loader.js:531:3)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Module.require (internal/modules/cjs/loader.js:637:17)
Nov 22 21:20:36 raspberrypi npm[3470]:     at require (internal/modules/cjs/helpers.js:22:18)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/mqtt.js:2:16)
Nov 22 21:20:36 raspberrypi npm[3470]:     at Module._compile (internal/modules/cjs/loader.js:701:30)
```

## What did you expect to happen
zigbee2mqtt should start and use default advanced settings but for network_key it should use my specific value.

## How to reproduce it (minimal and precise)
Install zigbee2mqtt on a rpi3 and use a config like this:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.178.201'
serial:
  port: /dev/ttyACM0
advanced:
  network_key: [11, 12, 13, 14 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26]
```

## Debug Info
zigbee2mqtt version: 1.7.1

Ton1965, 2019-11-23:
Hi,
It looks like a comma is missing between 14 and 15 in your network key. Most probably this is a root cause of the error.

runningman84, 2019-11-23:
I don't think so this was just an effort to hide my real network key... the real key does not suffer from this problem.

closed 2020-01-30 by stale[bot]