Koenkk/zigbee2mqtt #2392: IAS devices need to be reconfigured after controller change

skandragon, 2019-11-22:
# Bug Report
## What happened

I swapped out a controller with a new one (much better range in theory) -- all devices managed to reconnect and function, other than IAS devices.

I believe this is because the IAS endpoints are not reconfigured on startup, as I believe using the controller address of all 0s is not valid, so it must be an actual physical address.

## What did you expect to happen

IAS devices should continue working.

I think this could be done manually (as in "reconfigure this device fully") in the hopefully rare event when one must change out a controller.

## How to reproduce it (minimal and precise)

Change out a controller with a motion or door sensor in the network.

Koenkk, 2019-11-25:
yes you are right, they should indeed be reconfigured after a controller change. For now you can solve the problem by setting `interviewCompleted: false` in the database.db, and trigger the device to send a message (e.g. press reset button once).

skandragon, 2019-11-25:
I am not sure this is working as intended.  The specific device is a Bosh motion sensor ISW-ZPR1-WP13.  Based on my previous experience with this device, it accepts one command per message sent, so one must trigger the tamper switch about once every 2 seconds to make the config messages finally make it through.  However, even doing this, it doesn't seem to be bringing things online.

closed 2020-01-31 by stale[bot]