Koenkk/zigbee2mqtt #2393: Device level state exclusion?

j4c3, 2019-11-23:
Is it possible to exclude specific devices from publishing the cached state on startup?  

cache_state: false on specific devices in configuration.yaml does not seem to do it.

j4c3, 2019-11-26:
Based on state.js, I may work around this by changing the fz.LZL4B_onoff  to be something in this list, as opposed to 'level', which the original message used:

const dontCacheProperties = [
    'action', 'button', 'button_left', 'button_right', 'click', 'forgotten', 'keyerror',
    'step_size', 'transition_time', 'action_color_temperature', 'action_color',
    'action_group', 'group_list', 'group_capacity', 'no_occupancy_since',
    'step_mode', 'transition_time', 'duration', 'elapsed', 'from_side', 'to_side',
];

I'm thinking 'click'.

Koenkk, 2019-11-27:
What are you trying to accomplish, could you give some more context?

j4c3, 2019-11-27:
With a remote, if you turn a group to on, that is saved to the state file.  If something else modifies the group, the state file retains the value from the remote, and on startup publishes that state.  

It doesn't seem that important, but could turn something on at a time when you really might not want.  But, it looks like the fix already exists, since you're excluding those specific properties from the state file.

Koenkk, 2019-11-27:
So what the problem now?

j4c3, 2019-11-27:
None!

closed 2019-11-27 by j4c3