Koenkk/zigbee2mqtt #2394: Haozee Smoke Detector (Heiman HS1A-N) not working

Hugyleviking, 2019-11-23:
# Bug Report
## What happened
Pairing with zigbee2mqtt but not working "NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?"

Received message from unsupported device with Zigbee model 'SmokeSensor-N-3.0
Successfully interviewed '0x00158d000292e92a', device has successfully been paired

data form database:
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000292e92a","nwkAddr":36133,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-N-3.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,1282,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SmokeSensor-N-3.0","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":20,"stackVersion":2,"hwVersion":16,"dateCode":"2018.9.27","swBuildId":"2000-0001"}}},"binds":[]}},"appVersion":20,"stackVersion":2,"hwVersion":16,"dateCode":"2018.9.27","swBuildId":"2000-0001","zclVersion":2,"interviewCompleted":true,"meta":{}}

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608

groenmarsmannetje, 2019-11-23:
Try to restart. I had something similar and it appeared when I restarted Home Assistant and zigbee2mqtt.

Hugyleviking, 2019-11-23:
I have restart before posting. Thank you. I think it's new material, little different from HS1SA original. Mine is HS1SA-N.

Hugyleviking, 2019-11-24:
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d000292e92a","supported":false}}

Koenkk, 2019-11-25:
This device is not supported yet, please check: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

It will probably work with the following definition:
```js
{
        zigbeeModel: ['SmokeSensor-N-3.0'],
        model: 'HS3SA',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [fz.heiman_smoke, fz.battery_200],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

Hugyleviking, 2019-11-25:
I have try. I have Hass.io add-on on debian 10.
But I have no much skill to do what is asked on the page  https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Hugyleviking, 2019-11-25:
I have copied and modified the device.js file with instructions from Docker.

prankousky, 2019-11-30:
Will this be included in future versions?

Apparently, there are many different versions of this Heiman smoke detecter sold under different names and/or without exact model information.

I just received a device today that resembles [this one](https://www.zigbee2mqtt.io/devices/HS3SA.html) 100 percent, but due to lack of an exact model number description on the shop page, I believe I received another model as well. It pairs without any trouble, but is listed as an unsupported device as well. And just like OP I don't have the skills to include this by myself.

lechk82, 2019-12-17:
> This device is not supported yet, please check: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
>
> It will probably work with the following definition:
>
> ```js
> {
>         zigbeeModel: [...]

Successfully tested a Haozee Smoke Detector (zigbeeModel SmokeSensor-N-3.0) with this configuration. The device reported
`"battery": 100, "linkquality": 38, "smoke": true,"battery_low": false`
after pressing the alarm test button.

Koenkk, 2019-12-17:
@lechk82 thanks for confirming, will be support in the zigbee2mqtt dev branch in a few hours.

closed 2019-12-17 by Koenkk

Hugyleviking, 2019-12-17:
Thank You all very much !!!!

Hugyleviking, 2019-12-18:
Working well !!  Thank You !!

ChumKiu, 2020-01-07:
> This device is not supported yet, please check: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
>
> It will probably work with the following definition:
>
> ```js
> {
>         zigbeeModel: [...]

@Koenkk,
thank for your work!
I kindly ask if you can let me know some details around the right usage of the keyword meta: {configureKey: 1}, I see that somethings is is written with 2 instead 1, please I'm try to better understand how to start in coding some adapter for new devices.
Thank!
Best regards

Koenkk, 2020-01-08:
@ChumKiu if the configure key changes, it indicates to zigbee2mqtt that the configure method has to be executed again. So this configure key should only be bumped when the configure method changes.

ChumKiu, 2020-01-09:
@Koenkk thank for your answer, so for example, if is the first time that I've to set the configure this will set to 1, if in future I've to change something I've to increment this to 2 and so on? There is a way to reset it again to 1 or this will have conseguence on the actually paired devices into the network?
Sorry but I'm still learing how to build my devices...
Thank!

Koenkk, 2020-01-09:
Yes, everytime you change something in the `configure` increment it, BUT never decrease the value, e.g. somebody else could still be on `configure` `1` , if you would than change it to `1` again the user won't have it's device reconfigured.

ChumKiu, 2020-01-09:
@Koenkk 
Thank for the explanation, so this act as one indicator for compatibility, it have nothing to do effectively with the device itself, so I can let it set to 1 and make all the change I need to my device definition until this is definitively and may be deployed to some final release. I'm right?
Also I've a doubt, if a manufacturer improve the device, I may have need to update the configuration for the new device starting from some data, changing this configureKey will affect also the "old" device that was configured with an old parametrization and may be the device will not behave properly?
Thank!

Koenkk, 2020-01-09:
> I need to my device definition until this is definitively and may be deployed to some final release. I'm right?

yes you are right

> Also I've a doubt, if a manufacturer improve the device, I may have need to update the configuration for the new device starting from some data, changing this configureKey will affect also the "old" [...]

true, but in practice this (almost) never happens

zander, 2020-01-20:
I got this issue too, with the same hardware, then updated (git pull to master, npm install) and now I get;

```
zigbee2mqtt:error 2020-01-20 19:31:36: Failed to configure 'alarm/one', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
at Timeout._onTimeout (/home/services/zigbee/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
at listOnTimeout (internal/timers.js:531:17)
at processTimers (internal/timers.js:475:7))
```

Does the config work for others, and is it just me that doesn't have a functioning smoke sensor?

thomaschampagne, 2020-03-09:
@zander I got the same issue right now on my **HS1SA-N** and latest docker image (on 2020-03-09) on raspberry pi 4.

@Koenkk Does this speak to you? Thanks for your help!

**zigbee2mqtt logs:**

```javascript
zigbee2mqtt:error 2020-03-09 14:02:28: Failed to configure 'ent_smoke', attempt 1 (Error: Bind 0x00158d0002755950/1 genPowerCfg from '0x00124b0008b80ba0/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms),
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23),
    at Generator.throw (<anonymous>),
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

**database.db added entry:**
```json
{
  "id": 21,
  "type": "EndDevice",
  "ieeeAddr": "0x00158d0002755950",
  "nwkAddr": 5245,
  "manufId": 4619,
  "manufName": "HEIMAN",
  "powerSource": "Battery",
  "modelId": "SmokeSensor-N-3.0",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        3,
        1,
        1280,
        1282,
        2821
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "SmokeSensor-N-3.0",
            "manufacturerName": "HEIMAN",
            "powerSource": 3,
            "zclVersion": 2,
            "appVersion": 20,
            "stackVersion": 2,
            "hwVersion": 16,
            "dateCode": "2018.9.27",
            "swBuildId": "2000-0001"
          }
        },
        "ssIasZone": {
          "attributes": {
            "iasCieAddr": "0x0000000000000000",
            "zoneState": 1
          }
        },
        "genPowerCfg": {
          "attributes": {
            "batteryPercentageRemaining": 200
          }
        }
      },
      "binds": [
        {
          "cluster": 1,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0008b80ba0",
          "endpointID": 1
        }
      ]
    }
  },
  "appVersion": 20,
  "stackVersion": 2,
  "hwVersion": 16,
  "dateCode": "2018.9.27",
  "swBuildId": "2000-0001",
  "zclVersion": 2,
  "interviewCompleted": false,
  "meta": {
    "configured": 1
  },
  "lastSeen": 1583762758383
}

```

Koenkk, 2020-03-09:
Can you make sure to keep the device awake during pairing? (by e.g. pressing a button on it constantly).

thomaschampagne, 2020-03-09:
@Koenkk Perfect !! It worked :) !! This PR https://github.com/Koenkk/zigbee2mqtt.io/pull/239 for the documentation update.

```bash
zigbee2mqtt:info  2020-03-09 21:17:39: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0002755950/smoke/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.smoke }}","device_class":"smoke","state_topic":"zigbee2mqtt/ent_smoke","json_attributes_topic":"zigbee2mqtt/ent_smoke","name":"ent_smoke_smoke","unique_id":"0x00158d0002755950_smoke_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0002755950"],"name":"ent_smoke","sw_version":"Zigbee2mqtt 1.11.0","model":"Smoke detector (HS3SA)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}',
zigbee2mqtt:info  2020-03-09 21:18:13: Successfully configured 'ent_smoke',
zigbee2mqtt:info  2020-03-09 21:18:11: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":21,"smoke":false,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:10: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":23,"smoke":true,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:09: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":13,"smoke":false,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:09: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":10,"smoke":true,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:08: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":13,"smoke":true,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:07: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":21,"smoke":false,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:06: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":28,"smoke":false,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:05: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":34,"smoke":true,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:04: MQTT publish: topic 'zigbee2mqtt/ent_smoke', payload '{"battery":100,"linkquality":28,"smoke":false,"battery_low":false}',
zigbee2mqtt:info  2020-03-09 21:18:03: Configuring 'ent_smoke',
zigbee2mqtt:info  2020-03-09 21:18:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"ent_smoke"}}',
zigbee2mqtt:error 2020-03-09 21:18:03: Failed to interview 'ent_smoke', device has not successfully been paired,
zigbee2mqtt:info  2020-03-09 21:17:39: MQTT publish: topic 'homeassistant/sensor/0x00158d0002755950/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/ent_smoke","json_attributes_topic":"zigbee2mqtt/ent_smoke","name":"ent_smoke_linkquality","unique_id":"0x00158d0002755950_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0002755950"],"name":"ent_smoke","sw_version":"Zigbee2mqtt 1.11.0","model":"Smoke detector (HS3SA)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}',
zigbee2mqtt:info  2020-03-09 21:17:39: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0002755950/battery_low/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.battery_low}}","device_class":"battery","state_topic":"zigbee2mqtt/ent_smoke","json_attributes_topic":"zigbee2mqtt/ent_smoke","name":"ent_smoke_battery_low","unique_id":"0x00158d0002755950_battery_low_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0002755950"],"name":"ent_smoke","sw_version":"Zigbee2mqtt 1.11.0","model":"Smoke detector (HS3SA)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}',
zigbee2mqtt:info  2020-03-09 21:17:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"ent_smoke"}}',
zigbee2mqtt:info  2020-03-09 21:17:36: Starting interview of 'ent_smoke',
zigbee2mqtt:info  2020-03-09 21:17:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"ent_smoke"}}',
zigbee2mqtt:info  2020-03-09 21:17:36: Device 'ent_smoke' joined,
zigbee2mqtt:info  2020-03-09 21:17:24: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0","commit":"31e5678","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
```

Hugyleviking, 2020-03-11:
> Koenkk/zigbee2mqtt.io#239

Hi,

no more working with the last update.

Best regards.

Hugyleviking, 2020-03-13:
After deconect ans reconnect USB Cc2531 dongle + restart zigbee2mqtt,
working again.