Koenkk/zigbee2mqtt #2395: installing Zigbeemqtt as a Windows10 service

regissg, 2019-11-23:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

regissg, 2019-11-23:
I need some help to install Zigbeemqtt as a service on Windows 10. 
I am able to launch node red as a service with pm2 but I do not know how to do it for zigbeemqtt.
Thanks in advance for your help

carldebilly, 2020-01-22:
You can start a linux VM on your machine and install docker in it.

Or you can wrap Z2M into a new package and publish it like what danielwelch is doing for the HASS.IO add-in. https://www.thepythoncorner.com/2018/08/how-to-create-a-windows-service-in-python/

closed 2020-03-29 by stale[bot]

stevensegers, 2024-06-30:
I ran into this question too. Managed to make it work.
As Zigbee2Mqtt is an npm package, the question can be rephrased as "How to run an npm package as a Windows service?"
There seem to be many solutions for that. I tried https://github.com/suyeonORG/qckwinsvc2, which was pretty straight-forward.

To summarize:
```
npm i qckwinsvc2 -g
qckwinsvc2 install
```
Answer the few questions, and you're done.

Check the README page for more details.

Note: I assume that npm i is supposed to add the directory where qckwinsvc2.cmd lands to your path. (or it's supposed to be in your path already). For me that did not work, so I had to use the full path to the script: %appdata%\npm\qckwinsvc2.cmd