Koenkk/zigbee2mqtt #2396: New feature: Touchlink factory reset

Koenkk, 2019-11-23:
Place holder, soon....

tobimori, 2019-11-25:
```
zigbee2mqtt:error 2019-11-25 15:01:14: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SREQ '--> AF - interPanCtl - {"cmd":1,"data":[16]}' failed with status '186' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:245:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58))
```

Crashes for me, too.

tobimori, 2019-11-25:
Not working for me. GU10 White&Color Newest Gen (with BT)

lolorc, 2019-11-25:
oh i guess i just found an usage for my old cc2531 :)

Koenkk, 2019-11-26:
Also works for non hue bulbs now (tested with Innr and TRADFRI).

sjorge, 2019-11-26:
So it only works when it is close to the controller right? You can do it like with join where you have the bulb near a router?

Koenkk, 2019-11-26:
Yes it has to be close to the coordinator (demo: https://www.youtube.com/watch?v=kcRj77YGyKk).

bascloin, 2019-11-29:
Hi there,

I get the following error after sending the MQTT message (on 1.7.1+dev (commit #5e16581)):

info  2019-11-29 22:04:46: Starting touchlink factory reset...
error 2019-11-29 22:04:46: Failed to call 'BridgeConfig' 'onMQTTMessage' (TypeError: this.herdsman.touchlinkFactoryReset is not a function
    at Zigbee.touchlinkFactoryReset (/opt/zigbee2mqtt/lib/zigbee.js:249:30)
    at BridgeConfig.touchlinkFactoryReset (/opt/zigbee2mqtt/lib/extension/bridgeConfig.js:336:42)
    at BridgeConfig.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/bridgeConfig.js:314:44)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))

Could you advise me on this?

Koenkk, 2019-11-29:
@bascloin in the zigbee2mqtt directory, execute `rm -rf node_modules && npm install`.

bascloin, 2019-11-29:
That was it... thanks a lot man!

ejalal, 2019-11-30:
Hi @Koenkk 
Can't get this to work with the hassio zigbee2mqtt-edge image (Zigbee2mqtt 1.7.1+dev)
The bulbs blinks when I try to factory reset them, but then they doesn't pair (I use CC2531 USB dongle)
Actually the very first bulb (1 out of 4) paired with the stable release 1.7.1 without any problems, but all 3 other bulbs don't want to pair at all, this why I found this topic.
My bulbs (all 4 brand new) are Hue white A60 bulb E27 bluetooth (8718699673147)
Really need some help here, but I can get you some logs if you want to take a look at it

ejalal, 2019-11-30:
I just paired one of the failing bulbs with the Hue Bluetooth app and it paired fine, it's detected as a LWA001 in the app, but has a different model number printed on the bulb itself which is 9290018216 which is different from the one supported (8718699673147)

sjorge, 2019-12-01:
Just reset a `Philips Hue white and color ambiance E26/E27/E14 (9290012573A)` using this methode.... I had join disable.

I then moved the bulb to the new spot, enabled join and turned on the bulb. It joined by using the nearby router and all is well.

So you can both reset and join at the coordinator or only do the reset step. This is great!

ejalal, 2019-12-02:
It did work in the end, I just had to literally reboot the raspberry pi, not just home assistant or zigbee2mqtt containers.

Conphilpott, 2019-12-15:
Brilliant addition, was finally able to get my hue go 2.0 light to enter pairing mode using this!

stevevg, 2019-12-17:
Worked like a charm on two Philips Hue 9290018216, using zigbee2mqtt 1.8.0 and home assistant (hassio) 0.100.3 on raspberry pi 3. Thanks!

aurelijusrozenas, 2019-12-27:
Hi. I don't suppose there would be a way to send factory reset message through the router version of CC2531? I would be handy walking around with powerbank and router and pairing light bulbs.

Koenkk, 2019-12-27:
No thats not possible

ikbenfrank, 2020-01-24:
I did a factory reset using a CC2531 and on zigbee2mqtt 1.9 and it worked perfectly on my Ikea LED1545G12. Thanks

myradon, 2020-01-25:
I was able to factory reset Philips Hue Bulb through z2m touchlink (v1.9.0). Also did the "10 second reset" with Hue switch. Hue Bulb starts to flash but it will Not expose itself to z2m. Also did power cycle it. Nothing. Hue bridge will find the 'new' lamp. In log I see this;
`zigbee2mqtt:info  2020-01-25 21:21:08: Starting touchlink factory reset...

zigbee2mqtt:debug 2020-01-25 21:21:24: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset\' with data ''

zigbee2mqtt:info  2020-01-25 21:21:24: Starting touchlink factory reset...

zigbee2mqtt:info  2020-01-25 21:21:27: Successfully factory reset device through Touchlink

zigbee2mqtt:info  2020-01-25 21:21:27: Successfully factory reset device through Touchlink

zigbee2mqtt:debug 2020-01-25 21:25:01: Saving state to file /app/data/state.json

(node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '17' (expected '0')

    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)

    at Generator.next (<anonymous>)

    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)

(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)

(node:16) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

(node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '17' (expected '0')

    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)

    at Generator.next (<anonymous>)

    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)

(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)

zigbee2mqtt:debug 2020-01-25 21:30:01: Saving state to file /app/data/state.json

(node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '17' (expected '0')

    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)

    at Generator.next (<anonymous>)

    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)

(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 3)`

lancelon, 2020-01-25:
Anyone know if this will work with IKEA blinds? Having the most awful trouble trying to pair a blind that just won't react in any way. Wondering about reflashing the firmware to the CC2531?! Could that help? CC2531 still talking to all other existing devices but doesn't want to pair any new ones.

theFork, 2020-02-09:
Does this still require running on the dev branch? I would like to use this in production without upgrading my working setup.

sjorge, 2020-02-09:
@theFork this is in 1.10 AFAIK

talondnb, 2020-02-16:
I've tried this multiple times with a new LED1738G7 that I'm trying to re-pair and unfortunately it won't work. I receive the successful message for the reset and the bulb goes into a glowing state, but trying to re-pair always fails the interview.

Anything else I can do to try and re-pair the LED1738G7?

theFork, 2020-02-16:
@talondnb  For Tradfri devices such as the LED1738G7, there is a switching sequence to initiate a factory reset. There is a link to a video showing how to do this right here: https://www.zigbee2mqtt.io/devices/LED1738G7.html

talondnb, 2020-02-16:
> @talondnb For Tradfri devices such as the LED1738G7, there is a switching sequence to initiate a factory reset. There is a link to a video showing how to do this right here: https://www.zigbee2mqtt.io/devices/LED1738G7.html

Sorry I should have also mentioned I’ve tried the usual methods. I just cannot seem to re-pair the globe no matter what I try.

I’ve had to roll back to a snapshot to gain control of the globe again.

One thing to mention is that I was able to re-pair a Xiaomi temp sensor with no issues.

timdonovanuk, 2020-02-17:
Crazy question, but the whole "bring the device close (< 10 cm)" makes me nervous - how does this work? What if devices, due to a big fat antenna on the CC1352P-2, have amazing link quality and you go and issue the factory_reset command? Bang..you've just factory reset all your devices...

timdonovanuk, 2020-02-17:
Also, can you use a second spare cc2531(in coordinator mode) to do this (i.e. not the coordinator you want the device to join)?

Koenkk, 2020-02-18:
> Crazy question, but the whole "bring the device close (< 10 cm)" makes me nervous - how does this work?

The coordinator will send out a touchlink factory reset request, if any devices receives it it will check if it is close (based on the linkquality). 

> Also, can you use a second spare cc2531(in coordinator mode) to do this (i.e. not the coordinator you want the device to join)?

Yes, but make sure permit_join is set false at the instance of zigbee2mqtt.

Dylantje, 2020-02-22:
Dear... I try to reset a hue bulb color fist edition.
Only the reset is working ok.
Bud the bulb will not learn into 1.10 version.
I try, remote 1 and 0 untill the bulb blinck
light go on and off
then the remote buttons relase
no thing happend.
same thing 
make the bulb powerless and power up
no learning bulb...?

And the touchlink i do not understand where and how i to use this.
Is there perhaps a better explane for touchlink?
And with touclink i reset only one bulb or all philips bulbs, or the whole setup from the HUE?

Sorry noob here :-(

IVI053, 2020-03-25:
I'm trying touchlink with a 7099860PH - no success.
It always just fails.

```
   zigbee2mqtt:info  2020-03-25 00:19:37: Starting touchlink factory reset...
   zigbee2mqtt:info  2020-03-25 00:19:37: MQTT publish: topic 'z2m/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
   zigbee2mqtt:warn  2020-03-25 00:20:02: Failed to factory reset device through Touchlink
   zigbee2mqtt:info  2020-03-25 00:20:02: MQTT publish: topic 'z2m/bridge/log', payload '{"type":"touchlink","message":"reset_failed","meta":{"status":"failed"}}'
```

Device is right next to my CC26X2R1.

Koenkk, 2020-03-25:
@IVI053 I doubt if this device supports Zigbee (cannot find it on the internet)

IVI053, 2020-03-25:
https://www.zigbee2mqtt.io/devices/7099860PH.html ¯\_(ツ)_/¯
And there are some descriptions on the web that it can be paired with Hue gateway. Also has "smartlink" logo on it what is afaik a Philips-term for ZLL.

Interestingly I found in the morning that my whole zigbee network died shortly after I tried the touchlink reset through the coordinator. After login in to my z2m instance I found it with the following:

```
 (node:12686) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 5000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
(node:12686) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)
(node:12686) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 5000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:12686) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5)
(node:12686) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 5000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:12686) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 6)
(node:12686) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 5000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:12686) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 7)
(node:12686) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 3)
(node:12686) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 4)
(node:12686) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 5)
(node:12686) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 6)
(node:12686) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 7)
```

Koenkk, 2020-03-25:
@IVI053 do you use a CC1352P or CC2652R? The crash after touchlink reset has been fixed in the latest dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

IVI053, 2020-03-25:
Yes a CC2652R. Will update this week and test. Could it be, that this is also the cause for not being able to get the 7099860PH to work?

Koenkk, 2020-03-25:
Yes, touchlink fails because of that.

IVI053, 2020-04-12:
I finally found some time to change to your linked dev firmware. Got a different error now:
```
zigbee2mqtt:error 2020-04-12 23:03:41: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SRSP - AF - interPanCtl after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

Koenkk, 2020-04-13:
@IVI053 could you provide the herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

IVI053, 2020-04-19:
Sorry for the delay but since my son was born I can not always find the time to test immediately.
Here is the output with herdsman debug:

```
gbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
zigbee2mqtt:info  2020-04-19 14:37:29: Starting touchlink factory reset...
zigbee2mqtt:info  2020-04-19 14:37:29: MQTT publish: topic 'z2m/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
  zigbee-herdsman:controller:touchlink Set InterPAN channel to '11' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - interPanCtl - {"cmd":1,"data":[11]} +38s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,36,16,1,11,60] +38s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,16,0,117] +31s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,16,0,117] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 16 - [0] - 117 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - interPanCtl - {"status":0} +38s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - interPanCtl - {"cmd":2,"data":[12]} +14ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,36,16,2,12,56] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,16,0,117] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,16,0,117] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 16 - [0] - 117 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - interPanCtl - {"status":0} +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000ffff","destendpoint":254,"dstpanid":65535,"srcendpoint":12,"clusterid":4096,"transid":14,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[17,0,0,1,0,0,0,4,18]}} +12ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,29,36,2,2,255,255,0,0,0,0,0,0,254,255,255,12,0,16,14,0,30,9,0,17,0,0,1,0,0,0,4,18,196] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:touchlink Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms' +527ms
  zigbee-herdsman:controller:touchlink Restore InterPAN channel +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - interPanCtl - {"cmd":0,"data":[]} +501ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,1,36,16,0,53] +501ms
zigbee2mqtt:error 2020-04-19 14:37:36: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SRSP - AF - interPanCtl after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))

```

Koenkk, 2020-04-20:
Congratz with your son @IVI053 

I don't understand yet why this fails.
- Just to be sure, in the meantime TI released an update of their SDK, please try this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin
- Can you sniff the traffic on channel 11 while doing the touchlink reset: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

duesseldorferjung, 2020-04-23:
Wow!!! After spending hours and hours trying to factory reset my Hue Go (7146060PH) I finally tried the touchlink and it worked like a charm! Awesome work! Where is the buy me (you) a coffee link?

Koenkk, 2020-04-23:
@duesseldorferjung you can click the sponsor button on the top (https://github.com/koenkk/zigbee2mqtt). thanks!

mdhom, 2020-04-29:
I got the following error when trying to Touchlink-Reset my IKEA bulb (LED1732G11):
```
info  2020-04-29 09:43:50: Starting touchlink factory reset...
error 2020-04-29 09:44:10: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SREQ '--> AF - interPanCtl - {"cmd":1,"data":[23]}' failed with status '186' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58))
```
I've seen your response up top to run `rm -rf node_modules && npm install` in the zigbee2mqtt directory, but as i am running it within docker, i don't know where to execute that command?

My bridge config:
```
{"version":"1.11.0","commit":"31e5678","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"debug","permit_join":false}
```

Thanks for you help!!

Koenkk, 2020-04-29:
@mdhom can you try to stop zigbee2mqtt, replug the stick, start zigbee2mqtt and try again? `rm -rf node_modules && npm install` won't fix it.

mdhom, 2020-04-30:
Thanks, no error is shown anymore.. will try it with the IKEA bulb again.

aussybob, 2020-05-03:
@Koenkk for some reason I don't get a response when using the zigbee2mqtt/bridge/config/touchlink/factory_reset atleast I don't see anything in the log that the packet has been received. I can publish other topics to zigbee2mqtt and they work..
I am using home assistant and the hassio zigbee2mqtt add-on.
does this not work on the hassio addon?

zigbee2mqtt version 1.7.1 (commit #unknown)
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

aussybob, 2020-05-03:
when running the **zigbee2mqtt-edge** add-on the zigbee2mqtt/bridge/config/touchlink/factory_reset cmd works..

IVI053, 2020-05-11:
> * Just to be sure, in the meantime TI released an update of their SDK, please try this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Did not change. Still

```
zigbee2mqtt:error 2020-05-11 22:00:15: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SRSP - AF - interPanCtl after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/z2m/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

> * Can you sniff the traffic on channel 11 while doing the touchlink reset: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Will try this next and report my findings.

dchwilk, 2020-06-17:
Does this only work with CC? I'm trying a ConBee2 and get this error message:
```
Jun 17 14:29:20 x npm[2294]: zigbee2mqtt:info  2020-06-17 14:29:20: Starting touchlink factory reset...
Jun 17 14:29:20 x npm[2294]: zigbee2mqtt:info  2020-06-17 14:29:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
Jun 17 14:29:20 x npm[2294]: (node:2309) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'stack' of undefined
Jun 17 14:29:20 x npm[2294]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:355:103)
Jun 17 14:29:20 x npm[2294]:     at processTicksAndRejections (internal/process/task_queues.js:97:5)
Jun 17 14:29:20 x npm[2294]: (node:2309) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
```

Testet with 1.12.2-dev and 1.14.0-dev

Koenkk, 2020-06-17:
@dchwilk indeed not supported at the moment. You can ask the Conbee team to add support for it here: https://github.com/Koenkk/zigbee-herdsman/issues/72

fishpepper, 2020-06-23:
> Congratz with your son @IVI053
>
> I don't understand yet why this fails.
>
>     * Just to be sure, in the meantime TI released an update of their SDK, please try this firmware: [...]
I get the same error during touchlink reset. I used the following firmware:
https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200417.zip 

Debug log: https://pastebin.com/0USpF2bV

closed 2020-08-01 by stale[bot]

SimonMerrett, 2020-08-24:
` mosquitto_pub -d -t zigbee2mqtt/bridge/config/touchlink/factory_reset -m ""`
for anyone wishing to send the command from the command line using mosquitto. Perhaps @Koenkk might add to the docs on touchlink?

copystring, 2020-09-10:
Get a desk lap. Put bulb in desk lamp. Put desk lamp next to coordinator.

sshaikh, 2020-09-10:
Good idea but they're GU10s and I don't have a spare socket. I'll hoist a laptop!

sjorge, 2020-09-10:
I have this E27 one form ikea https://www.ikea.com/be/nl/p/skallran-tafellampvoet-donkergrijs-metaal-70420130/

I also have E27 -> E14 and E27 -> GU10 adaptors that I use with this, so far that covered all my lights except led panels, so those get the 6x on/off cycle

Fend0r, 2020-10-24:
I am getting this error (not mentioned so far):

Oct 25 01:01:17 homebridge npm[2604]: Zigbee2MQTT:info  2020-10-25 01:01:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"reset_started","meta":{"status":"started"},"type":"touchlink"}'
Oct 25 01:01:20 homebridge npm[2604]: Zigbee2MQTT:error 2020-10-25 01:01:20: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: Write for 'BUFFER' not available
Oct 25 01:01:20 homebridge npm[2604]:     at BuffaloZnp.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:259:19)
Oct 25 01:01:20 homebridge npm[2604]:     at ZpiObject.createPayloadBuffer (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zpiObject.js:87:21)
Oct 25 01:01:20 homebridge npm[2604]:     at ZpiObject.toUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zpiObject.js:38:29)
Oct 25 01:01:20 homebridge npm[2604]:     at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:279:34)
Oct 25 01:01:20 homebridge npm[2604]:     at Generator.next (<anonymous>)
Oct 25 01:01:20 homebridge npm[2604]:     at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
Oct 25 01:01:20 homebridge npm[2604]:     at new Promise (<anonymous>)
Oct 25 01:01:20 homebridge npm[2604]:     at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
Oct 25 01:01:20 homebridge npm[2604]:     at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:277:41)
Oct 25 01:01:20 homebridge npm[2604]:     at Queue.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:29

Any ideas?

spurgelaurels, 2020-11-10:
Bought a new ZZH, CC2651, running the 3.x.0 CC26X2R1_20201026 firmware. (Switching over from a CC2531 if that matters...)

Sengled and Xiaomi devices all pairing well, but when I run Touchlink to reset my Hues, the adaptor locks up with the below error and the system needs a reboot to continue. (Might only need a restart of Z2M and a re-plug of the adaptor, but my HASS box is pretty far away). 

Running latest edge branch of Z2M in HASS.

**touchlink scan logs**
```Zigbee2MQTT:info  2020-11-10 11:01:52: Start Touchlink scan
Zigbee2MQTT:error 2020-11-10 11:02:09: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:info  2020-11-10 11:02:09: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error"}'```

**Restarting Z2M logs**
```Zigbee2MQTT:info  2020-11-10 11:12:33: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-11-10 11:12:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:error 2020-11-10 11:12:52: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-10 11:12:52: Failed to start zigbee
Zigbee2MQTT:error 2020-11-10 11:12:52: Exiting...
Zigbee2MQTT:error 2020-11-10 11:12:52: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)```

spurgelaurels, 2020-11-10:
I found my Nortek stick buried in a box, so I'm going to use that on my laptop to reset these bulbs, but if you need any further debug information from my setup/situation, I'm happy to help.

gabric098, 2021-01-31:
> I found my Nortek stick buried in a box, so I'm going to use that on my laptop to reset these bulbs, but if you need any further debug information from my setup/situation, I'm happy to help.

Same error here @spurgelaurels . I'm running ZZH stick. Doesn't this stick support touchlink? I also have a spare CC2531 somewhere. Shall I use that one instead?

Thanks.
Gab

877dev, 2021-04-09:
I'm having the same issue, also with ZZH and Philips Hue bulb:

```
Zigbee2MQTT:info  2021-04-09 12:29:09: Start Touchlink factory reset of first found device
Zigbee2MQTT:error 2021-04-09 12:29:30: Request 'zigbee2mqtt/bridge/request/touchlink/factory_reset' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:info  2021-04-09 12:29:30: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/factory_reset', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error"}'
```

Morphy99, 2023-11-11:
I'm trying to reset some Philips Hue bulbs with Touchlink but every time I scan for devices it only finds bulbs in use by Z2M which I don't want to reset. I'm holding the bulb next to the coordinator (Sonoff Dongle + CC2652P). Tried powering on/off bulb restarting Z2M, unplugging Sonoff dongle. Where am I going wrong here?

[1 more reply]