Koenkk/zigbee2mqtt #2397: Can we expand the table on the per-device page to list what they support over reporting?

sjorge, 2019-11-23:
Can we expand the table on the per-device page to list what they support over reporting?

e.g. the Philips Hue bulbs only report genOnOff, Trådfri bulbs report genOnOff, genLevel and generm the color one.

I was hunting for alternatives to the Hue as the lack of reporting is a pain, and I am not happy with the Trådfri colors. Which basically resulted in me having to take a guess at which ones support reporting, I picked Innr which had good reviews color wise but it was impossible to find info on the reporting capabilities... luckily it reports everything.

But it would be great to have that info listed a wel!

sjorge, 2019-11-24:
Maybe just a separate page as this seems to be mostly the same between model on brands.

I am already collecting some data for my own use, just waiting on my E14 geldopto bulb and I'll post the little table here.

Koenkk, 2019-11-25:
This can easily be added to docgen, e.g. https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docgen/device_page_notes.js#L196 is an entry for all hue bulbs. Feel free to make a PR.

sjorge, 2019-11-27:
| Brand           | Support            |                    |                    |                    | Reporting          |                    |                     |                    | Change Color While Off |
| :---            | :---:              | :---:              | :---:              | :---:              | :---:              | :---:              | :---:               | :---:              | :---:                  |
|                 | On/Off             | Brighness          | Color              | Color Temp         | On/Off             | Brighness          | Color               | Color Temp         |                        |
| Philips Hue     | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :no_entry_sign:*   | :no_entry_sign:     | :no_entry_sign:    | :no_entry_sign:        | 
| Trådfri         | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :no_entry_sign:    | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:  | :no_entry_sign:    | :heavy_check_mark:     |
| Innr            | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark:  | :heavy_check_mark: | :no_entry_sign:        |
| GLEDOPTO        | :heavy_check_mark: | :heavy_check_mark: | :heavy_check_mark: | :no_entry_sign:    | :no_entry_sign:    | :no_entry_sign:    | :no_entry_sign:     | :no_entry_sign:    | :no_entry_sign:        |

* zigbee2mqtt will manual poll for change if a binding updates the bulb.

sjorge, 2019-11-27:
Maybe a seperate page with the overview is easier?
I tried a few hue and tradfri RGB bulbs and one Innr and GLEDOPTO... the firmware seems to be mostly consistant between within the same brand.

Koenkk, 2019-11-27:
Looks great, I think the best place is here: https://www.zigbee2mqtt.io/information/report.html

sjorge, 2019-11-27:
PR opened https://github.com/Koenkk/zigbee2mqtt.io/pull/169, saldy the checkmarks are github specific it seems.

closed 2020-01-26 by sjorge