Koenkk/zigbee2mqtt #2398: Timeout with Osram Flex

Root-DE, 2019-11-23:
# Bug Report
## What happened
Ive got a Timeout multiple times (70%) when i want to change some stats of an Osram Flex running. But Sometimes it work, therefore I dont know how to reproduce it. Furthermore if it succeded with the message …"r":255,"g":255,"b":"255"... the Flex isint complete white it is light blue (not such the Hue Bulb and -Strips, the rgb for them seems fine) but thats an other issue
## Debug Info
zigbee2mqtt version: 1.7.1
CC2531 firmware version:
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}
log_level: debug
https://pastebin.com/V5YrU8eR
Osram Flex in database.db
https://pastebin.com/DnPvpV97

Root-DE, 2019-11-25:
After unpair and pair the timeout failure did not occur so far, but there is a difference between the hue Strip and the Osram Flex, if you send nonvalid Data to hue like "color_temp" : 500 the bulbs change to the nearest, but if you use Osram Flex (Version is in last Post) you need to pick a value between [151,370] 151 and 370 are inclusive

Root-DE, 2019-11-27:
but to change the "color" with 
```
{
  "color": {
    "r": 0,
    "b": 128,
    "g": 0
  },
  "state": "on",
  "transition": 0.4
}
```
dont work complete sometimes yes, sometmies no

Koenkk, 2019-11-28:
Could it be a range issue? (https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html)

Root-DE, 2019-11-29:
No, it think its near enough, there are several Devices with a greater distance

closed 2020-02-04 by stale[bot]