Koenkk/zigbee2mqtt #2399: Support a new device - leaves the network when joining

cemonneau, 2019-11-24:
Hi,

I've been trying since a couple of days adding the support of a device from Legrand (Europe Manufacturer), and especially a roller shutter switch (https://www.legrand.fr/pro/catalogue/42566-version-celiane-with-netatmo/interrupteur-filaire-connecte-celiane-with-netatmo-pour-volet-roulant-blanc).

I know there is some issues like the #1994, but as it's not the same product & same errors. I prefered to open a new issue.

I manage to make the switch connect to my CC2531, but a soon as it joins, it throws a bunch of logs and then leaves the network. If I try to join again (by using the reset button on the switch), it then throws more errors : my current solution is to remove the database & clean the configuration.yml file as the device is still listed.
First attempt logs: https://pastebin.com/6H12D7ev
Second attempt logs: https://pastebin.com/a8kZNk8y

I tried with 2 firmwares : last one stable on 1.2 & 3.0 (as it seems the device is zigbee 3.0 I tried), the logs are identical. Zigbee2mqtt was the last dev version. The network only has the coordinator and that device.

My device.js contains the current following code:
```
{
    zigbeeModel: [" Shutter switch with neutral\u0000\u0000\u0000"],
    model: 'Shutter',
    vendor: 'Legrand',
    description: 'Test ', 
    supports: '',
    fromZigbee: [fz.cover_position_tilt], 
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
}
```

I've also gathered some info on Legrand (especially Netatmo products as it's the ones with Zigbee) from other githubs projects. The current device has not been updated (I dont have the gateway), some people have reported that after an update it may require a router between the devices & the coordinator. The update also adds more functionnalities on light switches (dimmer) and on outlets (power mettering), no information on the shutter switch yet.
There is a manufacterer specific cluster (0xfc01) and more info on this pull : https://github.com/dresden-elektronik/deconz-rest-plugin/pull/962/files

I'm a tech engineer and have been reading a lot on the zigbee protocol recently, I have some ground knowledge if necessary to help patch things if required.

Any help would be greatly appreciated, thanks !

Koenkk, 2019-11-25:
The device seems to pair successfully so that looks OK. I don't know yet why it almost immediately leaves the network. Perhaps sniffing could give some pointers: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

cemonneau, 2019-11-25:
I could not figure out why it would leave. The only thing that could explain it to me is the coordinator not sending a response to the 2/3 first requests of the device (type 'readResponse', cluster 'genBasic'). 
I'm not sure what I could add to respond to those requests.

I do not have the Legrand Gateway, but if I can't figure it out I may try to get one to sniff an "official" network.

Koenkk, 2019-11-26:
What does it try to read, can you share the wireshark logs?

cemonneau, 2019-11-27:
Ho, 

The request "type 'readResponse', cluster 'genBasic'" is in the log I provided above, and present 3 times before the device leaves the network.
```
zigbee2mqtt:debug 2019-11-16T21:18:23: Received Zigbee message from '0x00047400008b44a2', type 'readResponse', cluster 'genBasic', data '{"modelId":" Shutter switch with neutral\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2019-11-16T21:18:23: Received message from unsupported device with Zigbee model ' Shutter switch with neutral'
zigbee2mqtt:warn  2019-11-16T21:18:23: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

I did not tried yet the wireshark as I'm not sure it would be usefull so sniff between the CC2531 and the device. If you feel it would provide more usefull data I will flash one and sniff everything to report it back.

Koenkk, 2019-11-27:
This is expected, zigbee-herdsman reads information from the device (e.g. modelid, powersource, etc.). The `readResponse` are answers to these read requests. Probably not the reason why the device is leaving.

cemonneau, 2019-12-01:
Hi,

I got my hands on the official gateway, changed a CC2531 to be a sniffer, and learned a LOT of things. I'm planning to write something somewhere.
It has been a real pain to be able to link the shutter switch even to the official gateway as the problem was the same : join and then leave. It took me the whole day to find something to join, and it worked well after (I had to power off and on BOTH devices at the same time & initiate a permit join just after).

What i can resume:
- the gateway has a built in coordinator & router (probably the power outlet as it's a "double" device). Sometimes the network trames goes through the router, sometimes not.
- there is some useless but always nice cluster (identify (cluster 3) makes the LED blink in blue) 
- the LED stays green 15min when the configuration succeeds
- and some other data on reporting
- not all channels seems to be supported ([25;+[) & I haven't found a way to change the gateway channel
- wireless switches seems to be able to join the CC2561 network, but their commands are scene commands (at least for the one I tried), i will investigate more once the shutter works

I wasn't able to reproduce the join on the network made by the CC2531 BUT I found some things in the wireshark logs that could explain it (see screens linked, or PM me for logs)
1) the coordinator sends a "Confirm Key, SUCESS" packet
2) the coordinator sends a "Bind Request Src (cluster ID: 0x000f)", same for 0x0102, 0x0003

What is interesting is that when the switch wasn't able to join the network the "Bind Response, Status: Sucess" (after the Bind request) wasn't present, I may be linked.

I'm not sure what 1 is. 
Regarding 2, are those things configured with 
```
configure: async (device, coordinatorEndpoint) => {
       const endpoint = device.getEndpoint(1);
       await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
       await configureReporting.onOff(endpoint);
},
```
If it is the case, could you give me more info on what those commands are, I'm not 100% what they do. Can we use an hex number instead of 'genOnOff' (didn't found the dictionary)

Thanks !

Koenkk, 2019-12-03:
> What is interesting is that when the switch wasn't able to join the network the "Bind Response, Status: Sucess" (after the Bind request) wasn't present, I may be linked.

This is expected, when the device didn't join successfully it cannot respond to commands. For the other commands please refer to the zigbee documentation (should be described well there). But note that if it is even difficult to pair with the original gateway, this might require a lot of debugging to make it pair with the CC2531. Time wise it will probably be easier to buy an already supported devices (as it doesn't seem to be very zigbee "compliant").

sjorge, 2019-12-03:
If I am not mistaken they are similar to the soon to be released niko devices that use an enocean module... so we need green power support I think?

cemonneau, 2019-12-04:
Regarding Niko according to a friend working in that area it's not the same pairing process/code, but I haven't tested at all.

From what I saw during my tests the wireless switch might be green power BUT works with zigbee2mqtt without troubles (at least the pairing worked, after it sends scenes commands and I think zigbee2mqtt doesn't have the support, can you confirm @Koenkk ? ).

@Koenkk I'm ready to work on the integration as I already have those devices installed and well I'm not ready to replace them ^^. Zigate supports Legrand devices without troubles, so it's doable.
The main problem I currently have is just understanding the zigbee2mqtt code and doing the link with what it's supposed to do according to the specifications.

I think the problem with pairing with the official gateway is "only" a security in their gateway as it's explained (but badly) in their documentation, their devices seems to match the spefications, and the peoples working on it on the Zigate project didn't had a lot of troubles (it seems). I will improve my benchmark table to get a better environment to work with.

Is it possible to get more info on what part of the code should be doing (or does not support) 
1) "Confirm Key, SUCESS" packet
2) "Bind Request Src (cluster ID: 0x000f)" (I think it's the code I mentionned, didn't had the time to trace everything yet before doing some tests)

Koenkk, 2019-12-05:
1. This comes from the CC2531 firmware, not from zigbee2mqtt or herdmsan.
2. This one can be done by zigbee2mqtt, an example can be found here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L762 but just `genBasic` instead (as that is what is shown in your sniffer logs).

cemonneau, 2019-12-07:
Hi,

I'm back with great news : I nailed it and successfully integrated Legrand netatmo Switches.
There is some little things to change in zigbee2mqtt-herdsman as it was lacking things required for me to integrate it.

Short version, every powered devices from Legrand ask every other devices on the network since how long there are poweron, and if the value is too high, it leaves the network.
I think it's a security to force people to shutdown their electric installation when doing the installation.
The device sends a read attribute to the coordinator and it needs to respond, that's what I did.

I basically modified the controller.js(ts) file to send global read to their respective device handler (plugged on 'fromZigbee' currently). In my custom handler I sent the required 'readResponse' and it worked. I added some code to be able to transmit the original frame too (for the transactionSequenceNumber).
The zclFrameConverted.ts had to be a little bit modified too to allow the variables to not disapear when looping. 
The endpoint.ts too was modified : the readResponse was modified to allow custom attributes with type (similar to the write function).

I will do longer tests to ensure they are not dropping and then I will see to integrate the proper clusters.

How do you want to proceed @Koenkk  to expose the "global read" (read attribute in zigbee) to devices as it's the main thing required.
I can do a push but I thought you might prefer to handle this part.

For anyone reading my current CC2531 version is the last version with zigbee3.0, I will try later with the 1.2 version.

Koenkk, 2019-12-07:
Can you make a pr of your current changes? I will probably not immediately merge them but allows me to think about a proper solution.

cemonneau, 2019-12-07:
Will do this week-end on respective projects to allow you to see what's needed.

I can also report 2 switches are connected since 4 hours and stable with a CC2531 separated by 1m.
I also found the right values for the identify cluster (allows way more things like colors than the official app does), I'm on it get a nice support too.
The cluster closuresWindowCovering seems to work properly but I need to do more tests.
I'm still not sure about the binaryInput report I'm getting, weird. I will investigate.

Should I rename this thread Legrand Netatmo Support ?

Koenkk, 2019-12-08:
Yes you can rename it.

cemonneau, 2019-12-08:
@Koenkk I made the pull request to zigbee-herdsman and added an example of the code I'm currently using to test.

I can also report that I flashed the CC2531 back to the 1.2 version and it paired well too again, and was way easier than with the official gateway.

I also tried another switch that received an OTA update from the official gateway and it works too, and contrary to the Zigate (from what I red) it does not require another router on the network (yeah). The network was only the CC2531 and the switch and no troubles in 1 hour of tests.

cemonneau, 2019-12-09:
@romainalca I saw you were interested by this support on another thread, could you indicate me the products you have ? If I have them I will ensure they works.

Currently the support should be for 1 - shutters, 2 - wireless switch (classic or enter/leave, it seems there is a difference), 3 - "normal" switch.

I can speak french if necessary

cemonneau, 2019-12-09:
I ranned into something while doing some tests with wireless switch today.
They do not report their battery status/voltage by default. The only way to get it is to make a read call on the "deviceAnnounce" event. 
In the best world, it would be better to not call it at every announce to preserve the battery : it may be neglibeable, or not, not sure on the long term, but still. 
Is there a way (store ?) to make a counter and call it every X call ? I'm not sure about the philosophy you have or if we could make that per device dependent/modifiable (or a user setting ?).

I'm almost done with the support, but only working on 1 switch, I wasn't able to pair another switch, but it may be related to the device itself (that's the one allowing the permit join, it may not be possible to "unjoin" it from the official gateway :( )

cemonneau, 2019-12-15:
Okay so here is my initial commit.
I'm not creating a PR yet as I want to finish to check some attributes on my real installation (this test bench if powered off when I'm done so I'm not getting all values/reports).

Most devices have a "configure" button that sends an identify command to the coordinator.
I added a custom from, but the value stays when we look at the report in MQTT. Is there a way to make it disapear after X seconds ? I dont feel it should stay as it's not how it works on the official gateway : every devices LED turn green, and then off done by the coordinator sending an identify to everyone.

Here is the corresponding code.
https://github.com/cemonneau/zigbee-herdsman-converters/blob/legrand/converters/fromZigbee.js#L3220

Koenkk, 2019-12-15:
To avoid the value being cached, change `return {identify: 'configuration'};` to `return {action: 'identify'};`

cemonneau, 2019-12-17:
I hitted a second wall today, some products do not respect at all (it seems) the specification. I'm unable to get either a node descriptor or list endpoints or anything else. 
The affected products are the same as the ones not in the list of the zigate (https://zigate.fr/produits-legrand-compatibles-zigate/). 
I will try to see what I can do but I'm going to focus on the ones that works "out of the box" (wireless switch, shutter switch & power outlet) for now.

cemonneau, 2019-12-17:
I made some progress on 1 device : the wired light switch. I made it join but it was painfull (more).
It requires another router on the network, and every device from Legrand needs to be powered "at the same time" due to their security. But it also requires for the device to join to see a "permit join" while on (and within 10s/20s). On the official box when powered on & in join mode it spams "permit join" during the first 30s. 
I passed that part by sending a permit off & permit on to force a frame to be sent (the configuration.yml was on permit_join=true).
What's your idea on that part @Koenkk  : documentation ? or spam more permit joins ? 

Another product is searching for a device with the ability to deliver OTA (thanks to the sniffer), so unable to use it outside of the box.
The wired switch doesn't have every function available without OTA it seems (like dimming, even if it reports it). Peoples on the zigate forum talked about that, I still need to check the firmware version.

Koenkk, 2019-12-19:
@cemonneau let's for now at it to the docs.

cemonneau, 2019-12-22:
I'm back, and with "bad" news. 
I've been working on adding new devices support, and received an OTA from the manufacturer box.
Only 1 product was affected (shutter switch, the other wasn't working as it needed an OTA to work...), and ... it does not join anymore. 
I think I have the answer but couldn't find anything related in every project. I suspected it's in the zstack but I need your expertise on that @Koenkk [](url)
Please find in attachment an example of:
- a working join
- a "failed" one (on zigbee2mqtt)
- a working join on another product (not updated)

I believe the problem is the device not sending a data request after the association response, thus not receiving the transport key.
Is there a way to "force" that on our side (application) and not doing something on the zstack ?

I'm on the 1.2 coordinator version, haven't tested on 3.0 yet but I suspect the result will be the same.

Screen from the official gateway (working example) after OTA
[image]

Failed join (infinite loop) with zigbee2mqtt
[image]

Successfull join on zigbee2mqtt of another product
[image]

cemonneau, 2019-12-26:
Got a bit of time to play, and new results.
I was able to join the device after I upgraded to 3.0, nothing else to change. The diference was as mentioned above, the transport key was sent without the need of a "data request". I will try to flash back to 1.2 and see if the problem is still there.

In my recent commit I added the support of the "Master Switch" (Home/Leave button, and the center button put the network in permit join mode on the original gateway). There is a small (big?) thing that I still need to monitor : it seems that the specific command to prevent the leave is sent (sometimes) before the full pairing, thus the converter not beeing able to respond. A patch was needed to "deviceReceive.js" (easiest way i found) to send the answer BUT when I tried to repair twice after the full support it worked without that patch, not sure why.
That product wont work out of the box as it requires an OTA (or at least an answer to an OTA request), after pairing to the official box it works.

I just need to rebase and I think an initial support could be reviewed & merged. More products will be added once(if) I get them.

cemonneau, 2019-12-26:
I had a problem with the genPollCtrl cluster. The master switch is by default on fastPoll (draining a lot the battery, lost 20%(3v battery) in 1hour of test). I initially wanted to respond to a frame sent by the device, but I was unable to send the reply (not sure why as the logs where showing no errors).
I ended up sending another command (and not responding).
It works but it's not as clean as I would like.
See https://github.com/cemonneau/zigbee-herdsman-converters/blob/legrand/devices.js#L6519 for the part I'm talking about

cemonneau, 2020-01-03:
I'm back with more feedback.
I migrated my network (test) to the new CC1352P-2 (light bulbs/legrand shutter switch/legrand wireless switch).
All those products worked with 0 troubles.

The one that I wasn't able to pair is the "master switch", same problem as above. 
It sends a read command BEFORE pairing, as the converter is not yet associated it cannot answer, and thus is leaving. My patch above works.
Is there a way to implement the answer to that command (legrand specific/security) at a higher level ?
@Koenkk

Koenkk, 2020-01-07:
@cemonneau I guess you are talking about this code? https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L441    

What about adding a general onEvent method which is always called when a zigbee message is received (event when it does not have a modelID yet).

cemonneau, 2020-01-07:
@Koenkk yes that's exactly that part of the code, I should have mentioned it.
The idea of a global onEvent seems to be the best I think (most generic if required for other specific things). Maybe it could be filtered (or  exposed in the event) the manufacturerCode(of the frame) to ensure the right response "in case of".

Koenkk, 2020-01-08:
@cemonneau done, see https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/index.js#L35

alexisfradel, 2020-01-08:
Hello all,

This topic is really insteresting even if a bit too technical for me :-(

I have switched my branch to dev but i'm still not able to pair my Legrand Netatmo switch.
The device is listed on the MQTT topic but then the LED goes to red+blue.

Would you have a kind of a procedure to pair these devices?

In any case, thanks a lot for all your work.
Alexis

cemonneau, 2020-01-09:
@Koenkk perfect, thanks ! Will test and migrate if working in the next patch I'm preparing

cemonneau, 2020-01-09:
Hi @alexisfradel,

What kind of device do you have as Switch ? (wireless, wired, double button ?).
In theory the procedure is start zigbee2mqtt, permit join, power on the device (it needs to be power off due to a security couting the seconds since on). After that the LED should become green after 5/10s.
The zigbee network MUST be on the channel 11 too (haven't been able to join on any other sadly).
A device on Zigbee 3.0 seems to reduce bugs (at least one).

If you could provide your logs too that would be perfect !

I will probably work on the documentation this week, it's on my todo.

alexisfradel, 2020-01-09:
Hello @cemonneau 

Thank you for your reply.

I have micromodules. I think that my mistake is that I havent swith them off before pairing.
Reading the thread, I thought that it was not necessary after your last improvements.
I will try again later today.

Additional question: how do we define the Zigbee channel?

Best regards

cemonneau, 2020-01-09:
@alexisfradel 
I do not have Micromodules so they wont currently stay in the network (paired) due to a security. It will be fixed with a new commit I'm working on (currently we need to add a specific code for each Legrand device, that won't be the case after). 
I will need your zigbee2mqtt log to add them if possible (you can PM the logs me if you want).

The channel can be debine in the configuration.yml, it's by default on 11.

alexisfradel, 2020-01-09:
@cemonneau 
Indeed I have noticed in the logs that the micromodules are not supported.
Do you want me to do some test to confirm that what you plan on doing will work?
I can also try to modify some files on my RPI in order to try and find the specific code.
Could you please just point me in the right direction? Where is it done for the switch for example?

Finally, please find below my logs:
zigbee2mqtt:debug 2020-01-09 18:19:51: Received Zigbee message from '0x00047400001f05a6', type 'readResponse', cluster 'genBasic', data '{"modelId":" Micromodule switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-09 18:19:51: Received message from unsupported device with Zigbee model ' Micromodule switch'
zigbee2mqtt:warn  2020-01-09 18:19:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +10ms
  zigbee-herdsman:controller:device Interview - got 'modelId,manufacturerName,powerSource' for device '0x00047400001f05a6' +89ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":15095,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":2,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,3,0,0,0,1,0,2,0]}} +85ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,247,58,1,1,0,0,2,0,30,9,16,3,0,0,0,1,0,2,0,254] +85ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +85ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +31ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,38,68,129,0,0,0,0,247,58,1,1,0,34,0,96,151,151,0,0,18,24,3,1,0,0,0,32,2,1,0,0,32,0,2,0,0,32,66,247,58,19,217] +33ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,38,68,129,0,0,0,0,247,58,1,1,0,34,0,96,151,151,0,0,18,24,3,1,0,0,0,32,2,1,0,0,32,0,2,0,0,32,66,247,58,19,217] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 38 - 2 - 4 - 129 - [0,0,0,0,247,58,1,1,0,34,0,96,151,151,0,0,18,24,3,1,0,0,0,32,2,1,0,0,32,0,2,0,0,32,66,247,58,19] - 217 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":15095,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":34,"securityuse":0,"timestamp":9934688,"transseqnumber":0,"len":18,"data":{"type":"Buffer","data":[24,3,1,0,0,0,32,2,1,0,0,32,0,2,0,0,32,66]}} +34ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":32,"attrData":2},{"attrId":1,"status":0,"dataType":32,"attrData":0},{"attrId":2,"status":0,"dataType":32,"attrData":66}],"Cluster":{"ID":0,"attributes":{"zclVersion":{"ID":0,"type":32,"name":"zclVersion"},"appVersion":{"ID":1,"type":32,"name":"appVersion"},"stackVersion":{"ID":2,"type":32,"name":"stackVersion"},"hwVersion":{"ID":3,"type":32,"name":"hwVersion"},"manufacturerName":{"ID":4,"type":66,"name":"manufacturerName"},"modelId":{"ID":5,"type":66,"name":"modelId"},"dateCode":{"ID":6,"type":66,"name":"dateCode"},"powerSource":{"ID":7,"type":48,"name":"powerSource"},"appProfileVersion":{"ID":8,"type":48,"name":"appProfileVersion"},"swBuildId":{"ID":16384,"type":66,"name":"swBuildId"},"locationDesc":{"ID":16,"type":66,"name":"locationDesc"},"physicalEnv":{"ID":17,"type":48,"name":"physicalEnv"},"deviceEnabled":{"ID":18,"type":16,"name":"deviceEnabled"},"alarmMask":{"ID":19,"type":24,"name":"alarmMask"},"disableLocalConfig":{"ID":20,"type":24,"name":"disableLocalConfig"}},"name":"genBasic","commands":{"resetFactDefault":{"ID":0,"parameters":[],"name":"resetFactDefault"}},"commandsResponse":{}}},"address":15095,"endpoint":1,"linkquality":34,"groupID":0}' +63ms
zigbee2mqtt:debug 2020-01-09 18:19:51: Received Zigbee message from '0x00047400001f05a6', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":0,"stackVersion":66}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-09 18:19:51: Received message from unsupported device with Zigbee model ' Micromodule switch'
zigbee2mqtt:warn  2020-01-09 18:19:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:controller:device Interview - got 'zclVersion,appVersion,stackVersion' for device '0x00047400001f05a6' +60ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,0,4,42,239,1,1,0,73,0,244,103,158,0,0,8,24,218,10,0,0,33,7,0,42,239,29,98] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4,42,239,1,1,0,73,0,244,103,158,0,0,8,24,218,10,0,0,33,7,0,42,239,29,98] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,0,4,42,239,1,1,0,73,0,244,103,158,0,0,8,24,218,10,0,0,33,7,0,42,239,29] - 98 +1ms

cemonneau, 2020-01-09:
@Koenkk just tested (and it works perfectly) & moved the code to the onEvent (see https://github.com/cemonneau/zigbee-herdsman-converters/commit/24c4337746e3bcdc6836b37aa3f1cddd73b5f4ff#diff-168726dbe96b3ce427e7fedce31bb0bcR41) BUT would it be possible to add manufacturerCode (sometimes called manufacturerID) somewhere to filter messages ? Currently it's unavailable is the device is not paired. On my version I added it to meta (in data) taken from the Frame Header

cemonneau, 2020-01-09:
@alexisfradel can you please fill me on some details on the device:
- does it has a physical LED ?
- do you know if there is a dimming functionnality ?
- could you give me the line in data/database.db matching the device ? (to check the exposed functionalities)
I made an experimental support on my repository. If you want to test you need to apply all the modification to the respective files made on https://github.com/Koenkk/zigbee-herdsman-converters/pull/891/files (the code is based on the latest dev branch).

alexisfradel, 2020-01-09:
Hello @cemonneau 

I just saw that you included a first draft for the micromodule. I'm currently downloading it on my RPI to test.
I will keep you updated :-)

Regarding the information you have requested:
- it has a physical LED (colors: red, blue and green)
- on my Legrand (Home +) there is only on/off
- here is the entry in my database.db file:
{
  "id": 11,
  "type": "Router",
  "ieeeAddr": "0x00047400001f05a6",
  "nwkAddr": 19841,
  "manufId": 4129,
  "manufName": " Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
  "powerSource": "Mains (single phase)",
  "modelId": " Micromodule switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 256,
      "inClusterList": [
        0,
        3,
        4,
        6,
        5,
        15,
        64513
      ],
      "outClusterList": [
        0,
        64513,
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": " Micromodule switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
            "manufacturerName": " Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
            "powerSource": 1,
            "zclVersion": 2,
            "appVersion": 0,
            "stackVersion": 66,
            "hwVersion": 21,
            "dateCode": " \u000020190529\u0000\u0000\u0000\u0000\u0000",
            "swBuildId": "001c"
          }
        }
      },
      "binds": [
        
      ]
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [
        
      ],
      "outClusterList": [
        33
      ],
      "clusters": {
        
      },
      "binds": [
        
      ]
    }
  },
  "appVersion": 0,
  "stackVersion": 66,
  "hwVersion": 21,
  "dateCode": " \u000020190529\u0000\u0000\u0000\u0000\u0000",
  "swBuildId": "001c",
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {
    
  }

cemonneau, 2020-01-09:
@alexisfradel can you please edit your post (i think it broke the log) and add ``` before and after it ?
it will do something like
```
some log
```

alexisfradel, 2020-01-09:
I have tried to include your latest version and here are the new logs
We can see that the device joined the network and then leaved. 

Hope that helps:

> zigbee-herdsman:controller:log Device announce '0x00047400001f05a6' +21s
  zigbee-herdsman:controller:log Device '0x00047400001f05a6' announced with new networkAddress '15095' +0ms
zigbee2mqtt:debug 2020-01-09 22:35:59: Device '0x00047400001f05a6' announced itself
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +10ms
zigbee2mqtt:info  2020-01-09 22:35:59: Configuring '0x00047400001f05a6'
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":19841,"srcaddr":"0x00047400001f05a6","srcendpoint":1,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b00193e523a","dstendpoint":1} +2m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,129,77,166,5,31,0,0,116,4,0,1,6,0,3,58,82,62,25,0,75,18,0,1,0] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
zigbee2mqtt:error 2020-01-09 22:36:09: Failed to configure '0x00047400001f05a6', attempt 2 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7))
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,47,223,1,1,0,65,0,28,218,99,0,0,7,24,192,10,0,0,16,0,47,223,29,228] +13s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,47,223,1,1,0,65,0,28,218,99,0,0,7,24,192,10,0,0,16,0,47,223,29,228] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,47,223,1,1,0,65,0,28,218,99,0,0,7,24,192,10,0,0,16,0,47,223,29] - 228 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":57135,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":6543900,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,192,10,0,0,16,0]}} +13s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":192,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":"genOnOff","commands":{"off":{"ID":0,"parameters":[],"name":"off"},"on":{"ID":1,"parameters":[],"name":"on"},"toggle":{"ID":2,"parameters":[],"name":"toggle"},"offWithEffect":{"ID":64,"parameters":[{"name":"effectid","type":32},{"name":"effectvariant","type":32}],"name":"offWithEffect"},"onWithRecallGlobalScene":{"ID":65,"parameters":[],"name":"onWithRecallGlobalScene"},"onWithTimedOff":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"commandsResponse":{}}},"address":57135,"endpoint":1,"linkquality":65,"groupID":0}' +13s
zigbee2mqtt:debug 2020-01-09 22:36:12: Received Zigbee message from '0x00158d0003f3af3f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-09 22:36:12: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f3af3f', payload '{"battery":100,"voltage":3015,"contact":true,"linkquality":65}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,171,122,166,5,31,0,0,116,4,0,0,0,158] +33s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,171,122,166,5,31,0,0,116,4,0,0,0,158] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [171,122,166,5,31,0,0,116,4,0,0,0] - 158 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":31403,"extaddr":"0x00047400001f05a6","parentaddr":0} +33s
  zigbee-herdsman:controller:log Device '0x00047400001f05a6' joined +33s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x00047400001f05a6' accepted by handler +1ms
  zigbee-herdsman:controller:log Device '0x00047400001f05a6' is already in database with different networkAddress, updating networkAddress +0ms
  zigbee-herdsman:controller:log Not interviewing '0x00047400001f05a6', completed 'true', in progress 'false' +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,171,122,171,122,166,5,31,0,0,116,4,0,142,203] +739ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,171,122,171,122,166,5,31,0,0,116,4,0,142,203] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [171,122,171,122,166,5,31,0,0,116,4,0,142] - 203 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":31403,"nwkaddr":31403,"ieeeaddr":"0x00047400001f05a6","capabilities":142} +742ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,201,171,122,166,5,31,0,0,116,4,0,0,0,0,156] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 201 - [171,122,166,5,31,0,0,116,4,0,0,0,0] - 156 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":31403,"extaddr":"0x00047400001f05a6","request":0,"removechildren":0,"rejoin":0} +10s
  zigbee-herdsman:controller:log Device leave '0x00047400001f05a6' +10s
  zigbee-herdsman:controller:log Removing device from database '0x00047400001f05a6' +1ms

cemonneau, 2020-01-09:
@alexisfradel  Are you sure to have updated the dependencies ? (npm update). it requires the latest version of everything (zigbee2mqtt, zigbee-herdsman & zigbee-herdsman-converters).
Can you edit the log present in https://github.com/Koenkk/zigbee2mqtt/issues/2399#issuecomment-572766394 ? (I'm almost sure it's missing some critical parts as "inClusterList")

alexisfradel, 2020-01-09:
No i haven't run nom update but rather manually updated the files in the module folder.
i'm currently doing it and i let you know.

alexisfradel, 2020-01-09:
Where i'm not fully sure if i will pull your latest changes it's because i have only cloned the zigbee2mqtt repository.
So i have switched to the DEV branch on this repo and then launch the npm update.
To be honest, i'm not familiar with npm :-(

cemonneau, 2020-01-09:
npm is basically a dependency manager. when you switched to the dev branch if you did not run "npm update" it didn't updated/upgraded dependencies, thus 99% sure "breaking" some of the stuff required for the latest patch I made

alexisfradel, 2020-01-09:
@cemonneau 
I'm probably doing something wrong since, when i'm doing 'npm update', i do not get your latest devices.js file (i cannot find the micromodule section).

Here is the procedure I do:
- go to /opt/zigbee2mqtt folder
- rm -rf node_modules => to remove the old modules
- npm update

Thank you again for your support

PS: maybe there is another communication method that could be easier for you?

cemonneau, 2020-01-11:
@alexisfradel just added you to a private projet to be able to talk / exchange more data

alexisfradel, 2020-01-12:
@cemonneau I have joined the project. Thanks
I will probably have some time tonight to do some additional tests.
I will send you my log files and findings.

ozer59, 2020-01-12:
Hello 
@cemonneau  thank's a lot for support shutter with neutral works perfect for me auto integration in home assistant too
i have 1 Legrand connected power outlet works but no auto integration in home assistant 
i have 2 legrand micromodule works but no auto integration in home assistant 
i have 1 067723 same as 067773 Wireless remote switch works after change model in devices.js  but no auto integration in home assistant 
and 1 067646 pairing ok but not work can you help me ? 
and how to process for home assistant auto integration homeassistant.js 

thank's in advance 
Soory for my poor english

sjorge, 2020-01-12:
@ozer59 getting them working in zigbee2mqtt and having the discovered by ha are two separate things.

They need to be added to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L474 for ha to discover them.

cemonneau, 2020-01-12:
Thanks for tour feedback @ozer59 
HA support is planned for this week, I wanted to have a working integration before doing it. I will probably write the documentation this week too.
Can you give me the modeled you added for 067723 please ? 
Regarding 067646 did it joined ? If yes, can you give me the line matching in data/database.db?
Just to know (simple curiosity) where did you get those product (France? Online. I have some troubles to find some of those products)

cemonneau, 2020-01-12:
@ozer59 just realized you said the shutter works in HA ?! Never saw it appears/recognized and I didn't added the support. Maybe someone did it but I didn't saw the commit

ozer59, 2020-01-12:
with homeassistant.js (dev branch legrand ) shutter works in HA 
for 067723 i use 067773  is the but with another color from dev branch legrand 
i am french electrician I buy from my local dealer if you want i can help you to get those product 
thank's a lot 

for 067646:
{"id":2,"type":"EndDevice","ieeeAddr":"0x0004740000892ee5","nwkAddr":28210,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Battery","modelId":" Shutters central remote switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":515,"inClusterList":[0,3,15,32,1],"outClusterList":[3,258,0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Shutters central remote switch","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":3,"zclVersion":2,"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020171221\u0000\u0000\u0000\u0000\u0000","swBuildId":"012\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000��!\u0002\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u000e�\u0014\u0000 \u0001\u0000\u0000\u00008\u0007\u0003\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}}},"binds":[]}},"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020171221\u0000\u0000\u0000\u0000\u0000","swBuildId":"012\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000��!\u0002\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u000e�\u0014\u0000 \u0001\u0000\u0000\u00008\u0007\u0003\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1578859033613}

ozer59, 2020-01-12:
i have this in my homeassistant.js but power consumption dont work in ha for micromodule and connected power outlet 

    '067776': [cfg.cover_position],
    '067723': [cfg.sensor_action, cfg.sensor_battery],
    '067771': [cfg.switch],
    '064873': [cfg.sensor_action],
	'064888': [cfg.light_brightness, cfg.sensor_power],
    'K3004C': [cfg.switch],
	'067775': [cfg.switch, cfg.sensor_power],

cemonneau, 2020-01-13:
@ozer59 
Regarding 067646 could you add to devices.js the following code, subscribe to everything in Mqtt and see if the actions are properly reported ?
```
{
        zigbeeModel: [
            ' Shutters central remote switch',
        ],
        model: '067646',
        vendor: 'Legrand',
        description: 'Wireless shutter switch',
        supports: 'action',
        fromZigbee: [
            fz.identify,
            fz.ignore_basic_report,
            fz.cover_position_tilt,
            fz.battery_3V
        ],
        toZigbee: [],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        },
        onEvent: async (type, data, device, options) => {
            await legrand.read_initial_battery_state(type, data, device);

            if (data.type === 'commandCheckin' && data.cluster === 'genPollCtrl') {
                // TODO current solution is a work around, it would be cleaner to answer to the request
                const endpoint = device.getEndpoint(1);
                const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
                /* await endpoint.command('genPollCtrl', 'checkinRsp', {
                    startfastpolling: false,
                    fastpolltimeout: 0,
                }, {
                    transactionSequenceNumber:data.meta.zclTransactionSequenceNumber,
                    manufacturerCode: 0x1021, disableDefaultResponse: true
                }); */
                await endpoint.command('genPollCtrl', 'fastPollStop', {}, options);
            }
        },
    },
```
(I didn't bought this product as I prefered to use wireless light switch or wireless home/away switch for my shutters).

Regarding 067723 modifying to 067773 is useless, only the modelId matters. I took the 067723 reference as every paper doc Legrand has sends back to this one (that's how i decide which reference I use when they have variants).

Thanks for the home assistant code, I will take a look later and add power consumption.
I dont think the microcromodule has power consumption feature though (i can check if you send me the line in database.db to see the exposed features).

NB: I can speak french if you have some troubles to translate some sentences (but prefer english as this project is global).

cemonneau, 2020-01-13:
@ozer59 here is my corrected homeassistant.js (I will make a PR once your shutter remote works).
I tested on the power outlet and it reports the power properly (in watts). 
067646 is empty as I'm waiting your feedback.
```
    '067776': [cfg.cover_position],
    '067773': [cfg.sensor_action, cfg.sensor_battery],
    '067771': [cfg.switch],
    '067775': [cfg.switch, cfg.sensor_power],
    '064888': [cfg.switch],
    '064873': [cfg.sensor_action, cfg.sensor_battery],
    '067646': [],
    'K3004C': [cfg.switch],
```

ozer59, 2020-01-13:
for 067646 actions are properly reported thank you 
but 067775 power outlet not reports power consuption strange maybe need update 
yes microcromodule has power consumption feature but brightness no ? 

info db connected outlet 
{"id":9,"type":"Router","ieeeAddr":"0x0004740000096589","nwkAddr":18823,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,6,5,64513],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020180115\u0000\u0000\u0000\u0000\u0000","swBuildId":"00e\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000��!\u0002\u0001\u0000\u0000\u0000\u0000\u0001\u0001u\u0010\u0000 \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"haElectricalMeasurement":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdae","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdae","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdae","endpointID":1}]}},"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020180115\u0000\u0000\u0000\u0000\u0000","swBuildId":"00e\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000��!\u0002\u0001\u0000\u0000\u0000\u0000\u0001\u0001u\u0010\u0000 \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1,"configured":3},"lastSeen":1578932223779}

info db micromodule 
{"id":5,"type":"Router","ieeeAddr":"0x00047400001d50f1","nwkAddr":11209,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Micromodule switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,6,5,15],"outClusterList":[0,25],"clusters":{"genBasic":{"attributes":{"modelId":" Micromodule switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020171026\u0000\u0000\u0000\u0000\u0000","swBuildId":"006\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000��!\u0002\u0001\u0000\u0000\u0000\u0000\u0001\u0001�\u0010\u0000 \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"genBinaryInput":{"attributes":{"presentValue":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdae","endpointID":1},{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5fdae","endpointID":1}]}},"appVersion":0,"stackVersion":64,"hwVersion":1,"dateCode":" \u000020171026\u0000\u0000\u0000\u0000\u0000","swBuildId":"006\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000��!\u0002\u0001\u0000\u0000\u0000\u0000\u0001\u0001�\u0010\u0000 \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1,"configured":2},"lastSeen":1578935383349}

cemonneau, 2020-01-13:
@ozer59 can you update your homeassistant.js and use the following for the wireless shutter switch : 
```
    '067646': [cfg.cover_position, cfg.sensor_battery],
```
It should be good with that.

On your outlet it might require an update yes (I didn't updated mine/never connected it to the official gateway even). The cluster required for electrical measurement is the 2820, but when we look at your cluster(cluster=exposed functionnality) (from database.db, see "inClusterList":[0,3,4,6,5,64513],"outClusterList":[0,64513,25] ) it's not in it. Mine reports "inClusterList":[0,3,4,6,5,64513,2820],"outClusterList":[0,64513,25]. However, your dateCode (fimware version I think) is 20180115, mine is 2018-07-18. An update will probably fix that.
Same for the micromodule, the power measurement is not reported, an update might be required. Please try, remove the module from the database.db file & pair it again after the upgrade. We will then see if it reports the cluster.  It does not have the brightness feature too (maybe after OTA).

ozer59, 2020-01-13:
when i try to use i have this message 
and no reaction in ha 

debug 2020-01-13 21:01:50: Device '0x0004740000892ee5' announced itself
debug 2020-01-13 21:01:50: Received Zigbee message from '0x0004740000892ee5', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID 0
debug 2020-01-13 21:01:50: No converter available for '067646' with cluster 'closuresWindowCovering' and type 'commandDownClose' and data '{}'
debug 2020-01-13 21:01:50: Received Zigbee message from '0x0004740000892ee5', type 'raw', cluster 'manuSpecificLegrandDevices', data '{"type":"Buffer","data":[21,33,16,0,5,229,46,137,0,0,116,4,0]}' from endpoint 1 with groupID 0
debug 2020-01-13 21:01:50: No converter available for '067646' with cluster 'manuSpecificLegrandDevices' and type 'raw' and data '{"type":"Buffer","data":[21,33,16,0,5,229,46,137,0,0,116,4,0]}'
debug 2020-01-13 21:01:51: Received Zigbee message from '0x0004740000892ee5', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
debug 2020-01-13 21:01:51: No converter available for '067646' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":1}'
debug 2020-01-13 21:01:51: Received Zigbee message from '0x0004740000892ee5', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
info  2020-01-13 21:01:51: MQTT publish: topic 'zigbee2mqtt/0x0004740000892ee5', payload '{"battery":"100.00","voltage":3000,"linkquality":96}'

cemonneau, 2020-01-14:
@ozer59 You should have that code in your devices.js
```
    {
        zigbeeModel: [
            ' Shutters central remote switch',
        ],
        model: '067646',
        vendor: 'Legrand',
        description: 'Wireless shutter switch',
        supports: 'action',
        fromZigbee: [
            fz.identify,
            fz.ignore_basic_report,
            fz.cover_position_tilt,
            fz.battery_3V
        ],
        toZigbee: [],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        },
        onEvent: async (type, data, device, options) => {
            await legrand.read_initial_battery_state(type, data, device);

            if (data.type === 'commandCheckin' && data.cluster === 'genPollCtrl') {
                // TODO current solution is a work around, it would be cleaner to answer to the request
                const endpoint = device.getEndpoint(1);
                const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
                /* await endpoint.command('genPollCtrl', 'checkinRsp', {
                    startfastpolling: false,
                    fastpolltimeout: 0,
                }, {
                    transactionSequenceNumber:data.meta.zclTransactionSequenceNumber,
                    manufacturerCode: 0x1021, disableDefaultResponse: true
                }); */
                await endpoint.command('genPollCtrl', 'fastPollStop', {}, options);
            }
        },
    },
```

TRuHa83, 2020-01-18:
Hi @cemonneau.

I've been using Legrand's Valena Next series without a gateway integrated with zigbee2mqtt for a long time thanks to the great work of @Koenkk.

I love seeing people also interested in the project.

I currently have in use:
    - 2 Socket (7 418 11)
    - 2 Switch (7 418 10)
    - 1 Micromodule (0 648 88)
    - 1 Wireless switch (7 418 13)

Everything working except the wireless switch. Not counting that I don't get socket consumption data or brightness control on the switch.

If I can help with something, you just have to ask for it.
I will be aware of the progress, thank you very much.

[146 more replies]

closed 2020-09-05 by github-actions[bot]