Koenkk/zigbee2mqtt #2401: Problem heiman

kwetnico, 2019-11-24:
Hello i use zigbee2mqtt cc2531 with raspberry pi 4 and Jeedom. 

I have 15 devices to connect on my cc2531 , you are one that works very very well.  However I bought a siren for my brand alarm Heiman HS2WD-E , it is well recognized I have the batteries, voltages, status.  But impossible to get off the alarm and the light .. I have as indicated here: https://www.zigbee2mqtt.io/devices/HS2WD-E.html passed the command with mosquitto_pub but it does not recognize the syntax. :
The alarm can be trigged by publishing to : 

mosquitto_pub -t zigbee2mqtt/siren/set-m {"warning": {"duration": 60, "mode": "emergency", "strobe": true}}

 Could you give me the exact syntax with mosquitto?  Thank you

Root-DE, 2019-11-24:
Hey i think you make 2 Mistakes, at first you need to use …../Sirene/set -m …
(you dont have the space between set and "-m")
and the second ist, if you use publish via console, you need to escape the Quotes therfore you should try:
mosquitto_pub -t zigbee2mqtt/sirene/set -m "{\"warning\": {\"duration\": 60, \"mode\": \"emergency\", \"strobe\": true}}"

kwetnico, 2019-11-24:
> Hey i think you make 2 Mistakes, at first you need to use …../sirene/set -m …
> (you dont have the space between set and "-m")
> and the second is, if you use publish via console, you need to escape the Quotes therfore you should use:
> mosquitto_pub -t zigbee2mqtt/sirene/set -m "{\"warning\": {\"duration\": 60, \"mode\": \"emergency\", \"strobe\": true}}"

hello,

oh yes !!! 
thank you very much this just works (I have no ear lol)
By cons the light does not light I just the siren ..

Now how can I integrate this command in the Jeedom application if you know?

Thank you

Root-DE, 2019-11-24:
No Problem, but I dont know about the Integration in Jeedom

But maybe this will help you
https://diyprojects.io/jeedom-mqtt-associate-diy-iot-dashboard/#.Xdqa-0BFxoI

closed 2020-01-31 by stale[bot]