Koenkk/zigbee2mqtt #2402: Innr RB 250 C publishes invalid payload

sjorge, 2019-11-24:
# Bug Report
## What happened
My Innr RB 250 C (fw 2.1) seems to publish an invalid payload for color

`{"x":0.4085,"y":0.3894,"saturation":59.8},"color_mode":1}`

Looking at the zigbee spec:
```
5.2.2.2.1.1 CurrentHue Attribute
The CurrentHue attribute contains the current hue value of the light. It is updated
as fast as practical during commands that change the hue.
The hue in degrees shall be related to the CurrentHue attribute by the relationship
Hue = CurrentHue x 360 / 254 (CurrentHue in the range 0 - 254 inclusive)
If this attribute is implemented then the CurrentSaturation and ColorMode
attributes shall also be implemented.

5.2.2.2.1.2 CurrentSaturation Attribute
The CurrentSaturation attribute holds the current saturation value of the light. It is
updated as fast as practical during commands that change the saturation.
The saturation shall be related to the CurrentSaturation attribute by the
relationship
Saturation = CurrentSaturation/254 (CurrentSaturation in the range 0 - 254 inclusive)
If this attribute is implemented then the CurrentHue and ColorMode attributes
shall also be implemented.
```

## What did you expect to happen
It's probably the firmware at fault here, but I don't think we should publish this incomplete information if we detect it.  

## How to reproduce it (minimal and precise)
It happend after the first color change after pairing, I tried both setting via color_xy and via color_temp and it results in the same updated payload (saturation gets update usually), color_mode does correctly reflect 1 or 2 depending on the methode used to change the color.

I cannot get it to color_mode 0 which would be hue/sat, which makes me think the firmware on the bulb is incorrect.

## Debug Info
zigbee2mqtt version: 1.7.0+dev
CC253X firmware version:

Koenkk, 2019-11-25:
I don't understand the problem yet (but I've also never looked seriously into `color_mode` and `saturation`.). The converter logic can be found here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L819. Values are taken straight from the device.

sjorge, 2019-11-25:
As far as I understand the specifications, the device needs to provide hue **and** saturation (color_mode = 0x00), but it looks like this bulb only provides saturation... so no hue... so it's not useful at all :(

I also tried setting hue + sat.... to see if it would switch to color_mode 0x00 and output both values... but it does work as there are no convertors.

Koenkk, 2019-11-25:
Probably that is missing indeed, please try adding it.

sjorge, 2019-11-25:
Will give it a go next time I am at my computer.

sjorge, 2019-11-25:
There is a 'currentHue' value in the payload!

```zigbee2mqtt:info  2019-11-25 20:54:19: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":31,"brightness":254,"color_temp":555,"color":{"x":0.6818,"y":0.2907,"saturation":100,"hue":1.4},"color_mode":1}'```

I get that result by adding this to the block

```

                if (msg.data['currentHue']) {
                    result.color.hue = precisionRound(msg.data['currentHue'] / (65535 / 360), 1);
                }
```

However I am not sure the calculation is correct, the above should be red

sjorge, 2019-11-25:
```
The hue in degrees shall be related to the CurrentHue attribute by the relationship
Hue = CurrentHue x 360 / 254 (CurrentHue in the range 0 - 254 inclusive)
```

I will try with the math from the spec... but now the bulb dropped form the network :/

sjorge, 2019-11-25:
The good news is that the currentHue data is present! But I can't seem to get a correct value out of it....

```
5.2.2.2.1.1 CurrentHue Attribute
The CurrentHue attribute contains the current hue value of the light. It is updated
as fast as practical during commands that change the hue.
The hue in degrees shall be related to the CurrentHue attribute by the relationship
Hue = CurrentHue x 360 / 254 (CurrentHue in the range 0 - 254 inclusive)
If this attribute is implemented then the CurrentSaturation and ColorMode
attributes shall also be implemented.
5.2.2.2.1.2 CurrentSaturation Attribute
The CurrentSaturation attribute holds the current saturation value of the light. It is
updated as fast as practical during commands that change the saturation.
The saturation shall be related to the CurrentSaturation attribute by the
relationship
Saturation = CurrentSaturation/254 (CurrentSaturation in the range 0 - 254 inclusive)
If this attribute is implemented then the CurrentHue and ColorMode attributes
shall also be implemented.
```

```
                if (msg.data['currentHue']) {
                   result.color.hue = precisionRound((msg.data['currentHue'] * 360) / 254, 0);
                }
```

sjorge, 2019-11-25:
Getting somewhere... with https://www.rapidtables.com/convert/color/hsv-to-rgb.html I get more or less the colors I set...

RED:
```zigbee2mqtt:info  2019-11-25 21:13:25: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":42,"brightness":254,"color_temp":555,"color":{"x":0.692,"y":0.2953,"saturation":100,"hue":360},"color_mode":1}```
GREEN:
zigbee2mqtt:info  2019-11-25 21:14:45: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":36,"brightness":254,"color_temp":153,"color":{"x":0.1724,"y":0.7468,"saturation":98,"hue":116},"color_mode":1}'
BLUE:
zigbee2mqtt:info  2019-11-25 21:15:26: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":39,"brightness":254,"color_temp":153,"color":{"x":0.1356,"y":0.0418,"saturation":100,"hue":242},"color_mode":1}'

I did have to modify currentSaturation though...

```typescript
            if (
                msg.data['currentX'] || msg.data['currentY'] || msg.data['currentSaturation'] ||
                msg.data['currentHue'] || msg.data['enhancedCurrentHue']
            ) {
                result.color = {};

                if (msg.data['currentX']) {
                    result.color.x = precisionRound(msg.data['currentX'] / 65535, 4);
                }

                if (msg.data['currentY']) {
                    result.color.y = precisionRound(msg.data['currentY'] / 65535, 4);
                }

                if (msg.data['currentSaturation']) {
                    result.color.saturation = precisionRound(msg.data['currentSaturation'] / 2.54, 0);
                }

                if (msg.data['enhancedCurrentHue']) {
                    result.color.hue = precisionRound(msg.data['enhancedCurrentHue'] / (65535 / 360), 1);
                }

                if (msg.data['currentHue']) {
                   result.color.hue = precisionRound((msg.data['currentHue'] * 360) / 254, 0);
                }
            }
```

if you think this looks good, I will do a PR

sjorge, 2019-11-26:
Results with tweaked currentSaturation and new currentHue look good, atleast after reconverting a few colors I tried... will open a PR.

sjorge, 2019-11-26:
Koenkk/zigbee-herdsman-converters#773

sjorge, 2019-11-26:
Not sure how to fix toZigbee.js though... as these bulbs do not support the enhanceHueMove thingy... so we need a way to identify which of the two a bulb supports for writing?

For now I am happy with having the value returned correctly. I mostly set color.hex or color.rgb anyway.

sjorge, 2019-12-18:
The read support for merged a while back, closing this. Still can't set normal 'Hue' but I haven't figure out a way to diferentiate between enhancedMoveHue or moveHue commands based on device.

closed 2019-12-18 by sjorge