Koenkk/zigbee2mqtt #2403: List of "router" devices and some help needed

xefil, 2019-11-24:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1.
CC253X firmware version: CC2530+CC2591 with 3.0.x (latest)

Hi all,
I've set-up succesfully zigbee2mqtt. Until here all ok.
The first test was a CC2531 with a CC2530+CC2591 more or less at 10mt of distance. All on version 1.2 firmware.
With this first setup I've noticed that the  CC2530+CC2591 was not working properly. Even following the setup, seems that after some time, it looses the connection to che coordinator. Not having any way to debug the router, I've decided to switch the router/coordinator, so now the coordinator is my  CC2530+CC2591 with 3.0.x firmware. The CC2531 is for the moment disconnected. The devices have worked good for some hours, then, most of them, had stopped to work. What is connected? Actually not much:
- two Xiaomi Proximity Sensors Mijia
- two Xiaomi Presence Sensors Mijia
- one Xiaomi Temp/humidity/pressure Sensor Aquara 
Actually only the temperature sensor is working. The others, distance not more than 5 meters, same room, are not more conneced.
I think I'll move back to CC2531 flashing it to 3.0.x and leaving for the moment the  CC2530+CC2591 disconnected. Then I would buy some "router ready" devices.
I've checked if there are interferences... well, from the DSL router (FritzBOX), I can scan the network. There is a little interference, but not so much. I think the channel change should not change much. Then, I've some NRF24L01+ around for a MySensors netowork too... Those are on 2,4Ghz as well, but I'm moving away to ESP8266.
So, some questions:
- any suggestion to improve?
- is there a list of "router ready" devices I could purchase? Something like plugs or bulbs. So I could increase the network.
- is a good idea to move to 3.0.x firmware ?

Thanks a lot!

Simon

skandragon, 2019-11-25:
I find that any Zigbee plug is usually a router.  Bulbs are almost never routers, as they are often connected to physical switches, and one does not want the zigbee network to reconfigure constantly on whims of people flipping switches.

xefil, 2019-11-25:
Hello @skandragon ,
Thanks for The hint!
Is there a way to find out which are routers for sure? I.e. in source code or somewhere.
Or just in case some suggested cheap devices?
Thanks!

pplucky, 2019-12-06:
> Is there a way to find out which are routers for sure? I.e. in source code or somewhere.

This info is present in file database.db (in the same location as configuration.yaml), after the device being added to zigbee2mqtt network.

closed 2020-02-11 by stale[bot]