Koenkk/zigbee2mqtt #2405: Cannot Add Device

wes1993, 2019-11-24:
# Bug Report
Cannot add other device, i'm trying with some ikea light and switch but not worinkg.
other devices works well.
## What happened
If i try to add new devices, with permit_join=true the addon wont work and i cannot add devices.
## What did you expect to happen
When i enable the permit_join and the search in my device i need to see zigbee2mqtt that can add the device.
## How to reproduce it (minimal and precise)
Eneble permit Join, Reset the device to enable pairing...
## Debug Info
I have one cc2531 with Router firmware and some Ikea Bulb and Smart Plug that i think works like a router. see the image attached.
zigbee2mqtt version: 1.7.1
CC2531 firmware version: Z-Stack_Home_1.2 (default).

Thanks a Lot
Stefano
[image]
[image]
[image]
[image]

nldroid, 2019-11-24:
Do you see anything in the zigbee2mqtt log?

wes1993, 2019-11-24:
Thanks a lot for your reply,
for what i see i can't have find something relevant... 
i will attach the log, i have enebled the permit_join so after this log i think you can find something.
Thanks
St
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3883911/log.txt)
efano

tsprivate, 2019-11-26:
I have the same issue. I'm running ~30 device including some routers
- Xiaomi sockets (acting as router)
- IKEA router
- CC2530_CC2591
There should be an issue with any device limit as I understan.

My current version is 1.7.1

I tried as well to delete a device but this is not recognized either

wes1993, 2019-11-26:
What firmware do you have installed? The sourcerouting or the normal?
Where you have your Home Assistant environment? Hassio? What version?
Thanks a lot
Stefano
I hope we can find a solution

ptvoinfo, 2019-11-26:
@wes1993 
Your image looks good. Did you try to pair your device near the coordinator?

wes1993, 2019-11-26:
Thanks a lot ptvo for your reply :-D
Yes, i have tried near the coordinator, the first two ikea bulb works like a charm but the other.... nothing... 
I think that the coordinator has only 7 devices connected directly, is correct?

ptvoinfo, 2019-11-26:
@wes1993 I've counted 11 direct connections with the coordinator. So, your device can be connected. I can only offer to try another device.

wes1993, 2019-11-26:
how can you have seen that i have the 11 devices directly to the coordinator?
also if i try with the sourcerouting firmware is better for my network devices?
Really thanks for your help
Stefano

FixxCZ, 2019-11-26:
Same here, I was adding Ikea bulb like week ago without any issue. And now I can't add even devices I have been using before. Was trying ikea on/off switch, Xiaomi motion sensor, xiaomi switch. I have 9 devices in the network and one router to extend signal. I even enabled the joining in the config permanently and restarted the server but still not pairing with any of those three devices. I have motion sensor and switch in my network already so I know it should work and how to pair it. Runnig the laters ZigbeetoMQTT.

wes1993, 2019-11-27:
Hello to everyone!!
Finally i think i have solved the problem!!!
I have flashed the CC2531 with the sourcerouting FW instead of the normal FW, after that i have executed zigbee2mqtt with permit_join=true and after about 10 minutes all my devices, including the battery has joined the network again.
after that i have added successfully the other devices. Now my map is this
[image]

tsprivate, 2019-11-27:
how can I check the FW for source-routing. I used the default image at the time (~1year ago)

wes1993, 2019-11-28:
Do you have Home Assistant in Hassio?

tsprivate, 2019-11-28:
I'm running it via MQTT an Node-Red

wes1993, 2019-11-28:
If you have Hassio you can install this addon:
https://github.com/yllibed/Zigbee2MqttAssistant
And in the info you can read the version.
You can also install the new FW without programmer.

tsprivate, 2019-11-28:
thats what the log shows on startup

`Nov 28 23:09:22 noderedserver npm[2161]: zigbee2mqtt:info  2019-11-28T12:09:22: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-28.23-09-20'
Nov 28 23:09:22 noderedserver npm[2161]: zigbee2mqtt:info  2019-11-28T12:09:22: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
Nov 28 23:09:22 noderedserver npm[2161]: zigbee2mqtt:info  2019-11-28T12:09:22: Starting zigbee-herdsman...
Nov 28 23:09:24 noderedserver npm[2161]: zigbee2mqtt:info  2019-11-28T12:09:24: zigbee-herdsman started
Nov 28 23:09:24 noderedserver npm[2161]: zigbee2mqtt:info  2019-11-28T12:09:24: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'
Nov 28 23:09:24 noderedserver npm[2161]: zigbee2mqtt:info  2019-11-28T12:09:24: Currently 29 devices are joined:
Nov 28 23:0`

FixxCZ, 2019-11-28:
@wes1993 Thanks for the tip! That should be bundled with Zigbee2MQTT. I installed on Hassio but can't see the option to update the FW on my coordinator. Can you navigate me where to find it?

wes1993, 2019-11-28:
not from Zigbee2MQTT but from Windows PC.
Read this: https://github.com/Koenkk/zigbee2mqtt/issues/320

I think you have the normal FW not the SourceRouting.
Download the sourcerouting
DO NOT USE THE BIN INSIDE THE ZIP

FixxCZ, 2019-11-28:
I reflashed it via CC debugger before I read your post. Anyway it works like charm now. I moved from Feb 2019 firmware to latest 20190608 version. Whole operation takes few minutes and as expected no re-pairing needed. TRADFRI ON/OFF switch (E1743) paired on first try 5 meters away from coordinator.

wes1993, 2019-11-28:
Perfect!!! :) for the next time you know that you can do also without debugger!! ;)
But if you have installed the 08062019 you have installed the normal and not the sourcerouting...
With this FW you could have problem if add other devices because the router managment is not so good with this FW

wes1993, 2019-11-28:
My advice is to install the sourcerouting if you have some routers in your house

tsprivate, 2019-11-28:
How to install without debugger? I'm using Zigbee2MQTT on RPI and NodeRed ... dont have HASSIO

limes007, 2020-01-04:
I had a lot of trouble adding IKEA Tradfri lights too (Zigbee2mqtt v1.8.0). Trying to add the devices ends in the messages: "Device ... joined", "Starting interview of ...", "Failed to interview...", again "Starting interview of...", "Failed to interview...". I tried a lot of times, but nothing changes. I tried also the trick of "slowly turn off" from here (https://github.com/Koenkk/zigbee2mqtt/issues/2355#issuecomment-555673674), but this had no effect.

So I put the CC2531 stick on my PC and flashed the sourcerouting firmware (20190619) as suggested above. (Before: Default 20190608).

After restarting Zigbee2mqtt the first bulb could be paired in seconds. But not the second. Now it was different: The Zigbee2mqtt log does not even show the message "Device ... joined". It shows nothing even though the bulb has confirmed the reset flashing the light. I restarted Zigbee2mqtt several times, checked the permit_join flag. Nothing changed. So I removed one of my Xiaomi sensors and try to re-pair. Same effect, the log shows nothing. (BTW: The Xiaomi sensor had the problem of not shown as connected in the network map before, as described here: https://github.com/Koenkk/zigbee2mqtt/issues/1001)

So I thought about flashing one of the firmware versions again, but first I simply removed the CC2531 stick from my raspi, connected it again and restarted Zigbee2mqtt. Voila: The Xiaomi sensor and also the remaining two IKEA Tradfri bulbs can be paired in few seconds.

So I ponder if the other firmware had an effect in the first place. Or was if it simply the disconnecting from the raspi? But, the behavior war slightly different. Nevertheless, if someone has problems adding IKEA Tradfri bulbs I would recommend to temporarily remove the CC2531 stick.

closed 2020-03-11 by stale[bot]