Koenkk/zigbee2mqtt #2409: Unsupported device: HEIMAN smart remote controller: HS1RC-E

VenkateshGSRao, 2019-11-24:
# Bug Report
## What happened
Trying to pair HEIMAN smart remote controller: HS1RC-E

https://nl.aliexpress.com/item/32975159931.html?spm=a2g0s.9042311.0.0.177d4c4dWELfwJ

tried to resolve the issue by following:
https://github.com/Koenkk/zigbee2mqtt/issues/1581
but could not resolve the issue.

device id : 0x000d6f00104f28bd

getting this message:

`Interview - failed for device '0x000d6f00104f28bd' with error 'Error: Timeout - 49157 - 1 - 11 - 1280 - 11 after 15000ms`

despite clicking button on the remote continuously to keep it awake unable to pair.

**Entry in batabase.db :**
`{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00104f28bd","nwkAddr":49157,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"RC-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[0,1,3,1280],"outClusterList":[3,1281],"clusters":{"genBasic":{"attributes":{"modelId":"RC-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1"}}},"binds":[]}},"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2017.11.1","zclVersion":2,"interviewCompleted":false,"meta":{}}`

**Entry in devices.js**
```
    // HEIMAN Remote
    {
        zigbeeModel: ['RC-EM'],
        model: 'HS1RC-E',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'action',
        fromZigbee: [
            fz.battery_200,
            fz.heiman_smart_controller_armmode, fz.heiman_smart_controller_emergency,
	    ],
	    toZigbee: [],
	},

```
**Entry in fromZigbee.js**
```
    HS1RC_E_arm: {
        cid: 'ssIasAce',
        type: 'cmdArm',
        convert: (model, msg, publish, options) => {
            const action = msg.data.data['armmode'];
            delete msg.data.data['armmode'];
            const modeLookup = {
                  0: 'unlock',
                  1: 'home',
                  3: 'lock',
            };
            return {action: modeLookup[action]};
        },
    },
```

## What did you expect to happen

Pair successfully

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608

full debug log:
https://pastebin.com/ij5Gxd8K

bambouzel, 2019-11-24:
@VenkateshGSRao 
My devices.js looked like:
```
{
    zigbeeModel: ['RC-EM'],
    model: 'HS1RC-E',
    vendor: 'HEIMAN',
    description: 'Smart remote controller',
    supports: 'action',
    fromZigbee: [
        fz.battery_200, fz.ignore_power_change, fz.ignore_iaszone_change,
        fz.HS1RC_E_arm, fz.heiman_smart_controller_emergency,
    ],
    toZigbee: [],
}
```

But for the newest version of Z2M I just added 'RC-EM' to the existing 'HS1RC-M' device.  So they are treated as the same.

VenkateshGSRao, 2019-11-25:
Thanks for your reply.
Including 
fz.ignore_power_change, fz.ignore_iaszone_change 
In fromZigbee is producing error because they are not defined in fromZigBee.js
How to resolve that?

VenkateshGSRao, 2019-11-25:
Do you think by just adding RC-EM to existing definition in new Z2M will solve the problem?

bambouzel, 2019-11-26:
@VenkateshGSRao I would try to do that first, get the latest Z2M reelease, look at the devices.js and add 'RC-EM' to the 'HS1RC-M' definition if it is not yet there.
Then your device will be recognized as a HS1RC-M but will work correctly.
[log](https://github.com/Koenkk/zigbee2mqtt/issues/1957#issuecomment-556172180)

VenkateshGSRao, 2019-11-26:
I tried as you said. No luck.
I got one HS1RC-M also. It gets paired successfully on every attempt. But, on pressing buttons, nothing happens. Any idea?
Here is the log
> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2019-11-26 17:15:04: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-26.17-15-03'
zigbee2mqtt:debug 2019-11-26 17:15:04: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-11-23.18-23-32'
zigbee2mqtt:debug 2019-11-26 17:15:04: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2019-11-26 17:15:04: Starting zigbee2mqtt version 1.7.1+dev (commit #546a72f)
zigbee2mqtt:info  2019-11-26 17:15:04: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-26 17:15:04: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:info  2019-11-26 17:15:06: zigbee-herdsman started
zigbee2mqtt:info  2019-11-26 17:15:06: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2019-11-26 17:15:06: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2019-11-26 17:15:06: Currently 5 devices are joined:
zigbee2mqtt:info  2019-11-26 17:15:06: 0x00158d00033e8fdb (0x00158d00033e8fdb): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
zigbee2mqtt:info  2019-11-26 17:15:06: 0x00158d000303f8aa (0x00158d000303f8aa): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
zigbee2mqtt:info  2019-11-26 17:15:06: 0x00158d00033ad918 (0x00158d00033ad918): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
zigbee2mqtt:info  2019-11-26 17:15:06: 0x00124b001d436c34 (0x00124b001d436c34): SA-003-Zigbee - eWeLink Zigbee smart plug (Router)
zigbee2mqtt:info  2019-11-26 17:15:06: 0x000d6f00104f28bd (0x000d6f00104f28bd): HS1RC-E - HEIMAN Smart remote controller (EndDevice)
zigbee2mqtt:warn  2019-11-26 17:15:06: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-11-26 17:15:06: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-26 17:15:06: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-26 17:15:06: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-11-26 17:15:06: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2019-11-26 17:15:07: Connected to MQTT server
zigbee2mqtt:info  2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d00033e8fdb', payload '{"battery":100,"voltage":3092,"linkquality":120}'
zigbee2mqtt:info  2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d000303f8aa', payload '{"occupancy":false,"linkquality":131,"battery":100,"voltage":3015}'
zigbee2mqtt:info  2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d00033ad918', payload '{"battery":100,"voltage":3065,"linkquality":120,"contact":true}'
zigbee2mqtt:info  2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/0x00124b001d436c34', payload '{"state":"OFF","linkquality":141}'
zigbee2mqtt:info  2019-11-26 17:15:07: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1+dev","commit":"546a72f","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:info  2019-11-26 17:16:33: Device '0x005043c9a32f701d' joined
zigbee2mqtt:info  2019-11-26 17:16:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x005043c9a32f701d"}}'
zigbee2mqtt:info  2019-11-26 17:16:33: Starting interview of '0x005043c9a32f701d'
zigbee2mqtt:info  2019-11-26 17:16:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x005043c9a32f701d"}}'
zigbee2mqtt:debug 2019-11-26 17:16:33: Device '0x005043c9a32f701d' announced itself
zigbee2mqtt:debug 2019-11-26 17:16:38: Received Zigbee message from '0x005043c9a32f701d', type 'readResponse', cluster 'genBasic', data '{"modelId":"RC_V14","manufacturerName":"Heiman","powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-26 17:16:38: No converter available for 'HS1RC-M' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"RC_V14","manufacturerName":"Heiman","powerSource":3}'
zigbee2mqtt:debug 2019-11-26 17:16:39: Received Zigbee message from '0x005043c9a32f701d', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":3,"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2019-11-26 17:16:39: No converter available for 'HS1RC-M' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":3,"stackVersion":2}'
zigbee2mqtt:debug 2019-11-26 17:16:42: Received Zigbee message from '0x005043c9a32f701d', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20150427"}' from endpoint 1 with groupID 0
[... cut ...]
zigbee2mqtt:info  2019-11-26 17:16:42: Successfully interviewed '0x005043c9a32f701d', device has successfully been paired
zigbee2mqtt:info  2019-11-26 17:16:42: Device '0x005043c9a32f701d' is supported, identified as: HEIMAN Smart remote controller (HS1RC-M)
zigbee2mqtt:info  2019-11-26 17:16:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x005043c9a32f701d","model":"HS1RC-M","vendor":"HEIMAN","description":"Smart remote controller","supported":true}}'

komissar75, 2020-01-11:
Hi all, 
Does anybody successfully joined HEIMAN HS1RC-E?
I have similar problem and nothing helps :(

closed 2020-03-18 by stale[bot]