Koenkk/zigbee2mqtt #2410: [RFC] Group Home Assistant Auto-discovery

timstanley1985, 2019-11-24:
Groups are not currently auto-discovered by home assistant.  Can an option be added so that zigbee2mqtt groups can announce themselves to home assistant discovery?  This would be particularly helpful for lighting groups as no manual configuration would be required

sjorge, 2020-01-24:
I'm not sure how this would work though, as groups could contain a set of lights, but also a set of blinds for example.

How would we know which one to expose to HA?

Koenkk, 2020-01-25:
@sjorge in that case both blinds and light should be discovered, bulbs will ignore the blinds commands and vice versa.

nobodypb, 2020-09-01:
Are there currently any efforts to implement this? I would be very interested in the discovery of groups via the Home Assistant protocol. Because then I could easily implement this for WebThings Gateway and wouldn't have to query each device individually.

Or are pull requests welcome? I haven't poke that much into the code of zigbee2mqtt, yet. So I have no idea how complicated it would be to accomplish this, but I could give it a try. However, I would be very pleased if you could give me some hints for the implementation then.

rindlerblabla, 2021-04-21:
It's a much wanted functionality for me as well!

julien-w, 2021-04-24:
It would be a very big improvement. Perhaps the last must-have feature missing for me :-)

rindlerblabla, 2021-05-25:
Am still interested in this feature.

Koenkk, 2021-06-16:
HA group discovery is supported now for groups of lights, switches, locks and covers.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-16 by Koenkk

mdehaas, 2021-06-21:
Hi @Koenkk, any planning on when this is going to be included in the stable branch?

Koenkk, 2021-06-22:
@mdehaas next z2m release on 1 July

jd1900, 2021-07-01:
How is this implemented?
When turning on/off is a Zigbee command for each device sent or a command for the Zigbee group? I guess that the latest.

And..., I've noticed that when bootstraped if lights from a group are "on", the group is not "on". But once it is turned on/off it "syncs" with the group.

mdehaas, 2021-07-01:
Hi jd1900,

If lights from one group are partially on, the group is not on or off. But I feel that when creating a group in Zigbee the individual lights should not be controlled. What is your 'use-case'? And I think you are looking in HA? What do you see in the Z2M interface?

jd1900, 2021-07-01:
> Hi jd1900,
> 
> If lights from one group are partially on, the group is not on or off. But I feel that when creating a group in Zigbee the individual lights should not be controlled. What is your 'use-case'? And I think you are looking in HA? What do you see in the Z2M interface?

It happened in the HA interface.
And in groups with just one device after a reboot.
So it is not a big deal actually.

mdehaas, 2021-07-01:
I have a group with one light because I need to install the rest in the ceiling so I will try this tomorrow although I feel this is not a common scenario…

onedr0p, 2021-07-11:
I am currently experiencing this issue in home assistant. I have groups of 2 lights defined in zigbee2mqtt the entity in the GIF. When I toggle the group in home assistant the state is not updated correctly, until it is pressed again. Is this a known limitation of the groups feature?

[image]

Koenkk, 2021-07-12:
@onedr0p is this fixed after setting the color temperature once via HA?

onedr0p, 2021-07-12:
> @onedr0p is this fixed after setting the color temperature once via HA?

I can confirm this fixes it, thanks!

Morphy99, 2021-08-05:
How do we disable group discovery in HA? I don't need to use groups apart from when Binding my IKEA remotes to bulbs so I have them set up for this. But now I have extra entities in HA which are just duplicates of devices.