Koenkk/zigbee2mqtt #2411: Mueller Licht Scene

kotarokun, 2019-11-25:
Hello, 

i just updated the Zigbee2MQTT from shepherd to herdsman. 
I noticed that Müller Licht unfortunately still doesn't have scene support. So I asked if this could be added to Müllerlicht. For this lamp 404000_404005_404012 supported RGB should support the following additional function.

toZigbeeConverter:
`light_scene: {
        key: ['scene'],
        convert: (key, value, message, type, postfix) => {
            const cid = 'genBasic';
            if (type === 'set') {
                return {
                    cid: cid,
                    cmd: 'write',
                    cmdType: 'foundation',  // 'functional',
                    zclData: [{
                        attrId: 16389,
                        dataType: 32,
                        attrData: value
		    }],
	            cfg: {
				manufSpec: 1,
        		 	manufCode: 0x121b,   
		          },
                };
            }
        },
    },`

and device.js:
`    toZigbee: [.... tz.light_scene,`

That worked for Sherperd. I haven't tested it for Herdsman yet. 
Like it stand [here](https://forum.fhem.de/index.php/topic,95246.msg909441/topicseen.html#msg909441)

thanks a lot

Koenkk, 2019-11-25:
What exactly does it do?

kotarokun, 2019-11-25:
The Müller Licht have 6 light-Scenes on Board, so i can choose them via Remote-Control. I also like to choose them via MQTT ^^

Koenkk, 2019-11-25:
So what values can be put in? Could you write some documentation for this?

kotarokun, 2019-11-25:
Sorry, of course. 
Here is MQTT used for KODI
`scene:1,2,3,4,5,6 zigbee2mqtt/0x00158d0003247a3c/set {"$EVTPART0":$EVTPART1}`

Like above, you can select 1 to 6 😅

Koenkk, 2019-11-26:
thanks, could we map these numbers to some more friendly names? E.g. 1 = `fireplace`?

closed 2020-02-01 by stale[bot]

Albarion, 2021-04-02:
Is this implemented? It seems like I cannot use the scene buttons on my müller remote.