Koenkk/zigbee2mqtt #2412: Error: Could not locate the bindings file

jesperldk, 2019-11-25:
I just tried an update following the update guide. However, it does not start:

````
pi@openHelper2:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2019-11-25T13:04:31: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-25.14-04-28'
zigbee2mqtt:info  2019-11-25T13:04:32: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-11-25T13:04:32: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-11-25T13:04:32: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-25T13:04:32: Failed to start zigbee
zigbee2mqtt:error 2019-11-25T13:04:32: Exiting...
zigbee2mqtt:error 2019-11-25T13:04:32: Error: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/default/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/compiled/10.9.0/linux/arm/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/binding/node-v64-linux-arm/bindings.node
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/index.js:14:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-11-25T13_04_32_830Z-debug.log
````

I did try:
````
rm -rf node_modules
npm install
npm start
````
But it did not change anything.

Help appreciated, thank you.

Koenkk, 2019-11-25:
- What is the output of `git status`?
- Can you try the solutions from:
    - https://stackoverflow.com/questions/37458918/node-js-could-not-locate-the-bindings-file
    - https://stackoverflow.com/questions/27541586/error-could-not-locate-the-bindings-file-tried-56

jesperldk, 2019-11-25:
Thanks for the reply :-)
I got it working, but not quite sure exactly how. I tried several npm install, and with sudo I got an "not enough memory" - perhaps some process had stolen a lot of memory, and sometimes the install failed without an error because of that. Anyway, I did the first reboot in 7 months of my Pi Zero; a new install; some more reboots to get the stick to behave; and now it is running :-)

jesperldk, 2019-11-25:
Thanks for the reply!

I got it working, but not quite sure how.

I tried several npm install, and when running with sudo i got an not enough memory; perhaps the install failed differently without errors because of low memory?
I rebooted my Pi Zero for the first time in 7 months; a new npm install; several extra reboots to be able to see the stick; and now it runs :-)

closed 2019-11-25 by jesperldk

lancelon, 2020-02-07:
I've got this issue, trying all sorts, reinstalls, rebuilds, rm -rf of the directory etc

nothing helping so far!

```
root@raspbx:/media/usb/zigbee2mqtt# npm start

> zigbee2mqtt@1.10.0 start /media/usb/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-02-07 23:54:41: Logging to console and directory: '/media/usb/zigbee2mqtt/data/log/2020-02-07.23-54-40' filename: log.txt
zigbee2mqtt:info  2020-02-07 23:54:42: Starting zigbee2mqtt version 1.10.0 (commit #8df6be8)
zigbee2mqtt:info  2020-02-07 23:54:42: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-07 23:54:42: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-07 23:54:42: Failed to start zigbee
zigbee2mqtt:error 2020-02-07 23:54:42: Exiting...
zigbee2mqtt:error 2020-02-07 23:54:42: Error: Could not locate the bindings file. Tried:
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Debug/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Release/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Release/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/default/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/compiled/12.15.0/linux/arm/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/binding/node-v72-linux-arm/bindings.node
    at bindings (/media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
    at Module.load (internal/modules/cjs/loader.js:811:32)
    at Function.Module._load (internal/modules/cjs/loader.js:723:14)
    at Module.require (internal/modules/cjs/loader.js:848:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/media/usb/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/index.js:14:22)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-07T23_54_42_416Z-debug.log
```