Koenkk/zigbee2mqtt #2413: [question] Pairing Arduino+Xbee as device

lealoureiro, 2019-11-25:
Hello,

Is possible to send a receive messages from a custom firmware Arduino + Xbee, with a Zigbee2Mqtt coordinator? And how do I pair ir or set the address of coordinator etc?

I would like to connect a relay with my arduino and pair this "custom device" with my current zigbee2mqtt bridge to send messages and open my old style door bell button with the relay.

Thanks in advance.

Kind regards,
Leandro Loureiro

skandragon, 2019-11-25:
I have an Xbee connected to my network.  Here's what I did.

* Set the encryption key to the standard HA key.
* Enable joining.

Note that my Xbee has no profiles it will respond to other than the ones a standard Xbee responds to.  I use it mostly so I can run a network monitoring tool.

lealoureiro, 2019-11-26:
> * Set the encryption key to the standard HA key.
You mean the encryption key defined in configuration of zigbee2mqtt? 

And which address i set in xbee firmware?

closed 2020-02-01 by stale[bot]