Koenkk/zigbee2mqtt #2415: Supported device (Paul Neuhaus) not really functional as expected

walkerprog, 2019-11-25:
# Bug Report
## What happened

Hey Koenkk,
first of all thank you very much for this very cool software. It's exactly what I need for my home automation.
After I got my Tradfri devices to work more or less the way I want I bought an Paul Neuhaus 100.110.39 as it is listed as supported device and I wanted no issues at integration.
But with exactly this devices I ran into issues I can't understand / resolve on my own.

What is working:
joining
turning on and off works
changing brightness works
setting ANY value for color_temp to switch from color to white works

issue 1:
When setting the color the light turns off.
No color change after turning on again. The MQTT message declares state: OFF and brightness: 0 although only the color was send. This too happens if the color command is send with brightness and state. It doesn't matter if color is send as rgb, xy or hsv.
command send (example):
-m '"state": "ON","brightness": 255,"color": {"hue": 360,"saturation": 100}'
result:
payload '{"state":"OFF","brightness":0,"linkquality":13,"color_temp":800,"color":{"x":0.618299624701903,"y":0.372495061588915},"device":{"friendlyName":"name","ieeeAddr":"xxx","networkAddress":xxx,"type":"Router","manufacturerID":4151,"manufacturerName":"Neuhaus Lighting Group ","powerSource":"Mains (single phase)","applicationVersion":1,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20150821","softwareBuildID":"1001-0007"}}'
never a change in color, this only changes when color is changed via remote

when color changed via remote I get:
Received Zigbee message from 'name', type 'commandEnhancedMoveToHueAndSaturation', cluster 'lightingColorCtrl', data '{"enhancehue":61937,"saturation":254,"transtime":5}' from endpoint 1 with groupID 1

issue 2:
remote itself is reported as unsupported - strange to me, expected the remote to also be supported
Without this I cannot trigger the on/off button of the remote so it seems I have to figure this out by supporting this "new" device

The issue is that changes made via remote are NOT reported, although reporting is activated:
Succesfully setup reporting for 'xxx' - 2 - genOnOff
Succesfully setup reporting for 'xxx' - 2 - genLevelCtrl
Succesfully setup reporting for 'xxx' - 2 - lightingColorCtrl

when I compare the logs from this device to my IKEA Tradfri devices it looks similar.
Nevertheless reporting with Tradfri works flawlessly. If I dimm a Tradfri Panel the new brightness is reported immidiately.
With the Neuhaus Panel nothing is ever reported if ANYthing is changed via remote

issue 3:
devices gets disconnect when zigbee2mqtt is restarted and needs to be rejoined.
As long as zigbee2mqtt runs everything is ok and the Panel can be power-cycled without disconnects
It can be on for hours or powerd down via wall switch for hours. If the device is powered it reports itself und is responding
After a restart of zigbee2mqtt (doesn't matter if the Panel is powered and on or off or powered off via wall switch, doesn't matter if the remote is paired or not) the device immidiately stops responding to anything from the coordinator. The remote is still functional in that moment and can control the device.

If you need further information please let me know and I would be happy to provide you the information needed for debugging this issues.

## What did you expect to happen

issue 1: setting the color should change the color, NOT turn off the light
issue 2: zigbee2mqtt should report device state on changes via remote like it does with IKEA Tradfri
issue 3: device should be still present without the need of rejoining

## How to reproduce it (minimal and precise)

join Paul Neuhaus 100.110.39 to coordinator
for issue 1: send described mqtt commands
for issue 2: nothing, after join device never reports to changes via remote
for issue 3: restart zigbee2mqtt

## Debug Info
zigbee2mqtt version: 1.7.1 (commit #b459c35)
CC253X firmware version: 20190608 on CC2531

Koenkk, 2019-11-27:
for issue 1, can you provide the debug log when sending the commands?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

walkerprog, 2019-11-27:
Hello Koenkk,

thanks for your reply.
I have set debug-logging since the very beginning.
But I miss a few things in logging, for example the send zigbee-commands.

here the log while sending the command for color turning out the light instead:
https://pastebin.com/B4LvQk71

walkerprog, 2019-11-27:
here the logs while changing the color via remote:
https://pastebin.com/MHCu27Fe

color changed perfectly

Koenkk, 2019-11-28:
I don't see any strange things in the log for now. Can you sniff the traffic while setting the color (and the panel turns off?) https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

walkerprog, 2019-11-28:
takes a few days, I need a second stick for this, don't want to flash the current one

stick is ordered, I'll post the dumps asap

walkerprog, 2019-11-30:
stick just arrived and traffic is sniffed.
seems to be wrong commands send out.
After triggering a color change via mqtt I get following in Wireshark:

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 61
    Command: Move to Level with OnOff (0x04)
    Payload
        Level: 0
        Transition Time: 0.0 seconds

After settings the brightness to 150 I get:

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 62
    Command: Move to Level with OnOff (0x04)
    Payload
        Level: 150
        Transition Time: 0.0 seconds

so zigbee2mqtt does not send a color change but an 0 brightness?
Of course the light is off with 0 brightness....

After setting the color with the remote I get:

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 0
    Command: Color Loop Set (0x44)
    Payload
        Update Flags: 0x01
            .... ...1 = Update Action: 1
            .... ..0. = Update Direction: 0
            .... .0.. = Update Time: 0
            .... 0... = Update Start Hue: 0
        Action: De-activate (0)
        Direction: Hue is Decrementing (0)
        Time: 0 seconds
        Enhanced Hue: 0

and:

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 1
    Command: Enhanced Move to Hue and Saturation (0x43)
    Payload
        Enhanced Hue: 3084
        Saturation: 254
        Transition Time: 0.5 seconds

Koenkk, 2019-12-04:
I think I already see the problem, the JSON you send is malformed.

```
Nov 27 20:34:31 linuxbox npm[5815]: zigbee2mqtt:debug 2019-11-27T19:34:31: Received MQTT message on 'zigbee2mqtt/LEDPanelNeuhaus/set' with data '"{color": {"hue": 150, "saturation": 100}}'
``` 

The payload should be `{"color": {"hue": 150, "saturation": 100}}` (" before color is missing)

walkerprog, 2019-12-04:
shame on me...
It was indeed the wrong order of the first two characters.
With the right order it's working now ^^

Thank you very very much, needed that hint

Do you have any ideas for the other issues?

Koenkk, 2019-12-05:
issue 2: https://www.zigbee2mqtt.io/information/report.html
issue 3: would be very strange, are you sure? can you provide logs?

walkerprog, 2019-12-05:
issue 2:

report: true IS set in configuration.yaml
IKEA Floalt-Panels work perfectly:
https://pastebin.com/5wiggjKz

Neuhaus-Panel is not working:
https://pastebin.com/CMa6HeQz
Brightness is set via remote AND via mqtt but there is nothing reported like the IKEA panels do

additionally, if the IKEA Pannels are power-cycled I get "Device 'xyz' announced itself" AND "Succesfully setup reporting for 'xyz' - number - channel" AND mqtt publishes with the current state
if the Neuhaus-panel is power-cycled I get the "Device 'xyz' announced itself" ONLY - nothing more

issue 3:
initial join of device is included above
It doesn't matter how often and how long the panel is powered off (even for hours or days) with the wall switch - when powering on the panel announces itself and reacts on any commands via remote AND mqtt.
But if the zigbee2mqtt daemon is restartet I get this:
https://pastebin.com/Re0bfTdx

after the "failed: 'Error: Timeout" I changed brightness and color via remote. No reporting, but brightness and color changed.

It doesn't matter if the Neuhaus-panel is powered on or disconnected from power via wall switch.
The remote is still working, but no mqtt-commands are sucessful, no matter how long I wait while the panel is powered on.
After factory reset and rejoin (and pairing the remote again) everythings (except reporting) works fine untill next restart of zigbee2mqtt daemon.

In the case you need any zigbee-dumps please tell me a secure way to provide you the pcapng-files.

sjorge, 2019-12-05:
Do Paul Neuhaus lights support reporting of anything other than On/Off? If not, maybe another candidate for the Hue bind workaround.

walkerprog, 2019-12-06:
What workaround do you mean?
Can you provide me additional info please?

sjorge, 2019-12-06:
I with some help of Koen implemented this for Philips Hue as they weren't reporting brightness changes with both Hue remote or Ikea remotes in this commit https://github.com/Koenkk/zigbee2mqtt/commit/a813663de5cf24110f2f7c943682bc3e8d4627eb#diff-c7f8b2826f26f7710ede1e542c3d0b92R48

walkerprog, 2019-12-06:
thanks, sounds interesting

I'll give this a try the next days

cron0mat, 2019-12-19:
I have issue no. 3 too.

I am running zStack30x from 20190425 on CC2531 and zigbee2mqtt 1.8.0 (master brunch)

After restart of the zigbee2mqtt service, Paul Neuhaus lamp is not available anymore.
I get message 
> Failed to ping 'wz_lampe'
for example, but corresponding remote control is online.

Is there any kind of "rejoin" needed before pinging ?

walkerprog, 2020-02-18:
with zigbee 1.10 (master brunch) it is somewhat working.
strange thing was: everything was broken after immediately after the update, the CC25xx reacted in a very strange way.
after flashing the exact firmware it was having to the stick, zigbee2mqtt was operational again, but I had to repair all devices.
Since then my Neuhaus keeps connected, even after powering of the raspberry and pulling out the stick for several hours.

I don't know, if it is the new version or if the firmware was somewhat corrupted (never flashed it before, already sold it with the right firmware).

after repairing every devices keeps connected during restarts, but the reporting is completely broken. If i trigger a button on the remote for the LED Panel, the LED Panel is controlled (expected), BUT: in the rare case there is any reporting, the new brightness is NOT reported in the context of the LED-Panel, but in the context of the (independant and not connected) tradfri LED Driver.

There is some other weird stuff so I will save my devices and configuration and completely start from scratch - perhaps something just went wrong during the update.

I will report after the new installation - guess I need some help with the mentioned "Hue bind workaround" - was way to stupid to get it running ^^

walkerprog, 2020-02-19:
after a whole day to get tradfri working as before:

issue 1 was rsolved before
issue 2 persistent - I'm to stupid to get the workarounnd flying
issue 3 now completely resolved with constellation:
zigbee2mqtt version: 1.10.0 (commit #8df6be8)
CC253X firmware version: 20190608 on CC2531
The Neuhaus panel now survives zigbee2mqtt restarts and even shutting down the raspberry pi

Thanks for all help so far!

closed 2020-04-26 by stale[bot]