Koenkk/zigbee2mqtt #2416: Ikea E1810 Remote - Not Reporting Battery

jeremywillans, 2019-11-26:
# Bug Report
## What happened
Ikea E1810 Remote Control fails to report battery value when pressing a button.

Tried Removing/Repairing, and waited 24h to see if it value would be eventually sent.

## What did you expect to happen
Battery percentage to be sent with command updates, same as existing E1524

## How to reproduce it (minimal and precise)
Pair E1810, press a button, action is sent, however no battery value is included

## Debug Info
`info  2019-11-26 08:39:58: Logging to directory: '/app/data/log/2019-11-26.08-39-57'`
`info  2019-11-26 08:39:58: Starting zigbee2mqtt version 1.7.1+dev (commit #d37d9ee)`
`info  2019-11-26 08:39:58: Starting zigbee-herdsman...`
`info  2019-11-26 08:39:59: zigbee-herdsman started`
`info  2019-11-26 08:39:59: Coordinator firmware version: '{"type":"zStack12","meta"{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`
...
E1810
`info  2019-11-26 11:36:19: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6816c0', payload '{"linkquality":78,"action":"toggle"}'`
`info  2019-11-26 11:36:19: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6816c0', payload '{"linkquality":78,"action":""}'`
`info  2019-11-26 11:36:21: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6816c0', payload '{"linkquality":80,"action":"toggle"}'`
`info  2019-11-26 11:36:21: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6816c0', payload '{"linkquality":80,"action":""}'`
...
E1524
`info  2019-11-26 11:45:26: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe07e4ca', payload '{"linkquality":78,"battery":60,"action":"toggle"}'`
`info  2019-11-26 11:45:26: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe07e4ca', payload '{"linkquality":78,"battery":60,"action":""}'`
`info  2019-11-26 11:45:28: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe07e4ca', payload '{"linkquality":83,"battery":60,"action":"toggle"}'`
`info  2019-11-26 11:45:28: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe07e4ca', payload '{"linkquality":83,"battery":60,"action":""}'`

Koenkk, 2019-11-26:
Did you pair this device recently? (while on zigbeee2mqtt 1.7.0 or 1.7.1?)

jeremywillans, 2019-11-26:
Paired on 1.7.1-dev about 15 hours ago, still not reporting battery

codewise-nicolas, 2019-12-04:
I just got started on zigbee2mqtt and I can see the same problem.
I have the E1810 and battery level is not in the MQTT payload.
Im on version 1.7.1

`{"version":"1.7.1","commit":"b459c35","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}`

`MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":11,"action":"brightness_up_click"}'`

Also tried 1.7.1-dev, and no battery details still
`{"version":"1.7.1+dev","commit":"ba26887","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}`

`MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":21,"action":"brightness_up_click"}'`

codewise-nicolas, 2019-12-04:
Im not sure what happened, but all of a sudden (magically almost) the 5-button remote/zigbee2mqtt is reporting the battery in the publishes.

> zigbee2mqtt:info  2019-12-04T11:53:45: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":0,"action":"brightness_up_click"}'
zigbee2mqtt:info  2019-12-04T12:02:56: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":63,"action":"arrow_right_click"}'
zigbee2mqtt:info  2019-12-04T12:02:57: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":68,"action":"brightness_down_click"}'
zigbee2mqtt:info  2019-12-04T12:03:02: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":44,"action":"brightness_up_click"}'
zigbee2mqtt:info  2019-12-04T12:03:03: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":44,"action":"brightness_up_click"}'
zigbee2mqtt:info  2019-12-04T12:03:04: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":44,"battery":87}'
zigbee2mqtt:info  2019-12-04T12:03:04: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":44,"battery":87,"action":"brightness_up_click"}'
zigbee2mqtt:info  2019-12-04T12:03:05: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":49,"battery":87,"action":"brightness_up_click"}'
zigbee2mqtt:info  2019-12-04T12:03:09: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":47,"battery":87,"action":"brightness_up_click"}'
zigbee2mqtt:info  2019-12-04T12:03:11: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe6f9aa5', payload '{"linkquality":42,"battery":87,"action":"brightness_up_click"}'

jeremywillans, 2019-12-04:
I suspect I just had a faulty one... Returned it to Ikea. Issue Closed

closed 2019-12-04 by jeremywillans